espressif/esp-thread-br #2: Enable Openthread Commissioner

motters, 2023-02-03:
Hi,

We've been testing your BR example using an ESP and RCP(nordic nrf52833 chip while we wait on the H2). We can create a thread network, connect to WiFi, etc, successfully.

We have tried to enable the openthread commissioner using the following steps:

- Set CONFIG_OPENTHREAD_COMMISSIONER=y
- Then on bootup we run the following:
  - ``commissioner start``
  - ``commissioner joiner add * JO1NU5``

All this goes successfully.

However, when devices try to commission using  the joining code "JO1NU5", they return error code 23 [not found].

Is the openthread commissioner supported / tested?

Cheers

chshu, 2023-02-05:
@motters The commissioner and joiner feature is supported and tested in our SDK.

Which Joiner are you using? `not found` means the Joiner didn't find any valid discovery response which allows it to join.
Could you try to add the specific joiner eui64 in the `commissioner joiner add` command? if it still fails, please share us the wireshark capture file.

motters, 2023-02-07:
@chshu I've tried adding the eui64 of the end device and it didn't work (see esp br log).

I've also attached the wireshark capture file containing multiple failed commissioning events.

Let me know if you need anything more.

[esp br logs.txt](https://github.com/espressif/esp-thread-br/files/10677114/esp.br.logs.txt)
[wireshark_OpenThread SnifferC9ZWZ1.pcapng.txt](https://github.com/espressif/esp-thread-br/files/10677116/wireshark_OpenThread.SnifferC9ZWZ1.pcapng.txt)

zwx1995esp, 2023-02-08:
@motters Could you please share your topology of your thread test network?
Which joiner device do you use?
Could you please share the complete command that you have executed on the commissioner and joiner?

motters, 2023-02-08:
@zwx1995esp Sure. On the ESP BR we run

```
wifi connect -s XXXX -p ZZZZ
ifconfig up
thread start 
commissioner start
commissioner joiner add ‘*’ J01NU5
```

Our devices are developed using Zephyr. They are programmed to commission to thread networks on bootup if they have no OT network credentials saved. The device's commissioning function is as below:

[image]

I found yesterday that to make commissioning successful i had to manually run the following on the device:

`ot panid 0x1234`

The made the panid on the device match that of the ESP BR.

I suppose the question is does the panid on the device have to match the ESP BR to commission?

I'm going to re-flash everything today to make sure the above is true. I should also say during development we've used the OT posix boarder router on a raspberry pi to test devices. This solution works very well but we'd much prefer a simpler ESP solution over a linux solution.

zwx1995esp, 2023-02-08:
`I suppose the question is does the panid on the device have to match the ESP BR to commission?`

Yeah, for the commissioner, only the device which has the same PANID and PSKC can join in the network.

closed 2023-02-22 by chshu