espressif/esp-thread-br #4: In RCP update, can esp32h2 only be upgraded with uart0 ?

gl-chenxiaosheng, 2023-02-22:
IDF version：commit https://github.com/espressif/esp-idf/commit/886e98a2c1311556eb6be02775d49703d6050222 (HEAD -> release/v5.0, origin/release/v5.0)

I use esp32s3 as ot_br and esp32h2 as ot_rcp to verificate RCP update. And find it only work when esp32h2 use uart0. 
If esp32h2 use uart1, it will error.
```shell
W(2685) OPENTHREAD:[W] Platform------: Wait for response timeout
W(2685) OPENTHREAD:[W] Platform------: RCP failure detected
W(2685) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
I (2695) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2705) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (4665) RCP_UPDATE: esp_rcp_update(231): Failed to connect to RCP
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759bd
0x403759bd: esp_restart_noos_dig at /home/helpsen/esp_v5.0/esp-idf/components/esp_system/esp_system.c:64 (discriminator 1)
```
Can esp32h2 only be upgraded with uart0 ?

chshu, 2023-02-23:
@gl-chenxiaosheng The RCP update feature binds to H2 uart0, since it's a feature hardcoded in ROM code. RCP update is done under H2 UART download mode.

The UART is configurable for ot_rcp firmware, after boots up, both UART0 and UART1 could be used for host and RCP communication.

gl-chenxiaosheng, 2023-02-23:
There have another question is that using example 'update_baudrate = 230400' it only upload about 15% will failed and retring. And then i swtich the update_baudrate = 115200, it will work fine. This may be caused by my connection with DuPont cable instabilitily. Would you guys have tested this update_baudrate, and what is the maximum value it can reach?

xieqinan, 2023-02-23:
@gl-chenxiaosheng  The 'update_baudrate = 230400' failure is due to incompatiable esp-idf version, you can use the tested IDF commit 14ef8d5. 
The update_baudrate is adaptive, whose value depends on your hardware support. The update_baudrate = 921600 is tested. Of course, the baudrate more higher, the updating speed more quickly.

gl-chenxiaosheng, 2023-02-24:
What does hardware support mean? Can I understand that the update rate of 921600 can be work when using both espressif chips?

xieqinan, 2023-02-24:
Yes, the update rate of 921600 is tested with ESP32s3 and ESP32H2.  The update rate also depends on the connection wires.

closed 2023-03-21 by chshu