espressif/esp-thread-br #5: Pause board router during busy CPU periods

motters, 2023-03-10:
We're currently testing the BR and found an issue where the RCP & ESP will reboot if the ESP is busy on another task, when something on OT needs servicing. 

For example, when loading the webpage under the following conditions causes the RCP & ESP to reboot. However when OT is not enable the webpage loads fine.

- Serve the webpage from an SD card (40MHz clock)
- Including "bootstrap.min.css" & "jquery.min.js" locally 
- Setup WiFi into APSTA mode

See the log of the failure below:

```
I(59308) OPENTHREAD:[I] Mac-----------: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:81, type:Beacon, src:0x0000, dst:None, sec:no, ackreq:no
I (60238) OTBR_WEB: -------------------------------------------
I (60238) OTBR_WEB: /index.html
I (60238) OTBR_WEB: -------------------------------------------
I (60248) SPIFFS: Reading /sdcard/index.html
E (60288) OPENTHREAD: dropping radio frame: Parse
E(60288) OPENTHREAD:[C] Platform------: Unexpected RCP reset: RESET_EXTERNAL
W(60288) OPENTHREAD:[W] Platform------: RCP failure detected
W(60298) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
I (60308) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (60318) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0xffffffff
I (60318) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (60328) diskio_sdmmc: sdmmc_read_blocks failed (-1)
I (60348) OTBR_WEB: -------------------------------------------
I (60348) OTBR_WEB: /static/style.css
I (60358) OTBR_WEB: -------------------------------------------
I (60368) SPIFFS: Reading /sdcard/static/style.css
E (60368) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (60368) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (60378) SPIFFS: Failed to open file.
E (60388) OTBR_WEB: style_css_get_handler(326): Fail to get style file.
I (60398) OTBR_WEB: -------------------------------------------
I (60398) OTBR_WEB: /static/bootstrap.min.css
E (60398) OTBR_WEB: Failed to stat file : /sdcard/static/bootstrap.min.css
W (60418) httpd_txrx: httpd_resp_send_err: 404 Not Found - File does not exist
I (60438) OTBR_WEB: -------------------------------------------
I (60438) OTBR_WEB: /static/jquery.min.js
E (60438) OTBR_WEB: Failed to stat file : /sdcard/static/jquery.min.js
W (60448) httpd_txrx: httpd_resp_send_err: 404 Not Found - File does not exist
I (60458) OTBR_WEB: -------------------------------------------
I (60468) OTBR_WEB: /static/restful.js
I (60468) OTBR_WEB: -------------------------------------------
I (60478) SPIFFS: Reading /sdcard/static/restful.js
E (60478) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (60488) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (60498) SPIFFS: Failed to open file.
E (60498) OTBR_WEB: script_js_get_handler(335): Fail to get script file.
E (62278) RCP_UPDATE: esp_rcp_update(231): Failed to connect to RCP
W (62288) wifi:age timer already deinit
W (62288) wifi:age timer already deinit
W (62288) wifi:hmac tx: ifx0 stop, discard
W (62298) wifi:hmac tx: ifx0 stop, discard
W (62298) wifi:hmac tx: ifx0 stop, discard
W (62298) httpd_txrx: httpd_sock_err: error in recv : 113
W (62298) wifi:hmac tx: ifx0 stop, discard
W (62308) wifi:hmac tx: ifx0 stop, discard
W (62308) wifi:hmac tx: ifx0 stop, discard
I (62328) esp_ot_br: WIFI_EVENT_STA_DISCONNECTED
W (62328) httpd_txrx: httpd_sock_err: error in recv : 113
W (62328) httpd_txrx: httpd_sock_err: error in recv : 113
W (62328) httpd_txrx: httpd_sock_err: error in recv : 113
W (62348) httpd_txrx: httpd_sock_err: error in recv : 113
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7012
load:0x40078000,len:15384
load:0x40080400,len:3816
0x40080400: _init at ??:?
```

A temporary fix i want to try is to pause the BR functionality while the webpage is served. 

The easiest way of doing this i think is `otLinkSetRadioFilterEnabled(esp_openthread_get_instance(), true);`

However this function is not available as OT needs the following set `OPENTHREAD_CONFIG_MAC_FILTER_ENABLE` see https://openthread.io/reference/group/api-link-link. How do we go about setting OPENTHREAD_CONFIG_MAC_FILTER_ENABLE?

Do you have any other suggestions on this issue?

Cheers.

xieqinan, 2023-03-14:
Firstly, the reason of RCP reboot maybe is that the UART drops the ack radio frame lead to br ocurrs RCP reboot. The CPU is busy will cause the UART's rx buffer overflow, you can try to enlarge the rx buffer.
Secondly, if the priority of reading data from SD card(or others) task is higher than ot_br task, will lead to the radio frame  is handled untimely. so you can load "bootstrap.min.css" & "jquery.min.js" from Internet instead of SD card, the above problem maybe is not existence.
Thirdly, you maybe fail to load webpage from SD card, the solution of loading webpage from SD card will test subsequently.
Finally, it is not good idea to pause the BR functionality while the webpage is served.

motters, 2023-03-14:
Could you expand on how to increase the UART buffer? 
From my understanding the buffer is set on bootup using `uart_driver_install`. This function also set the handler for the data which would be OT. I'm assuming this is executed in `esp_openthread_init` which is not publicly available (header only). 

I did try adjusting the thread priority however i'll give this another attempt.

My thoughts were to allow the web server time to serve the files, then re-enable the BR. The JS on the webpage can then be delayed slightly before attempting to load BR info.

xieqinan, 2023-03-14:
Full log is helpful for me to locate problem.

The UART buffer size can be modified by changing menuconfig's `OPENTHREAD_UART_BUFFER_SIZE` option(the feature is supported in latest IDF).

I think that the root of problem should be the BR drops the radio frame due to use uart interface by the provided log provided. Drop frame reasons maybe is due to the process of BR sends commands to RCP will wait the RCP respond, during this period the br-task is pending which leads to other tasks and ot-br task runs interspersed.
I thinks some solutions maybe is valid.
First, as you said, ensure the start-up process of br-web and ot-br task runs sequentially. 
Second, enlarge the rx buffer size to avoid the received frame been overwrited.
Third, ensure is ot-br task can be scheduled at the first time.
Fourth, Try to increate the number of failure recovery by modifing the macro of `OPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT`.

motters, 2023-03-14:
Also worth pointing out when the error occurs the following happens:

- BR attempts to read files from SD card and server to client
- OT on the BR drops an RCP radio frame
- OT on the BR drops another RCP radio frame
- The RCP resets unexpectedly (could this be due to not receiving a response from the BR?)
- BR detected the RCP failure and attempts to recover
- Reboot follows shortly after multiple SD card reading errors 

```
I (32926) OTBR_WEB: -------------------------------------------
I (32926) OTBR_WEB: /index.html
I (32926) OTBR_WEB: -------------------------------------------
I (32946) SPIFFS: Reading /sdcard/index.html
E (32996) OPENTHREAD: dropping radio frame: Parse
E (33036) OPENTHREAD: dropping radio frame: Parse
I (33036) OTBR_WEB: -------------------------------------------
E(33036) OPENTHREAD:[C] Platform------: Unexpected RCP reset: RESET_EXTERNAL
W(33046) OPENTHREAD:[W] Platform------: RCP failure detected
W(33046) OPENTHREAD:[W] Platform------: Trying to recover (1/1)I (33036) OTBR_WEB: /static/style.css
I (33066) OTBR_WEB: -------------------------------------------
I (33076) SPIFFS: Reading /sdcard/static/style.css

I (33076) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (33076) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (33086) diskio_sdmmc: sdmmc_read_blocks failed (265)
I (33086) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (33106) SPIFFS: Failed to open file.
E (33116) OTBR_WEB: style_css_get_handler(326): Fail to get style file.
I (33126) OTBR_WEB: -------------------------------------------
I (33136) OTBR_WEB: /static/bootstrap.min.css
I (33136) OTBR_WEB: -------------------------------------------
I (33146) SPIFFS: Reading /sdcard/static/bootstrap.min.css
E (33246) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (33246) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (33256) SPIFFS: Failed to open file.
E (33256) OTBR_WEB: style_css_get_handler(326): Fail to get style file.
I (33266) OTBR_WEB: -------------------------------------------
I (33276) OTBR_WEB: /static/jquery.min.js
I (33276) OTBR_WEB: -------------------------------------------
I (33276) SPIFFS: Reading /sdcard/static/jquery.min.js
E (33386) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (33396) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (33406) SPIFFS: Failed to open file.
E (33406) OTBR_WEB: script_js_get_handler(335): Fail to get script file.
I (33406) OTBR_WEB: -------------------------------------------
I (33426) OTBR_WEB: /static/restful.js
I (33426) OTBR_WEB: -------------------------------------------
I (33426) SPIFFS: Reading /sdcard/static/restful.js
E (33536) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (33536) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (33546) SPIFFS: Failed to open file.
E (33546) OTBR_WEB: script_js_get_handler(335): Fail to get script file.
E (35066) RCP_UPDATE: esp_rcp_update(231): Failed to connect to RCP
I (35086) esp_ot_br: WIFI_EVENT_STA_DISCONNECTED
W (35086) httpd_txrx: httpd_sock_err: error in recv : 113
W (35086) httpd_txrx: httpd_sock_err: error in recv : 113
W (35086) httpd_txrx: httpd_sock_err: error in recv : 113
W (35106) httpd_txrx: httpd_sock_err: error in recv : 113
W (35106) httpd_txrx: httpd_sock_err: error in recv : 113
ets Jul 29 2019 12:21:46
```

xieqinan, 2023-03-14:
Do you use the ESP-Thread-Border-Router? if not, you can try to disconnect the RST PIN in the ESP32H2.

xieqinan, 2023-03-15:
Your test is helpful.

`esp_openthread_register_rcp_failure_handler(update_rcp)` is a callback which works when rcp failure. the callback will try to update rcp firmware, `RCP reset: RESET_EXTERNAL` will cause it. 
When the br trys to update rcp, due to `RCP_UPDATE: esp_rcp_update(231): Failed to connect to RCP` causes the br and rcp reboot.

The reasion of `OPENTHREAD: dropping radio frame: Parse` maybe CPU busy or UART failure. We should keep `OPENTHREAD_UART_BUFFER_SIZE = 1024`, then  set `xTaskCreatePinnedToCore(ot_task_worker, "ot_br_main", 6144, xTaskGetCurrentTaskHandle(), 5, NULL,0);` and `xTaskCreatePinnedToCore(ot_task_br_web, "ot_task_br_web", 4096, base_path, 4, NULL,1);`

xieqinan, 2023-03-16:
Can you tell me your test environment includes esp-idf commit, esp-thread-br commit and hardware?

I try to reproduce the problem.

HeFeng1947, 2023-05-10:
Would adding serial port hardware flow control help with data drop?

chshu, 2023-06-14:
@motters Have you got the issue resolved? If yes, could you please close the issue.

chshu, 2023-07-17:
@motters Please reopne if any follow up questions.

closed 2023-07-17 by chshu