espressif/esp-thread-br #7: 启动时直接卡死并输出乱码

HeFeng1947, 2023-03-30:
IDF使用master分支：81e1e6599553a646a689ad51e32a5d48b34cfec5
硬件：H2+S3
使用IDF自带的ot_rcp和ot_br可以正常运行，但是S3跑esp-thread-br会在初始化阶段卡死。
尝试了网口的配置，log如下：
```
I (27) boot: ESP-IDF v5.1-dev-4264-g81e1e65995-dirty 2nd stage bootloader
I (27) boot: compile time Mar 30 2023 18:53:06
I (28) boot: chip revision: v0.1
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (78) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (85) boot:  3 ota_0            OTA app          00 10 00020000 002ee000
I (93) boot:  4 ota_1            OTA app          00 11 00310000 002ee000
I (100) boot:  5 web_storage      Unknown data     01 82 005fe000 0001e000
I (108) boot:  6 rcp_fw           Unknown data     01 82 0061c000 000af000
I (115) boot:  7 ot_storage       Unknown data     01 3a 006cb000 00002000
I (123) boot: End of partition table
I (127) boot: No factory image, trying OTA 0
I (132) esp_image: segment 0: paddr=00020020 vaddr=3c120020 size=42874h (272500) map
I (189) esp_image: segment 1: paddr=0006289c vaddr=3fc99f00 size=044d8h ( 17624) load
I (193) esp_image: segment 2: paddr=00066d7c vaddr=40374000 size=0929ch ( 37532) load
I (203) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=1151cch (1135052) map
I (407) esp_image: segment 4: paddr=001851f4 vaddr=4037d29c size=0cb90h ( 52112) load
I (428) boot: Loaded app from partition at offset 0x20000
I (458) boot: Set actual ota_seq=1 in otadata[0]
I (458) boot: Disabling RNG early entropy source...
I (469) cpu_start: Pro cpu up.
I (470) cpu_start: Starting app cpu, entry point is 0x403756fc
0x403756fc: call_start_cpu1 at /home/kelly/esp/esp-idf/components/esp_system/port/cpu_start.c:158

I (0) cpu_start: App cpu up.
I (484) cpu_start: Pro cpu start user code
I (484) cpu_start: cpu freq: 160000000 Hz
I (484) cpu_start: Application information:
I (487) cpu_start: Project name:     esp_ot_br
I (492) cpu_start: App version:      783e9fe-dirty
I (498) cpu_start: Compile time:     Mar 30 2023 19:25:58
I (504) cpu_start: ELF file SHA256:  e854bed7ec813aea...
I (510) cpu_start: ESP-IDF:          v5.1-dev-4264-g81e1e65995-dirty
I (517) cpu_start: Min chip rev:     v0.0
I (521) cpu_start: Max chip rev:     v0.99 
I (526) cpu_start: Chip rev:         v0.1
I (531) heap_init: Initializing. RAM available for dynamic allocation:
I (538) heap_init: At 3FCADEF0 len 0003B820 (238 KiB): DRAM
I (544) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (551) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (557) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (565) spi_flash: detected chip: gd
I (568) spi_flash: flash io: dio
I (572) sleep: Configure to isolate all GPIO pins in sleep state
I (579) sleep: Enable automatic switching of GPIO sleep configuration
I (586) app_start: Starting scheduler on CPU0
I (591) app_start: Starting scheduler on CPU1
I (591) main_task: Started on CPU0
I (601) main_task: Calling app_main()
I (651) esp_eth.netif.netif_glue: 02:00:00:12:34:56
I (651) esp_eth.netif.netif_glue: ethernet attached to netif
I (2651) ethernet_connect: Waiting for IP(s).
I (2651) ethernet_connect: Ethernet Link Up
I (3651) esp_netif_handlers: example_netif_eth ip: 192.168.113.186, mask: 255.255.255.0, gw: 192.168.113.1
I (3651) ethernet_connect: Got IPv4 event: Interface "example_netif_eth" address: 192.168.113.186
I (4611) ethernet_connect: Got IPv6 event: Interface "example_netif_eth" address: fe80:0000:0000:0000:0000:00ff:fe12:3456, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4611) example_common: Connected to example_netif_eth
I (4621) example_common: - IPv4 address: 192.168.113.186,
[... 42 lines cut ...]
I (479) cpu_start: Project name:     esp_ot_br
I (484) cpu_start: App version:      783e9fe-dirty
I (490) cpu_start: Compile time:     Mar 30 2023 19:36:58
I (496) cpu_start: ELF file SHA256:  7b467e8c343a29ed...
I (502) cpu_start: ESP-IDF:          v5.1-dev-4264-g81e1e65995-dirty
I (509) cpu_start: Min chip rev:     v0.0
I (514) cpu_start: Max chip rev:     v0.99 
I (518) cpu_start: Chip rev:         v0.1
I (523) heap_init: Initializing. RAM available for dynamic allocation:
I (531) heap_init: At 3FCAC3D0 len 0003D340 (244 KiB): DRAM
I (537) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (543) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (550) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (557) spi_flash: detected chip: gd
I (560) spi_flash: flash io: dio
I (564) sleep: Configure to isolate all GPIO pins in sleep state
I (571) sleep: Enable automatic switching of GPIO sleep configuration
I (578) app_start: Starting scheduler on CPU0
I (583) app_start: Starting scheduler on CPU1
I (583) main_task: Started on CPU0
I (593) main_task: Calling app_main()
Ya5��!1!�A�a!P!1!%A��1P�1�%5DEa1�P!5!1���!Ļ!1�1%�1�!���A�1P!5�!5���5!�5�%q1��a!�P!5!5�!1��1}�a��1!!��%A��!P5!����1�% !�!�1���eA�vP5!����1�1 ��a%P���!%�%�5��a%P���a5�%A�%P���!5�%A�%P11�1!�A��
```

尝试了开关menuconfig中的各项配置，但是都没有正常运行起来。看到RCP相关配置的波特率还是115200，IDF主分支已经切换到460800了。**目前该项目是否依赖于某个IDF的版本？**

chshu, 2023-03-30:
目前 esp-thread-br 为了支持 ESP32-H2beta2 芯片，是基于这个 IDF 版本 14ef8d5：
https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router#set-up-esp-idf

我们即将更新支持 ESP32-H2 MP 芯片。如果用最新的 IDF 主分支，需要把 br 波特率也改为 460800.

chshu, 2023-04-04:
@HeFeng1947 最新的 master 分支已经支持 H2 MP 芯片，波特率已经统一为 460800。

closed 2023-04-04 by HeFeng1947