espressif/esp-thread-br #8: esp-thread-br后续会集成ot-br-posix的restful API吗？

HeFeng1947, 2023-04-04:
目前很多生态对接OTBR都基于ot-br-posix的rest api接口，而esp-thread-br内部有一套简化的版本，但是不太足够，例如缺少网络拓扑等。如果能直接移植ot-br-posix那套C++的代码应该在后续维护和多生态适配都有一定优势。

chshu, 2023-04-06:
ot-br-posix 是基于 Linux Posix 的，ESP Thread BR 是基于 ESP- IDF (RTOS based)，所以 IP stack / mDNS / web-service 这些基础组件都不一样。

我们会对比一下 rest API，尽可能兼容 ot-br-posix.

HeFeng1947, 2023-04-07:
> ot-br-posix 是基于 Linux Posix 的，ESP Thread BR 是基于 ESP- IDF (RTOS based)，所以 IP stack / mDNS / web-service 这些基础组件都不一样。
> 
> 我们会对比一下 rest API，尽可能兼容 ot-br-posix.

感谢回复。我们现在正在和HA等第三方生态讨论基于ot-br-posix rest api的对接，如果esp方案可以兼容ot-br-posix的接口，那么我们就可以无缝接入了。

closed 2023-04-10 by HeFeng1947

chshu, 2023-05-11:
@HeFeng1947 我们更新了 rest API 接口：https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/codelab/web-gui.html#web-gui-rest-apis，保持和 ot-br-posix 兼容。

HeFeng1947, 2023-05-11:
> @HeFeng1947 我们更新了 rest API 接口：[https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/codelab/web-gui.html#web-gui-rest-apis，保持和](https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/codelab/web-gui.html#web-gui-rest-apis%EF%BC%8C%E4%BF%9D%E6%8C%81%E5%92%8C) ot-br-posix 兼容。

非常感谢，这个就是我们想要的，节省了我们自己开发的时间。