espressif/esp-thread-br #9: Issues building for ESP32-C6 + VS Code

beriberikix, 2023-04-12:
Hello! I recently got some of the new ESP32-C6 devkits and thought I try this out on one of them since I saw support was recently added. I'm using the VS Code extension, which also added support for the C6. However, I'm unable to build the example on any of my machines (Windows, macOS and Linux.)

Each error is slightly different but since I'm currently on my mac, I'll describe the error and my setup.

* x86 macbook pro
* macOS 13.0
* VS Code 1.77.3
* IDF Extension v1.6.1
* ESP-IDF on master (C6 support isn't in a stable release)

When I set up the workspace configuration and hit build, I get:

```
[5/1196] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
otadata,data,ota,0xf000,8K,
phy_init,data,phy,0x11000,4K,
ota_0,app,ota_0,0x20000,1600K,
ota_1,app,ota_1,0x1b0000,1600K,
web_storage,data,spiffs,0x340000,100K,
rcp_fw,data,spiffs,0x359000,500K,
ot_storage,data,58,0x3d6000,8K,
*******************************************************************************
[6/1196] cd /Users/jonathanberi/code/esp-thread-br/examples/basic..._border_router/build/esp-idf/main/spiffs_image/ot_rcp_0/rcp_image
FAILED: esp-idf/main/CMakeFiles/rcp_image_generation /Users/jonathanberi/code/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/main/CMakeFiles/rcp_image_generation 
cd /Users/jonathanberi/code/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/main && /Users/jonathanberi/code/esp-thread-br/examples/basic_thread_border_router/main/../create_ota_image.py --rcp-build-dir /Users/jonathanberi/esp-master/esp-idf/examples/openthread/ot_rcp/build --target-file /Users/jonathanberi/code/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/main/spiffs_image/ot_rcp_0/rcp_image
Traceback (most recent call last):
  File "/Users/jonathanberi/code/esp-thread-br/examples/basic_thread_border_router/main/../create_ota_image.py", line 94, in <module>
    main()
  File "/Users/jonathanberi/code/esp-thread-br/examples/basic_thread_border_router/main/../create_ota_image.py", line 73, in main
    fout, FILETAG_RCP_VERSION, os.path.getsize(rcp_version_path), offset)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen genericpath>", line 50, in getsize
FileNotFoundError: [Errno 2] No such file or directory: '/Users/jonathanberi/esp-master/esp-idf/examples/openthread/ot_rcp/build/rcp_version'
[11/1196] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aria.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "/usr/local/bin/bash '-c', 'ninja '" terminated with exit code: 1. 
```

I did not change any of the default configurations that are part of the example. 

Any ideas what I'm doing wrong or how to fix the build? Happy to also build on the other host OSes if that's helpful.

chshu, 2023-04-13:
Hi Jonathan:)

If you want to run the Thread BR on ESP32-C6 single SoC, please use the `ot_br` example from the ESP-IDF repo:
https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br
You can build and run the `ot_br` example without any change, just set-target to esp32c6.

The example in this `esp-thread-br` repo is designed for the two SoCs solution ([ESP BR board](https://github.com/espressif/esp-thread-br#esp-thread-border-router-board)), which builds in the RCP image together, so the RCP image could be downloaded to the ESP32-H2 (15.4) SoC from host SoC. The compile error indicates you haven't built the ot_rcp example in ESP-IDF, so it can't find the ot_rcp image.

beriberikix, 2023-04-13:
Thank you Shu, that worked! 
[image]

closed 2023-04-13 by beriberikix

chshu, 2023-04-13:
@beriberikix Goog to know!

Currently, we still recommend to use the two SoCs BR solution for production. Since there is only one RF path in ESP32-C6, which means Wi-Fi and Thread can't receive simultaneously, it will cause high packet loss rate comparing to the two SoCs soltuion.

beriberikix, 2023-04-13:
I saw that, understood. The challenge we have is that the H2 is [not available](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-H2-DEVKITM-1-N4/18109238) anywhere yet :)

Can this project work with other OpenThread compatible RCP radios from other vendors?

chshu, 2023-04-14:
The H2 will be public available soon:)

Yes, the ESP Thread BR could work with other RCP. While you can also use C6 as the RCP, that is:
- ot_br on ESP32-S3: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br
- ot_rcp on ESP32-C6: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp

Connect the two boards via UART, and run the two exampls as is.

a-d-r-i-a-n-d, 2023-04-16:
It's a shame that Wi-Fi and Thread can't receive simultaneously on the ESP32-C6. I thought that having a single microcontroller with Wi-Fi 6 as a thread border router would be a great idea. Could the packet loss be solved later on with changes in the firmware/software or is it a limitation with the design of the C6?

beriberikix, 2023-04-16:
This is usually the case in multi-radio designs, due to the physical limitations of having to share an RF path. This is very common among BLE + 802.15.4 MCUs, for example. Time slicing between the two modems becomes the main strategy. Usually software can improve to switch between the radios to a point it is "good enough" for a specific application, but it'll never be as performant as a two radio solution. 

I'm curious if the math has been done to figure out what's needed for Thread and/or Matter (typical use case) and if we can hope for a good enough solution!