espressif/esp-thread-br #13: 关于esp_openthread_lock_acquire（）的使用

HeFeng1947, 2023-05-30:
请问哪些ot接口是需要使用esp_openthread_lock_acquire（）做互斥操作的？
在esp_br_web_api.c中，部分调用ot接口时加了上锁（如：[handle_openthread_join_network_request](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#L426)），但是部分接口（如：[network_prefix_add](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#LL415C18-L415C36)）又没有。这方面esp是怎么考虑的呢？

zwx1995esp, 2023-05-31:
esp_openthread_lock_acquire() 的作用是保护 openthread 的 single instance。

我们在 ot-task 的main loop中，执行openthread相关处理时会take这个锁，当处理函数执行结束，ot-task process处理结束，不再调用任何API的时候会放掉该锁。

在其他的task中，如果想要使用openthread的API就需要拿这个锁，如果拿到了才可以调用。这个锁是递归锁，可以被take多次。

您问的这个问题中: [network_prefix_add](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#LL415C18-L415C36) 是一个static的函数，并且他在[handle_openthread_join_network_request](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#L426)这个函数中，后面这个函数是会在其他的task中执行的，所以他的处理中加了锁，[network_prefix_add](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#LL415C18-L415C36)这个函数在调用前，已经拿锁了([handle_openthread_join_network_request](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#L426) 456行拿了锁492行释放，[network_prefix_add](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#LL415C18-L415C36) 在489行调用)。 此外还有一个函数[handle_openthread_add_network_prefix_request ](https://github.com/espressif/esp-thread-br/blob/1a423f1ae146b046de1db1e9ca405e8d96b768d3/components/esp_ot_br_server/src/esp_br_web_api.c#L503)也是如此。

HeFeng1947, 2023-05-31:
> esp_openthread_lock_acquire() 的作用是保护 openthread 的 single instance。
>
> 我们在 ot-task 的main loop中，执行openthread相关处理时会take这个锁，当处理函数执行结束，ot-task process处理结束，不再调用任何API的时候会放掉该锁。 [...]

感谢解答。我可以简单的归纳为只要使用了esp_openthread_get_instance()获得的ot句柄，就需要进行加锁操作吗？

zwx1995esp, 2023-05-31:
是的

closed 2023-06-01 by HeFeng1947