espressif/esp-thread-br #14: TBR启动卡在NAT64初始化

HeFeng1947, 2023-06-15:
问题出现的比较随机，目前还没找到明确可以复现的方法。
出现问题时，openthread log最后都是输出到：
`
I (11386) OPENTHREAD: netif up
`
此时不会再有log输出（无论是openthread还是其他组件），ot cli输入指令无返回。
而正常时会输出：
`
I (11657) OPENTHREAD: netif up
I (11667) OPENTHREAD: NAT64 ready
I(11667) OPENTHREAD:[I] RouterTable---: Route table
`
初步怀疑是NAT64初始化失败，但没有找到很好的思路。

硬件是S3+H2，网口PHY使用DM9051
IDF commit：17451f1fb3d6485af5113d96b6836a7ce60efee8

HeFeng1947, 2023-06-16:
> Hi, 感谢您的关注和提出的issue。 我们内部测试也发现了这个问题，已经被解决了，是一个死锁相关的问题，相关的bugfix正在走我们内部的流程，预计很快就会同步到esp-idf的github上。如果您想快速实验，可以尝试手动修改openthread-lib:
> 
> > cd $IDF_PATH/components/openthread/lib
> > git checkout master(or commit hash: 36cb2202e10b5ba7484654962ca9e3ceb51f6d51)

感谢回复，我们稍后会测试新版本的lib。后续idf合并的时候，麻烦你在这个issue回复下。

zwx1995esp, 2023-06-16:
我们内部的已经merge了，github可能再等一下，最晚一周以后同步

zwx1995esp, 2023-06-21:
这个fix已经同步到github上了，请使用最新的idf进行测试。

closed 2023-06-21 by HeFeng1947