espressif/esp-thread-br #15: otPlatRadioGetTransmitPower()返回错误数值

HeFeng1947, 2023-06-16:
S3+H2的架构，H2规格书中最大发射功率为18dbm，而otPlatRadioGetTransmitPower()返回21。
测试发现，无论otPlatRadioSetTransmitPower()设置什么值，otPlatRadioGetTransmitPower()都只返回21.

zwx1995esp, 2023-06-19:
我们驱动这边默认会给power设置最大值，也就是上电之后默认就是配置最大的power。至于您说的H2规格书中最大发射功率为18dbm，目前官网上最新的手册是20dBm。但是我们硬件设计的最大power应该是21，这个还在和硬件的确认中，有了消息会回复您。

zwx1995esp, 2023-06-19:
刚才和硬件确认了，我们这边最大的功率就是20dBm，应该是之前理论设计是21，但是后面测下来发现在极限数值下有些芯片的一致性还不好(有的21dBm没问题，有些芯片21dBm有点问题)，所以就降低了一点最大的power。最大就是最新的datasheet写的20。

zwx1995esp, 2023-06-19:
后面会更新一版软件代码来更改一下最大的power。

HeFeng1947, 2023-06-19:
> 刚才和硬件确认了，我们这边最大的功率就是20dBm，应该是之前理论设计是21，但是后面测下来发现在极限数值下有些芯片的一致性还不好(有的21dBm没问题，有些芯片21dBm的定能不是那么好)，所以就降低了一点最大的power。最大就是最新的datasheet写的20。

之前规格书写的发射功率是以3为步进的，如果最大是20，那么otPlatRadioSetTransmitPower()要传入多少？

zwx1995esp, 2023-06-19:
目前的代码还是传入21，后面这个是软件map的，我改一下这个map，传20就可以了，但是建议传`components/hal/esp32h2/include/hal/ieee802154_ll.h` 文件中的 IEEE802154_TXPOWER_VALUE_MAX。

(相关的MR我准备提一下，走一下流程，差不多得一周的时间，不过这个应该没有关系，如果你现在的代码传入`components/hal/esp32h2/include/hal/ieee802154_ll.h` 文件中的 IEEE802154_TXPOWER_VALUE_MAX 的话后见面我的改动也是改这个，这样的话改动对你们来说就是无感的)

HeFeng1947, 2023-06-19:
> 目前的代码还是传入21，后面这个是软件map的，我改一下这个map，传20就可以了，但是建议传`components/hal/esp32h2/include/hal/ieee802154_ll.h` 文件中的 IEEE802154_TXPOWER_VALUE_MAX。 [...]

我看规格书上，发射功率是有步进3的。如果我设置了一个非步进倍数的值（例如：8），底层是会拟合到9还是到6？还是可以设置到8？

zwx1995esp, 2023-06-20:
Source Code: https://github.com/espressif/esp-idf/blob/master/components/ieee802154/driver/esp_ieee802154_pib.c#L48
在区间 [IEEE802154_TXPOWER_VALUE_MIN+3k, IEEE802154_TXPOWER_VALUE_MIN+3(k+1))  中的power会被map IEEE802154_TXPOWER_VALUE_MIN+3k

closed 2023-06-21 by HeFeng1947