espressif/esp-thread-br #17: 设置dataset active不生效

HeFeng1947, 2023-06-29:
之前一直基于旧版本开发测试，没有问题
IDF: 17451f1fb3d6485af5113d96b6836a7ce60efee8
esp_ot_cli_extension：0.3.0

最近更新到IDF：d2fb7240a713490b7a96d270ae31149e75c7ae0f esp_ot_cli_extension：0.4.6，出现了设置dataset active不生效的问题。
code中调用otDatasetSetActive（），返回正常且log打印正常，但是实际上不生效。后续在打开thread网络的时候，log显示：
···
E (935606) OPENTHREAD: esp_openthread_publish_meshcop_mdns(142): Failed to get OpenThread active dataset
···
测试了基于esp_ot_cli_extension的交互
串口输入：**dataset init new** **dataset commit active**，返回也是done，但是实际上也不生效。
串口输入：**dataset active**永远都是**Error 23: NotFound**

尝试了回退IDF和esp_ot_cli_extension的版本，应用层完全不变，功能正常。

HeFeng1947, 2023-06-29:
@zwx1995esp 请帮忙看下这个问题，ot组件存储网络信息到nvs这一部分的代码实现，近期有改动吗

chshu, 2023-06-29:
最近 IDF master 上我们更新了 nvs 的实现，删掉了 ot_storage：https://github.com/espressif/esp-idf/commit/4b86ee16fada1d1a32cbddcc55df5d057472cf2a

基于 IDF master 的代码要做如下改动：
[image]

HeFeng1947, 2023-06-29:
> 最近 IDF master 上我们更新了 nvs 的实现，删掉了 ot_storage：[espressif/esp-idf@4b86ee1](https://github.com/espressif/esp-idf/commit/4b86ee16fada1d1a32cbddcc55df5d057472cf2a)
> 
> 基于 IDF master 的代码要做如下改动： [image]

更新nvs配置之后测试正常，非常感谢！

closed 2023-06-29 by HeFeng1947