espressif/esp-thread-br #18: ESP32c3 + ESP32c6 OTBR

Gonta01, 2023-07-04:
Is it possible to use this two microcontrollers to create OTBR using [esp-thread-br](https://github.com/espressif/esp-thread-br) software?
Previously tested [ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) for ESP32c3 and [rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) for ESP32c6 and wired them using UART. It works. As I understand it isn't possible to get REST API with ot_br and ot_rcp, so is it possible to use esp-thread-br for combination of these microcontrollers?

Gonta01, 2023-07-04:
Is it possible to use this OTBR with Home Assistant?

chshu, 2023-07-04:
The esp-thread-br also works with ESP32-C3 + ESP32-C6 (RCP) combination, just configure the SDK properly:
- set `ESP_BR_BOARD_TYPE` to `ESP_BR_BOARD_STANDALONE`
- config `PIN_TO_RCP` to the correct pins, the reset and boot pins are only required if `OPENTHREAD_BR_AUTO_UPDATE_RCP` is enabled

Some comments regarding to your setup:
- C3 doesn't support external psram, need to check whether the C3 RAM resource is enough or not.
- ESP32-H2 is more suitable for RCP, which is more cost-effective, and also mess production now.

We haven't tested with Home Assistant yet, but since ESP Thread BR has passed the Thread 1.3 BR certificaiton, it should be compatible with any ecosystems which support standard Thread network.

closed 2023-07-17 by chshu