espressif/esp-thread-br #19: Build error: implicit declaration of function 'esp_openthread_auto_start'

FutureCow, 2023-07-09:
I saw this declaration was added 2 weeks ago, but when I try to build this latest version I got this error:

/esp-thread-br-main/examples/common/thread_border_router/src/border_router_launch.c:111:21: error: implicit declaration of function 'esp_openthread_auto_start'

chshu, 2023-07-10:
`esp_openthread_auto_start` is a new API introduced in esp-idf master, could you pull the latest esp-idf and try again.

FutureCow, 2023-07-10:
Thanks for the quick response. I'm quite new to ESP-IDF, so i tried to update to master (I'm using VSCode as IDE) but trying to build gives me this error:

`esp-idf/components/esp_hw_support/sleep_modes.c:148:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function)`

Also tried to setup a new esp-idf enviroment with ESP-IDF 5.2 PowerShell to build, building works if i don't select the webserver, with the webserver i got this error at the end:
```
-- Build files have been written to: C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/build/bootloader
Bootloader binary size 0x51b0 bytes. 0x2e50 bytes (36%) free.-main\examples\...br-main/examples/basic_thread_border_router/build/bootloader/bootloader.bin"
[1205/1207] Linking CXX executable esp_ot_br.elf
FAILED: esp_ot_br.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/build/esp_ot_br.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\esp_ot_br.elf.rsp -o esp_ot_br.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_join_network_request+0x64): undefined reference to `otJoinerStart'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x34): undefined reference to `otCommissionerGetState'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x38): undefined reference to `otCommissionerStart'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x3c): undefined reference to `otCommissionerAddJoinerWithDiscerner'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x40): undefined reference to `otCommissionerAddJoiner'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj): in function `handle_openthread_join_network_request':
C:/Espressif/frameworks/esp-thread-br-main/components/esp_ot_br_server/src/esp_br_web_api.c:468: undefined reference to `otJoinerStart'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj): in function `handle_openthread_network_commission_request':
C:/Espressif/frameworks/esp-thread-br-main/components/esp_ot_br_server/src/esp_br_web_api.c:601: undefined reference to `otCommissionerGetState'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Espressif/frameworks/esp-thread-br-main/components/esp_ot_br_server/src/esp_br_web_api.c:606: undefined reference to `otCommissionerStart'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Espressif/frameworks/esp-thread-br-main/components/esp_ot_br_server/src/esp_br_web_api.c:620: undefined reference to `otCommissionerAddJoinerWithDiscerner'
c:/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Espressif/frameworks/esp-thread-br-main/components/esp_ot_br_server/src/esp_br_web_api.c:623: undefined reference to `otCommissionerAddJoiner'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router\build\log\idf_py_stderr_output_37088 and C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router\build\log\idf_py_stdout_output_37088
```

chshu, 2023-07-10:
These error could be fixed by enabling the `commissioner` and `joiner` features under OpenThread component in menuconfig

xieqinan, 2023-07-11:
Hi @FutureCow,
The current REST API does not support accessing the `http://OTBR_IP:80/node/dataset/active` resource. If you want to retrieve the active dataset of Thread, you can use the `http://OTBR_IP:80/node/active-dataset-tlvs`. For more REST API options, you can refer to the [web-gui docs](https://github.com/espressif/esp-thread-br/blob/main/docs/en/codelab/web-gui.rst).

closed 2023-07-25 by chshu