espressif/esp-thread-br #21: Build error: undefined reference to esp_openthread_register_rcp_failure_handler, esp_openthread_rcp_deinit, mdns_hostname_get, mdns_delegate_hostname_set_address

DanTup, 2023-07-16:
I'm having trouble building the example project for an ESP-C6(-DevKitC-1). I'm completely new to ESP-IDF so I may be doing something wrong, but I'm not sure how to debug.

I'm using VS Code with the ESP-IDF extension. I've cloned this repo (e23be3a858cc3d23e294b13435dc4b1d844522ea) and the master branch of ESP-IDF (https://github.com/espressif/esp-idf/commit/4fc2e5cb95abb1e7339f11929bc9e916a9c3d15a).

When I click the build button in VS Code, here's the output I get:

```
[1/7] cd /Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_ot_br_server && /Users/danny/Dev/Microcontrollers/ESP-Tools/python_env/idf5.2_py3.9_env/bin/python /Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/spiffs/spiffsgen.py 0x19000 /Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/components/esp_ot_br_server/frontend /Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/examples/basic_thread_border_router/build/web_storage.bin --page-size=256 --obj-name-len=32 --meta-len=4 --use-magic --use-magic-len
[2/7] Performing build step for 'bootloader'
[1/1] cd /Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/examples/basic_thread_border_router/build/bootloader/esp-idf/esptool_py && /Users/danny/Dev/Microcontrollers/ESP-Tools/python_env/idf5.2_py3.9_env/bin/python /Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/examples/basic_thread_border_router/build/bootloader/bootloader.bin
Bootloader binary size 0x5400 bytes. 0x2c00 bytes (34%) free.
[3/5] Linking CXX executable esp_ot_br.elf
FAILED: esp_ot_br.elf 
: && /Users/danny/Dev/Microcontrollers/ESP-Tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/examples/basic_thread_border_router/build/esp_ot_br.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/esp_ot_br.elf.dir/project_elf_src_esp32c6.c.obj -o esp_ot_br.elf -L/Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/soc/esp32c6/ld   -L/Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/esp_rom/esp32c6/ld   -L/Users/danny/Dev/Microcontrollers/ESP-Projects/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_system/ld   -L/Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/esp_phy/lib/esp32c6   -L/Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/esp_coex/lib/esp32c6   -L/Users/danny/Dev/Microcontrollers/ESP-IDF-master/esp-idf/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/openthread/libopenthread.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__esp-serial-flasher/libespressif__esp-serial-flasher.a  esp-idf/iperf/libiperf.a  esp-idf/espressif__esp_ot_cli_extension/libespressif__esp_ot_cli_extension.a  esp-idf/espressif__esp_rcp_update/libespressif__esp_rcp_update.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/main/libmain.a  esp-idf/border_router_board/libborder_router_board.a  esp-idf/thread_border_router/libthread_border_router.a  esp-idf/esp_br_http_ota/libesp_br_http_ota.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/esp_ot_br_server/libesp_ot_br_server.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/espressif__esp_ot_cli_extension/libespressif__esp_ot_cli_extension.a  esp-idf/iperf/libiperf.a  esp-idf/border_router_board/libborder_router_board.a  esp-idf/esp_br_http_ota/libesp_br_http_ota.a  esp-idf/espressif__esp_rcp_update/libespressif__esp_rcp_update.a  esp-idf/espressif__esp-serial-flasher/libespressif__esp-serial-flasher.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/json/libjson.a  esp-idf/spiffs/libspiffs.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/ [... cut ...]
(.text.handle_host_update+0x162): undefined reference to `mdns_delegate_hostname_set_address'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I'm not entirely sure how to interpret these errors (I'm also a C++ newbie). It's also not clear to me if I need a specific commit of ESP-IDF to build this (since I can see there have been many commits mentioning openthread in ESP-IDF over the last few weeks, since this repo was last updated).

Any pointers would be appreciated. Thanks!

chshu, 2023-07-17:
@DanTup Seems you were trying to run the [basic_thread_border_router](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router) on ESP32-C6 single SoC, but it is not supported.

The ESP Thread BR is based on the two SoCs solution, BR and OT core stack run on host SoC Wi-Fi (ESP32-S3), OT RCP runs on 802.15.4 SoC (ESP32-H2). Could you please run the example following the docs: https://docs.espressif.com/projects/esp-thread-br, you can either purchase a ESP Thread Border Router board, or setup the same connection between S3 and H2 modules ([schematic](https://dl.espressif.com/dl/schematics/esp_thread_br_zigbee_gw_schematiccs_v1.2.pdf)).

It's possible to run [ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) on ESP32-C6 single SoC, but it's only a prototype, and not recommended for productization as documented in its readme:

> The example could also run on a single SoC which supports both Wi-Fi and Thread (e.g., ESP32-C6), but since there is only one RF path in ESP32-C6, which means Wi-Fi and Thread can't receive simultaneously, it has a significant impact on performance. Hence the two SoCs solution is recommended.

The [basic_thread_border_router](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router) supports the two SoCs board only.

zwx1995esp, 2023-07-18:
Hi, could you please pull the latest code for the `esp-thread-br`, we updated the mdns version just now and there should not be any compiling errors.
For your questions: could 2x C6 be supported by this repo? the answer should be yes, you can have a try, but you need to connect the two boards together manually. Refer to the pin map here: [standalone-modules](https://github.com/espressif/esp-thread-br#standalone-modules).

DanTup, 2023-07-26:
Yep, I'm seeing the same errors as @Gonta01 above (but no longer errors about mdns).

It's still very possible I'm doing something wrong. I tried to order some ESP32-H2 but the order was rejected by Mouser because of export restrictions (apparently they can't ship it from the US to the UK). Currently the only option I have is using ESP32-S2 and/or ESP32-C6 boards.

chshu, 2023-09-08:
@DanTup @Gonta01 The default configuration in the example works as is on the ESP Thread Border router board. Please follow the docs here for any customized configurations: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#configure-esp-thread-border-router

closed 2023-09-08 by chshu

DanTup, 2023-11-24:
Sorry I never responded here sooner. I ended up ordering the Border Router board and following [these instructions](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html) using the latest code, and I had no errors building :-)

I did hit a really strange issue where booting the BR would cause a server on my network to reboot all of its kubernetes pods though - I opened an issue at https://github.com/espressif/esp-thread-br/issues/45 with some questions to try and better understand exactly what the BR is doing.