espressif/esp-thread-br #22: esp-thread-br router build fails with Errors.

topraveenpaul, 2023-07-19:
Target hardware: The Wi-Fi based ESP Thread Border Router consists of two SoCs:
Host: Raspberry Pi 4 8GB on Ubuntu 20.04 LTS 64 bit.

Followed the exact steps detailed here: https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/dev-guide/build_and_run.html

Succesfully completed  until :
esp-idf/examples/openthread/ot-rcp
idf.py --preview set-target esp32h2
idf.py menuconfig
idf.py build

Encountered the below error while building the esp-thread-br.

cd esp-thread-br/examples/basic_thread_border_router
idf.py menuconfig
idf.py build

The build step failed with below errors. Please check the attachment for the complet logs.

Question: Is there a working code that I can simply build and flash it to the board? Is there any specific commits that is 100% working?

**_```
home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c: In function 'esp_ot_process_iperf':
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c:33:8: error: 'iperf_cfg_t' has no member named 'format'
   33 |     cfg.format = KBITS_PER_SEC;
      |        ^
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c:33:18: error: 'KBITS_PER_SEC' undeclared (first use in this function)
   33 |     cfg.format = KBITS_PER_SEC;
      |                  ^~~~~~~~~~~~~
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c:33:18: note: each undeclared identifier is reported only once for each function it appears in
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c:112:32: error: 'iperf_cfg_t' has no member named 'format'
  112 |                             cfg.format = idx;
      |                                ^
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c:117:65: error: 'iperf_cfg_t' has no member named 'format'
  117 |                 otCliOutputFormat("f:%sbit/s\n", strcmp(unit[cfg.format], "B") == 0 ? "\0" : unit[cfg.format]);
      |                                                                 ^
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_iperf.c:117:102: error: 'iperf_cfg_t' has no member named 'format'
  117 |                 otCliOutputFormat("f:%sbit/s\n", strcmp(unit[cfg.format], "B") == 0 ? "\0" : unit[cfg.format]);
      |                                                                                                      ^
[1148/1183] Building C object esp-idf/espressif__esp_ot_cli_extension/CMakeFiles/__idf_espressif__esp_ot_cli_extension.dir/src/esp_ot_udp_socket.c.objFAILED: esp-idf/espressif__esp_ot_cli_extension/CMakeFiles/__idf_espressif__esp_ot_cli_extension.dir/src/esp_ot_udp_socket.c.obj
```_**

Also,

**_```
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_udp_socket.c: In function 'esp_ot_process_mcast_group':
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_udp_socket.c:481:5: error: implicit declaration of function 'esp_openthread_task_switching_lock_release' [-Werror=implicit-function-declaration]
  481 |     esp_openthread_task_switching_lock_release();
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_udp_socket.c:496:5: error: implicit declaration of function 'esp_openthread_task_switching_lock_acquire' [-Werror=implicit-function-declaration]
  496 |     esp_openthread_task_switching_lock_acquire(portMAX_DELAY);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1149/1183] Building C object esp-idf/espressif__esp_ot_cli_extension/CMakeFiles/__idf_espressif__esp_ot_cli_extension.dir/src/esp_ot_ip.c.objninja: build stopped: subcommand failed.
HINT: Please check that the function name is correct. Also it is possible that you've forgot to import esp_openthread_task_switching_lock_release library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /home/pi/esp/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_35298 and /home/pi/esp/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_35298
```_**

[tbr_failed.txt](https://github.com/espressif/esp-thread-br/files/12100290/tbr_failed.txt)

chshu, 2023-07-20:
@topraveenpaul Seems you were using an old IDF version, could you fetch the latest IDF master branch and try again. Feel free to comment if any further issues.

topraveenpaul, 2023-07-20:
@chshu . Thank you for your response. Yes, it works after I feteched from the master. I will close the issue now.

closed 2023-07-20 by topraveenpaul