espressif/esp-thread-br #23: Border Router doesn't apply network settings after reboot

koenvervloesem, 2023-07-20:
I flashed this firmware to the Thread Border Router development board and I formed a network in the Web GUI with my own choices for network name, (X)PAN ID, network key, and channel. However, after a reboot the network status in the Web GUI as well as the `dataset commit active` command in the CLI show that it has set up a network with the default settings.

The custom network settings are still available after a reboot in the "Form Thread Network" web form, but they aren't applied anymore. Also a `dataset commit active` command in the CLI and rebooting after applying the changes in the "Form Thread Network" doesn't change this behavior. How do I make the firmware apply my custom network settings from the Web UI?

gytxxsy, 2023-07-24:
Currently, this problem occurs if you open `OPENTHREAD_BR_AUTO_START` via `idf.py menuconfig`. I would like to know if you use it? If so, you can turn off it as a temporary workaround, and we are actively working on this issue. We will keep you informed when there is an update.

gytxxsy, 2023-07-24:
> I flashed this firmware to the Thread Border Router development board and I formed a network in the Web GUI with my own choices for network name, (X)PAN ID, network key, and channel. However, after a reboot the network status in the Web GUI as well as the `dataset active` command in the CLI show that it has set up a network with the default settings.
> 
> The custom network settings are still available after a reboot in the "Form Thread Network" web form, but they aren't applied anymore. Also a `dataset commit active` command in the CLI and rebooting after applying the changes in the "Form Thread Network" doesn't change this behavior. How do I make the firmware apply my custom network settings from the Web UI?

Currently, this problem occurs if you open OPENTHREAD_BR_AUTO_START via idf.py menuconfig. I would like to know if you use it? If so, you can turn off it as a temporary workaround, and we are actively working on this issue. We will keep you informed when there is an update.

chshu, 2023-07-25:
Fixed in https://github.com/espressif/esp-thread-br/commit/aa6e6ca8f6d6984712e715e7cb6318e9282b48f7

closed 2023-07-25 by chshu

koenvervloesem, 2023-07-26:
Thanks for fixing this, I've confirmed it works now!