espressif/esp-thread-br #24: OTA update

FHFS, 2023-07-24:
Hello,

I'm trying to do a OTA update on the esp-thread-br. I've enabled `CONFIG_OPENTHREAD_BR_CREATE_OTA_IMAGE` in menuconfig, and the br_ota_image is generated in the build folder.

I'm serving through a python http server with: `python -m http.server -b 192.168.3.47 8070`

However when I call `ota download http://192.168.3.47:8070/br_ota_image` on the esp-thread-br, I get an error.  Does anyone know why?
```
> ota download http://192.168.3.47:8070/br_ota_image

I (26815) BR_OTA: Downloading http://192.168.3.47:8070/br_ota_image, RCP target file /rcp_fw/ot_rcp_1/rcp_image
I (29655) BR_OTA: subfile_info: tag 0xff size 84 offset 0
I (29655) BR_OTA: subfile_info: tag 0x0 size 72 offset 84
I (29655) BR_OTA: subfile_info: tag 0x1 size 24 offset 156
I (29655) BR_OTA: subfile_info: tag 0x2 size 20976 offset 180
I (29655) BR_OTA: subfile_info: tag 0x3 size 3072 offset 21156
I (29655) BR_OTA: subfile_info: tag 0x4 size 312896 offset 24228
I (29655) BR_OTA: subfile_info: tag 0x5 size 1577136 offset 337124
I (29665) BR_OTA: Downloaded 1108 bytes
…..
I (34715) BR_OTA: Downloaded 99412 bytes
E (40025) BR_OTA: download_ota_image(258): Failed to write OTA
E (40025) BR_OTA: esp_br_http_ota(292): Failed to download ota image
Failed to download imageI (40025) wifi:state: run -> init (0)
I (40025) wifi:pm stop, total sleep time: 1075574 us / 35919614 us
I (40025) wifi:<ba-del>idx:0, tid:0
I (40025) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (40075) wifi:flush txq
I (40075) wifi:stop sw txq
I (40075) wifi:lmac stop hw txq
I (40075) wifi:Deinit lldesc rx mblock:10
```

FHFS, 2023-07-28:
Does anyone have any idea how I can fix this? or what is causing this?

zwx1995esp, 2023-08-01:
@FHFS Hi, sorry for the delay, I will go on your issue right now. 
I try to reproduce your issue but failed, all things work well on my side. So could you please provide more details for your issue?
1.Do you try to use the OTA step shown in the [BR docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/dev-guide/ota_local_server.html)? And enable a http server using `openssl`?
2.You say that `it only downloads the first time after an erase.` But in your logs, the OTA image tried to be saved on `/rcp_fw/ot_rcp_1/rcp_image` which might be empty at the first time, if you re-run OTA, it should try to be saved on `/rcp_fw/ot_rcp_0/rcp_image`.[refer to the OTA Image Rollback Mechanism](https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/dev-guide/ota_update.html#the-rcp-image-rollback-mechanism).
3. For your logs, we can see that:
```
I (34715) BR_OTA: Downloaded 99412 bytes
E (40025) BR_OTA: download_ota_image(258): Failed to write OTA
```
the timestamp for the first line is 34715ms and the second is 40025ms, it means that after almost 5s sence the last downloaded data from the server. Actually our http client have a default timeout [`5s`](https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L145) for receiving data from server, if timeout it will return right now. So there might be some network issue(maybe some congestions or network broken)

FHFS, 2023-08-01:
> 1.Do you try to use the OTA step shown in the [BR docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/dev-guide/ota_local_server.html)? And enable a http server using openssl?

No, I am just using a python http server. I will try to follow the guide with openssl, and get back to you.

> 2.You say that it only downloads the first time after an erase. But in your logs, the OTA image tried to be saved on /rcp_fw/ot_rcp_1/rcp_image which might be empty at the first time, if you re-run [...]

I've included the logs of the first and second time downloading the OTA file:
First time:
```
I (1659169) OPENTHREAD: Received PIOI (1671149) OPENTHREAD: Received PIO
> ota download http://192.168.2.114:8070/br_ota_image

I (1677139) BR_OTA: Downloading http://192.168.2.114:8070/br_ota_image, RCP target file /rcp_fw/ot_rcp_1/rcp_image

I (1677279) BR_OTA: subfile_info: tag 0xff size 84 offset 0

I (1677279) BR_OTA: subfile_info: tag 0x0 size 72 offset 84

I (1677279) BR_OTA: subfile_info: tag 0x1 size 24 offset 156

I (1677279) BR_OTA: subfile_info: tag 0x2 size 21056 offset

I (1677279) BR_OTA: subfile_info: tag 0x3 size 3072 offset 2
I (1677279) BR_OTA: subfile_info: tag 0x4 size 318416 offset 24308

I (1677289) BR_OTA: subfile_info: tag 0x5 size 1597776 offset 342724

I (1677299) BR_OTA: Downloaded 1108 bytes
....
.... removed some lines showing downloaded byte count
....
I (1682309) BR_OTA: Downloaded 95316 bytes
I (1683129) OPENTHREAD: Received PIO
E (1688509) BR_OTA: download_ota_image(258): Failed to write OTA
E (1688509) BR_OTA: esp_br_http_ota(292): Failed to download ota image
Failed to download imageI (1688509) wifi:state: run -> init (0)
I (1688509) wifi:pm stop, total sleep time: 892964 us / 1684291987 us

I (1688509) wifi:new:<13,0>, old:<13,2>, ap:<255,255>, sta:<13,2>, prof:1
I (1688559) wifi:flush txq
I (1688559) wifi:stop sw txq
I (1688559) wifi:lmac stop hw txq
I (1688559) wifi:Deinit lldesc rx mblock:10
```

Second time:
```
I (133318) OPENTHREAD: Received PIO
> ota download I (146318) OPENTHREAD: Received PIO
> ota download http://192.168.2.114:8070/br_ota_image

I (152218) BR_OTA: Downloading http://192.168.2.114:8070/br_ota_image, RCP target file /rcp_fw/ot_rcp_1/rcp_image

E (152278) BR_OTA: download_ota_image(217): Failed to open target file
E (152278) BR_OTA: esp_br_http_ota(292): Failed to download ota image
Failed to download imageI (152278) wifi:state: run -> init (0)
I (152278) wifi:pm stop, total sleep time: 875004 us / 148052536 us

I (152278) wifi:new:<13,0>, old:<13,2>, ap:<255,255>, sta:<13,2>, prof:1
I (152318) wifi:flush txq
I (152318) wifi:stop sw txq
I (152318) wifi:lmac stop hw txq
I (152318) wifi:Deinit lldesc rx mblock:10
```

>  So there might be some network issue(maybe some congestions or network broken)

I don't think so, I've used curl and wget to download the file from multiple devices on the same wifi network and the file is downloaded without a problem.

zwx1995esp, 2023-08-02:
For 1. I have tried for almost 20 times, no errors occur and all passed.

For 2. for your test scenario, could you please share your whole test step by step? (maybe start from how to build the example and how to set up a http server and copy which file on the server and everythings related to your test environment, the more details the more better) I want to use your step to reproduce this issue.

FHFS, 2023-08-02:
I will do a write-up for my scenario. I will get back to you asap.

FHFS, 2023-08-03:
I've followed this [guide](https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/dev-guide/ota_local_server.html). It does not seem very comprehensive to me.

Lets start with my build setup:

Build command: `docker run --rm -v ${workspaceFolder}:/IoT/esp-thread-br -w /IoT/esp-thread-br/examples/basic_thread_border_router espressif/idf:latest@sha256:62387daaa65bfc7dae4f04a4b4c573a592a5a8ecca1a04018357ce5ee64205cb idf.py build`

esp-thread-br git commit: 7fd29e78ca2713c7e7e09a96d60820a0dc6d66ff

sdkconfig:
```
CONFIG_OPENTHREAD_BR_AUTO_UPDATE_RCP=y
CONFIG_OPENTHREAD_BR_CREATE_OTA_IMAGE=y
CONFIG_OPENTHREAD_BR_AUTO_START=y
CONFIG_OPENTHREAD_BR_START_WEB=y

CONFIG_OPENTHREAD_NETWORK_NAME="NEST-PAN-5961"
CONFIG_OPENTHREAD_NETWORK_CHANNEL=19
CONFIG_OPENTHREAD_NETWORK_PANID=0x5961
CONFIG_OPENTHREAD_NETWORK_EXTPANID="1952473fb4d78dda"
CONFIG_OPENTHREAD_NETWORK_MASTERKEY="ab11344873cc3fc3889ab62784e9e05d"
CONFIG_OPENTHREAD_NETWORK_PSKC="104810e2315100afd6bc9215a6bfac53"

CONFIG_EXAMPLE_WIFI_SSID="myrouter"
CONFIG_EXAMPLE_WIFI_PASSWORD="mypassword"
```

`CONFIG_OPENTHREAD_BR_CREATE_OTA_IMAGE` creates a OTA file in the build folder. I copy the file to a empty folder where I start the http or https server from the commandline.

http server called from commandline: `python -m http.server -b 192.168.3.47 8070`

https server called from python script:
``` python
import http.server, ssl, socketserver
context = ssl.SSLContext(ssl.PROTOCOL_TLS_SERVER)
context.load_cert_chain("ca_cert.pem","ca_key.pem")
server_address = ("192.168.3.47", 8070)
handler = http.server.SimpleHTTPRequestHandler
with socketserver.TCPServer(server_address, handler) as httpd:
    httpd.socket = context.wrap_socket(httpd.socket, server_side=True)
    httpd.serve_forever()
```

When generating the pem with: `openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365-nodes`
CommonName: "192.168.3.47"

Calling the script with: `python script.py` from the folder contain the `br_ota_image` file.

I can use my browser to go to `https://192.168.3.47:8070/` and download the `br_ota_image` file if I ignore the certificate warning.

It does not matter if I use a HTTP or HTTPS server. The result is the same on my esp-thread-br

I did the same thing again with a breakpoint on `components/esp_br_http_ota/src/esp_br_http_ota.c:258`. Here I can see `r` is -1 returned by `write_file_for_length`. It seems the write operation was retried 5 times. I also noticed the first part of the download goes pretty quickly, and slows down when reaching 70000 bytes and slows down further until 95000b

zwx1995esp, 2023-08-08:
Hi, I reproduce this issue, I can investigate this issue now. but it is too hard for me to reproduce it on my side....

FHFS, 2023-08-08:
Great news you were able to reproduce the issue. Why is it hard to reproduce for you? 

Also you were able to "ota download" without problems, which step in my writeup is different from your setup?

zwx1995esp, 2023-08-10:
@FHFS  Hi, Sorry for the delay, these days I jumped for another work, and I do not have much more time on it, I synced your issue to @gytxxsy and he will help you to investigate on your issues.

gytxxsy, 2023-08-10:
I'll be focusing on this issue recently and will update as soon as I make progress.

gytxxsy, 2023-08-11:
I have successfully reproduced this issue, it seems it is a problem caused by insufficient storage.
You can try modifying `esp-thread-br/examples/basic_thread_border_router/partition.csv` to temporarily fix this issue. Below is an example:
```javascript
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,        data, nvs,      , 0x6000,
otadata,    data, ota,      , 0x2000,
phy_init,   data, phy,      , 0x1000,
ota_0,      app,  ota_0,    , 1500K,
ota_1,      app,  ota_1,    , 1500K,
web_storage,data, spiffs,   , 100K,
rcp_fw,     data, spiffs,   , 820K,
```
After I use this configuration, I can solve the problem of OTA download failure. I hope this helps you. 
In addition, the size of the OTA image I use is different from yours, so this configuration may not be fully applicable to yours, and you may need to make some adjustments.

FHFS, 2023-08-11:
I've updated the partition table. Got some overflow and decided to shrink by optimizing for size instead of debug with `CONFIG_COMPILER_OPTIMIZATION_SIZE=y`

The download finishes, but I get stuck at the end. These are my logs:
```
I (122370) BR_OTA: 1436832/1436832 bytes
I (122380) esp_image: segment 0: paddr=001a0020 vaddr=3c110020 size=4366ch (276076) map
I (122410) esp_image: segment 1: paddr=001e3694 vaddr=3fc96800 size=04f00h ( 20224) 
I (122420) esp_image: segment 2: paddr=001e859c vaddr=4037
I (122420) esp_image: segment 3: paddr=001f0020 vaddr=42000020 size=103f18h (1064728) map
I (122560) esp_image: segment 4: paddr=002f3f40 vaddr=4037ba7c size=0ad50h ( 44368) 
I (122580) esp_image: segment 0: paddr=001a0020 vaddr=3c110020 size=4366ch (276076) map
I (122610) esp_image: segment 1: paddr=001e3694 vaddr=3fc96800 size=04f00h ( 20224)I (122620) esp_image: segment 2: paddr=001e859c vaddr=40374000 size=07a7ch ( 31356) 
I (122620) esp_image: segment 3: paddr=001f0020 vaddr=42000020 size=103f18h (1064728) map
I (122760) esp_image: segment 4: paddr=002f3f40 vaddr=4037ba7c size=0ad50h ( 44368) 
I (122960) wifi:state: run -> init (0)
I (122960) wifi:pm stop, total sleep time: 715764 us / 118751525 us

I (122960) wifi:new:<13,0>, old:<13,2>, ap:<255,255>, sta:<13,2>, prof:1
I (123010) wifi:flush txq
I (123010) wifi:stop sw txq
I (123010) wifi:lmac stop hw txq
I (123010) wifi:Deinit lldesc rx mblock:10
```

Questions:
1. Why doesn't the OTA update finish.
2. I don't understand why shrinking OTA_0/1 and expanding rcp_fw partitions fixes the first problem with downloading.

gytxxsy, 2023-08-14:
1. It seems that the OTA has been updated, but there is some failure when connecting to wifi, so it gets stuck.
2. Because the direct cause of the failure of the firmware update is the lack of rcp_fw space, so it needs to be expanded (reducing OTA_0/1 is to save more space for rcp_fw, because the flash of the BR board is only 4MB ).
3. I am not clear about this and trying to reproduce this problem.

gytxxsy, 2023-08-17:
There are some mistakes in my description above, corrected here:
In 1, the program doesn't get stuck when connecting to wifi, but when the OTA is done and the program restarts.
In this regard, we think that there may be some problems with the OTA image used, which can also explain why the hash is incorrect after restarting.

I've tried many times on my side, but can't reproduce this problem. However, you can make sure that you copied the correct OTA image to the correct server path. I thought this might be of some help.

FHFS, 2023-08-17:
I just tried it again with the br_ota_image I just generated. Same result. I will try this again with a fresh clone of the repository and let you know the results.

FHFS, 2023-08-17:
I also tried the same thing with a fresh clone.
1. I changed the partitions.csv
2. Changed sdkconfig
```
git diff examples/basic_thread_border_router/sdkconfig | grep '^[+-][^+-]'
+# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
-CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
-# CONFIG_ESPTOOLPY_FLASHFREQ_40M is not set
+# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
+CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
-CONFIG_ESPTOOLPY_FLASHFREQ="80m"
+CONFIG_ESPTOOLPY_FLASHFREQ="40m"
-# CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE is not set
+CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE=y
-# CONFIG_OPENTHREAD_BR_CREATE_OTA_IMAGE is not set
-# CONFIG_OPENTHREAD_BR_AUTO_START is not set
-# CONFIG_OPENTHREAD_BR_START_WEB is not set
+CONFIG_OPENTHREAD_BR_CREATE_OTA_IMAGE=y
+CONFIG_OPENTHREAD_BR_AUTO_START=y
+CONFIG_OPENTHREAD_BR_START_WEB=y
-CONFIG_EXAMPLE_WIFI_SSID="myssid"
-CONFIG_EXAMPLE_WIFI_PASSWORD="mypassword"
+CONFIG_EXAMPLE_WIFI_SSID="FreshTomato24"
+CONFIG_EXAMPLE_WIFI_PASSWORD="asdf1234"
-CONFIG_COMPILER_OPTIMIZATION_DEBUG=y
-# CONFIG_COMPILER_OPTIMIZATION_SIZE is not set
+# CONFIG_COMPILER_OPTIMIZATION_DEBUG is not set
+CONFIG_COMPILER_OPTIMIZATION_SIZE=y
-# CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP is not set
+CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP=y
-CONFIG_FREERTOS_TASK_FUNCTION_WRAPPER=y
-CONFIG_OPENTHREAD_NETWORK_NAME="OpenThread-ESP"
-CONFIG_OPENTHREAD_NETWORK_CHANNEL=15
-CONFIG_OPENTHREAD_NETWORK_PANID=0x1234
-CONFIG_OPENTHREAD_NETWORK_EXTPANID="dead00beef00cafe"
-CONFIG_OPENTHREAD_NETWORK_MASTERKEY="00112233445566778899aabbccddeeff"
+CONFIG_OPENTHREAD_NETWORK_NAME="NEST-PAN-5961"
+CONFIG_OPENTHREAD_NETWORK_CHANNEL=19
+CONFIG_OPENTHREAD_NETWORK_PANID=0x5961
+CONFIG_OPENTHREAD_NETWORK_EXTPANID="1952473fb4d78dda"
+CONFIG_OPENTHREAD_NETWORK_MASTERKEY="ab11344873cc3fc3889ab62784e9e05d"
-# CONFIG_OPENTHREAD_COMMISSIONER is not set
-# CONFIG_OPENTHREAD_JOINER is not set
+CONFIG_OPENTHREAD_COMMISSIONER=y
+CONFIG_OPENTHREAD_JOINER=y
-CONFIG_OPTIMIZATION_LEVEL_DEBUG=y
-CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG=y
-CONFIG_COMPILER_OPTIMIZATION_DEFAULT=y
-# CONFIG_OPTIMIZATION_LEVEL_RELEASE is not set
-# CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE is not set
+# CONFIG_OPTIMIZATION_LEVEL_DEBUG is not set
+# CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG is not set
+# CONFIG_COMPILER_OPTIMIZATION_DEFAULT is not set
+CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
+CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE=y
-# CONFIG_OTA_ALLOW_HTTP is not set
+CONFIG_OTA_ALLOW_HTTP=y
```
3. esp-idf commit 02f5b9a8982db2d59b593fab4e01ecffd35cfdff
4. build ot-rcp 
5. build esp-thread-br
6. serve br_ota_image
7. flash and monitor esp-thread-br
8. ota download http://192.168.3.47/br_ota_image

FHFS, 2023-08-18:
I meant to say that I got stuck at the same part as before

gytxxsy, 2023-08-18:
Yes, I got it am trying to reproduce the issue.

gytxxsy, 2023-08-18:
I used your config and commit and reproduced the wrong hash. However, the problem of program stuck was not reproduced. After testing, I think it is caused by the macro `CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE` being set to true, you can close it and try building the image again.
To close the macro, you can run `idf.py menuconfig` and diselect `Serial flasher config ---> Detect flash size when flashing bootloader`.
In fact, even without using OTA, but directly using `idf.py -p PORT flash`, the hash value will also be wrong (if you turn on the macro said above).

FHFS, 2023-08-21:
Indeed, when I disable the FLASHSIZE_UPDATE config, the wrong hash does not show.

I will try to debug the freezing, maybe I will get some hint about what is causing this.

FHFS, 2023-08-22:
When debugging I just hit "components/esp_ot_cli_extension/src/esp_ot_ota_commands.c line 42" esp_restart() and then it is stuck.

I've searched on esp-idf issues, maybe this issue [#9887](https://github.com/espressif/esp-idf/issues/9887) is related.

In this issue [anatol-grabowski](https://github.com/anatol-grabowski) mentions creating a task to restart using `xTaskCreate`. This works for me, but only with a larger stack size, 1024 stack overflows. 

[JustQY](https://github.com/JustQY) mentions `esp_system_abort` which also works. However using a task to restart seems to me to be the proper way to do this.

gytxxsy, 2023-08-23:
We usually don't create a new task to call `esp_restart`. Of course, if you suspect that the problem is caused by stack overflow, it's a good idea to increase the stack of [ot_task_worker](https://github.com/espressif/esp-thread-br/blob/f03f1cad7b9631091ee873b042b89a9b0e0bfb68/examples/common/thread_border_router/src/border_router_launch.c#L134) to test.
Can you give me some more information to reproduce the issue? The information needed are as follows:
1. idf commit ID and esp-thread-br commit ID.
2. The modification of the codes(such as `partitions.csv` I told you serveral days ago).
3. After fullclean and set target, which configurations did you change via menuconfig?

I  think reproducing the issue locally will help me to find the root reason of this issue.

FHFS, 2023-08-24:
> We usually don't create a new task to call esp_restart. Of course, if you suspect that the problem is caused by stack overflow, it's a good idea to increase the stack of [...]

I increased the stack of the task to 10240, and now it does continue a little bit further. See log:

``` logs

I (74574) BR_OTA: 1436672/1436912 bytes
I (74594) BR_OTA: 1436912/1436912 bytes
I (74594) esp_image: segment 0: paddr=001a0020 vaddr=3c110020 size=4367ch (276092) map
I (74644) esp_image: segment 1: paddr=001e36a4 vaddr=3fc96800 size=04f00h ( 20224) 
I (74654) esp_image: segment 2: paddr=001e85ac vaddr=40374000 size=07a6ch ( 31340) 
I (74654) esp_image: segment 3: paddr=001f0020 vaddr=42000020 size=103f44h (1064772) map
I (74864) esp_image: segment 4: paddr=002f3f6c vaddr=4037ba6c size=0ad60h ( 44384) 
I (74884) esp_image: segment 0: paddr=001a0020 vaddr=3c110020 size=4367ch (276092) map
I (74934) esp_image: segment 1: paddr=001e36a4 vaddr=3fc96800 size=04f00h ( 20224) 
I (74944) esp_image: segment 2: paddr=001e85ac vaddr=40374000 size=07a6ch ( 31340) 
I (74944) esp_image: segment 3: paddr=001f0020 vaddr=42000020 size=103f44h (1064772) map
I (75154) esp_image: segment 4: paddr=002f3f6c vaddr=4037ba6c size=0ad60h ( 44384) 
Done
I (75344) wifi:state: run -> init (0)
I (75344) wifi:pm stop, total sleep time: 804250 us / 71086740 us

I (75354) wifi:<ba-del>idx:0, tid:0
I (75354) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
                                                                           I (75364) wifi:flush txq
I (75364) wifi:stop sw txq
I (75364) wifi:lmac stop hw txq
I (75374) wifi:Deinit lldesc rx mblock:10
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375971
0x40375971: esp_restart_noos at /home/fhfs/Repos/idf50/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3820,len:0x1718
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbec
```

FHFS, 2023-08-24:
> Can you give me some more information to reproduce the issue? The information needed are as follows:
>
>     1. idf commit ID and esp-thread-br commit ID.
>     2. The modification of the codes(such as [...]

1. idf commit sha: `02f5b9a8982db2d59b593fab4e01ecffd35cfdff`, esp-thread-br commit sha: `7fd29e78ca2713c7e7e09a96d60820a0dc6d66ff`
2. Only changes in paritions.csv
3. After a fullclean I do not do a set target. I use [this sdkconfig file](https://pastebin.com/WsbRDUKz)

FHFS, 2023-08-28:
> I increased the stack of the task to 10240, and now it does continue a little bit further. See log:

I think here the esp32 did restart but the monitor had some problems reattaching.

gytxxsy, 2023-09-08:
I've tried the configuration you said above, but still can't reproduce the problem. I suggest you use our latest [ESP-THREAD-BR Release v1.0-beta ](https://github.com/espressif/esp-thread-br/releases/tag/v1.0-beta) to test it (with adding the partitions.csv changes I told you about). Also,  do you use the ESP Thread Border Router board v1.2 for test? If you are not using it but other hardware devices, please let me know and I will do analysis from this aspect. Thanks.

gytxxsy, 2023-09-08:
@FHFS In addition, basic_thread_border_router is just an example, so its codes are not unchangeable. If you can solve the current problem by modifying the task stack size, you can modify it directly and continue your development.

chshu, 2023-09-08:
No SDK change required, closing.

closed 2023-09-08 by chshu