espressif/esp-thread-br #25: Failed to find valid route for Mesh Local address

FHFS, 2023-07-27:
Hello, 

I've built and I am running the esp-thread-br project. I am able to connect to the local network with the nrf52840. 

I am trying to resolve a mDNS advertisment by calling `otDnsClientBrowse`, but I get a timeout with some debug lines saying "Failed to find valid route" for the mesh local address of the esp-thread-br. On the esp-thread-br debug logs I get some dropped udp packets with bad checksums.

When I then use the CLI to create a new dataset and set all the network credentials using:
```
thread stop
ifconfig down
dataset init new
dataset networkname NEST-PAN-5961
dataset panid 0x5961
dataset extpanid 1952473fb4d78dda
dataset networkkey ab11344873cc3fc3889ab62784e9e05d
dataset channel 19
dataset commit active
ifconfig up
thread start
```
Then the nrf52840 is able to get `otDnsClientGetDefaultConfig` and resolve the mDNS advertisment through SRP using `otDnsClientBrowse` and all routes are ok, no more errors missing routes or bad checksums. 

jwhui at OpenThread noticed the Mesh Local address was not correct(contained DEADBEEF). I've had this [discussion on OpenThread](https://github.com/openthread/openthread/discussions/9313) 

I've set the network credentials using `CONFIG_OPENTHREAD_NETWORK_` in sdkconfig. With `CONFIG_OPENTHREAD_BR_AUTO_START` enabled.
**Am I configuring things incorrectly, or is the Mesh Local not being initialized properly?**

zwx1995esp, 2023-07-27:
Hi, FHFS, thanks for issuing that, will reprodunce it on our board and do some further investigations.

zwx1995esp, 2023-07-27:
Hi @FHFS  Could you please share more details about your issue? It might be really helpful for us to reproduce this issue quickly. 
So for your issues, the first time, you used the auto start mode to enable a BR, and then connect the nrf52840 as a child and then you run the command `dns browse XXX` to browse some service might be published on the other host, and it can not browse while the nrf52840 print some logs `Failed to find valid route` like you said. Right?
If yes, I have a question about that: did you get the dataset from the br after the BR start and lead a thread network?(may be using `dataset active -x` to get the data set and then set the dataset to nrf52840)?

For the twice, you start a br using a manually way, at this time, the nrf52840 can browse the service.
and you still using the `dataset active -x` to get the data set and then set the dataset to nrf52840?

FHFS, 2023-07-27:
> If yes, I have a question about that: did you get the dataset from the br after the BR start and lead a thread network?(may be using dataset active -x to get the data set and then set the dataset to nrf52840)?

No, the credentials are given to the nrf52840 through defines. Like:
```
CONFIG_OPENTHREAD_NETWORKKEY="ab:11:34:48:73:cc:3f:c3:88:9a:b6:27:84:e9:e0:5d"
CONFIG_OPENTHREAD_NETWORK_NAME="NEST-PAN-5961"
CONFIG_OPENTHREAD_PANID=22881
CONFIG_OPENTHREAD_CHANNEL=19
CONFIG_OPENTHREAD_XPANID="19:52:47:3f:b4:d7:8d:da"
```

chshu, 2023-07-27:
@FHFS Could you read the dataset from both `esp-thread-br` and `nrf52840` using the command `dataset active`? We want to confirm whether the two are exactly the same.

FHFS, 2023-07-27:
esp-thread-br dataset:
```
0e080000000000010000000300001302081952473fb4d78dda0510ab11344873cc3fc3889ab62784e9e05d030d4e4553542d50414e2d35393631010259610410104810e2315100afd6bc9215a6bfac53
```

nrf52840 dataset:
```
00030000130102596102081952473fb4d78dda0510ab11344873cc3fc3889ab62784e9e05d030d4e4553542d50414e2d353936310708fd8bbdd9ef0b000004104decbca9a85f790b8eb574cd8315bc8a0c0402a0f77835060004001fffe00e080000639b0571825f
```

esp-thread-br dataset after dataset init new:
```
> dataset active -x
00030000130102596102081952473fb4d78dda0510ab11344873cc3fc3889ab62784e9e05d030d4e4553542d50414e2d353936310708fd8bbdd9ef0b000004104decbca9a85f790b8eb574cd8315bc8a0c0402a0f77835060004001fffe00e080000639b0571825f
```

zwx1995esp, 2023-07-28:
Hi, @FHFS  already reproduce this issue on our board, and this issue might be related to some security policy mismatch, so all the coap message can not ne received by BR, I will do more investigation ASAP(maybe next week), and sync the info to you soon.

zwx1995esp, 2023-07-28:
Another tips, if you want to run our BR example and play some features related to the BR, you can try with this:
when BR form a network, using this command to get the whole raw dataset
```
> dataset active -x
```

and set this dataset to the nrf52840 board using this command:
```
> dataset set active <dataset>
```
and on the nrf52840 board enable the thread network, this method can make sure that the nrf52840 board can join the BR network. And you can do some test or play some features using our BR.

FHFS, 2023-07-28:
@zwx1995esp I am running the nrf52840 without CLI. Configuring the network through `dataset init new` on the esp-thread-br works perfectly for me. OpenThread is working perfectly now. I just have some trouble with doing a OTA update on the esp-thread-br, see #24. I can connect with my nrf52840, resolve mDNS through SRP and reach the local network.

zwx1995esp, 2023-07-31:
@FHFS  Yes, copy that, I will have a look at this ASAP.

zwx1995esp, 2023-08-02:
[test.zip](https://github.com/espressif/esp-thread-br/files/12237562/test.zip)
Please try this patch, thsi patch also support config the mesh local prefix via menuconfig.

First unzip the `test.zip`

and then apply the patch on your local idf branch

```
cd <your-idf-path>
git apply test.patch
```

FHFS, 2023-08-02:
This works perfectly. When I was using `otDatasetCreateNewNetwork` to initialize the dataset, the dataset changed after a openthread device joined the network. Now it is correct from the start.

zwx1995esp, 2023-08-02:
I will tell you when this feature is merged, so you can use this patch first, the release version will come soon, just wait the merging process.

chshu, 2023-09-08:
@FHFS The change has been merged to IDF: https://github.com/espressif/esp-idf/commit/8eff9df2b073ba6c87de1d733742569f7f61ac0e

closed 2023-09-08 by chshu