espressif/esp-thread-br #26: Build error when enabling "OPENTHREAD_BR_AUTO_START" option

DrewOJensen, 2023-07-28:
I'm new to all of this so please forgive my ignorance.

When I perform a clean install of esp-idf & esp-thread-br, both contained in the same folder parent folder, I am able to build with the the default config setup. Here's  "ESP Thread Border Router Example" config menu option
[image]

Again, everything is default, no changes to the install yet. I then added the SSID & PSWD for the Wifi, and rebuilt no problem. I then enabled "OPENTHREAD_BR_AUTO_START" and when I attempt to build, I run into issues. 
[image]

Face value says it's an implicit declaration issue but not knowing any details of this setup, I wouldn't be surprised if it's something I'm possibly missing in the setup. As I mentioned earlier, I've already set the SSID & PSWD, and tested it previously on the first default build manually setting up wifi. Here is the output during the failing build, starting with the 1196 build steps.

`[1172/1196] Building C object esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.objFAILED: esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj 
/home/crazyharry/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1-dirty\" -DMD5_ENABLED=1 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DOPENTHREAD_FTD=1 -DSERIAL_FLASHER_BOOT_HOLD_TIME_MS=50 -DSERIAL_FLASHER_RESET_HOLD_TIME_MS=100 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/config -I/home/crazyharry/esp/esp-thread-br/examples/common/thread_border_router/include -I/home/crazyharry/esp/esp-idf/components/newlib/platform_include -I/home/crazyharry/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/crazyharry/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/crazyharry/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/crazyharry/esp/esp-idf/components/freertos/esp_additions/include -I/home/crazyharry/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/home/crazyharry/esp/esp-idf/components/esp_hw_support/include -I/home/crazyharry/esp/esp-idf/components/esp_hw_support/include/soc -I/home/crazyharry/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/crazyharry/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/crazyharry/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/crazyharry/esp/esp-idf/components/heap/include -I/home/crazyharry/esp/esp-idf/components/log/include -I/home/crazyharry/esp/esp-idf/components/soc/include -I/home/crazyharry/esp/esp-idf/components/soc/esp32s3 -I/home/crazyharry/esp/esp-idf/components/soc/esp32s3/include -I/home/crazyharry/esp/esp-idf/components/hal/esp32s3/include -I/home/crazyharry/esp/esp-idf/components/hal/include -I/home/crazyharry/esp/esp-idf/components/hal/platform_port/include -I/home/crazyharry/esp/esp-idf/components/esp_rom/include -I/home/crazyharry/esp/esp-idf/components/esp_rom/include/esp32s3 -I/home/crazyharry/esp/esp-idf/components/esp_rom/esp32s3 -I/home/crazyharry/esp/esp-idf/components/esp_common/include -I/home/crazyharry/esp/esp-idf/components/esp_system/include -I/home/crazyharry/esp/esp-idf/components/esp_system/port/soc -I/home/crazyharry/esp/esp-idf/components/esp_system/port/include/private -I/home/crazyharry/esp/esp-idf/components/xtensa/include -I/home/crazyharry/esp/esp-idf/components/xtensa/esp32s3/include -I/home/crazyharry/esp/esp-idf/components/lwip/include -I/home/crazyharry/esp/esp-idf/components/lwip/include/apps -I/home/crazyharry/esp/esp-idf/components/lwip/include/apps/sntp -I/home/crazyharry/esp/esp-idf/components/lwip/lwip/src/include -I/home/crazyharry/esp/esp-idf/components/lwip/port/include -I/home/crazyharry/esp/esp-idf/components/lwip/port/freertos/include -I/home/crazyharry/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/crazyharry/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/include -I/home/crazyharry/esp/esp-idf/components/openthread/include -I/home/crazyharry/esp/esp-idf/components/openthread/openthread/include -I/home/crazyharry/esp/esp-idf/components/esp_netif/include -I/home/crazyharry/esp/esp-idf/components/esp_event/include -I/home/crazyharry/esp/esp-idf/components/driver/include -I/home/crazyharry/esp/esp-idf/components/driver/deprecated -I/home/crazyharry/esp/esp-idf/components/driver/analog_comparator/include -I/home/crazyharry/esp/esp-idf/components/driver/dac/include -I/home/crazyharry/esp/esp-idf/components/driver/gpio/include -I/home/crazyharry/esp/esp-idf/components/driver/gptimer/include -I/home/crazyharry/esp/esp-idf/components/driver/i2c/include -I/home/crazyharry/esp/esp-idf/components/driver/i2s/include -I/home/crazyharry/esp/esp-idf/components/driver/ledc/include -I/home/crazyharry/esp/esp-idf/components/driver/mcpwm/include -I/home/crazyharry/esp/esp-idf/components/driver/parlio/include -I/home/crazyharry/esp/esp-idf/components/driver/pcnt/include -I/home/crazyharry/esp/esp-idf/components/driver/rmt/include -I/home/crazyharry/esp/esp-idf/components/driver/sdio_slave/include -I/home/crazyharry/esp/esp-idf/components/driver/sdmmc/include -I/home/crazyharry/esp/esp-idf/components/driver/sigma_delta/include -I/home/crazyharry/esp/esp-idf/components/driver/spi/include -I/home/crazyharry/esp/esp-idf/components/driver/temperature_sensor/include -I/home/crazyharry/esp/esp-idf/components/driver/touch_sensor/include -I/home/crazyharry/esp/esp-idf/components/driver/twai/include -I/home/crazyharry/esp/esp-idf/components/driver/uart/include -I/home/crazyharry/esp/esp-idf/components/driver/usb_serial_jtag/include -I/home/crazyharry/esp/esp-idf/components/driver/touch_sensor/esp32s3/include -I/home/crazyharry/esp/esp-idf/components/esp_pm/include -I/home/crazyharry/esp/esp-idf/components/esp_ringbuf/include -I/home/crazyharry/esp/esp-idf/examples/common_components/iperf/include -I/home/crazyharry/esp/esp-idf/components/esp_wifi/include -I/home/crazyharry/esp/esp-idf/components/esp_wifi/wifi_apps/include -I/home/crazyharry/esp/esp-idf/components/esp_phy/include -I/home/crazyharry/esp/esp-idf/components/esp_phy/esp32s3/include -I/home/crazyharry/esp/esp-idf/components/http_parser -I/home/crazyharry/esp/esp-idf/components/esp_http_client/include -I/home/crazyharry/esp/esp-thread-br/components/esp_br_http_ota/include -I/home/crazyharry/esp/esp-idf/components/app_update/include -I/home/crazyharry/esp/esp-idf/components/bootloader_support/include -I/home/crazyharry/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/crazyharry/esp/esp-idf/components/esp_app_format/include -I/home/crazyharry/esp/esp-idf/components/esp_partition/include -I/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_rcp_update/include -I/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp-serial-flasher/include -I/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp-serial-flasher/port -I/home/crazyharry/esp/esp-idf/components/nvs_flash/include -I/home/crazyharry/esp/esp-idf/components/spi_flash/include -I/home/crazyharry/esp/esp-idf/components/vfs/include -I/home/crazyharry/esp/esp-thread-br/examples/common/border_router_board/include -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router=. -fmacro-prefix-map=/home/crazyharry/esp/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj -MF esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj.d -o esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_rou [... cut ...]
[1177/1196] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_91684 and /home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_91684
`

I also get an error when I try to build with the web server GUI enabled, 
[image]
but first things first, I'd like to resolve this auto start issue first.

gytxxsy, 2023-07-28:
It seems that function "esp_openthread_auto_start" was not found when building the app. You can delete the folder "basic_thread_border_router/build" and try again. If the problem still cannot be resolved, it's a good idea to check if you use the latest commit.

Referenc commits: 
esp-idf(releasev5.1): 79b1379662b547f6eb0e5fed33df70587b03d99c
esp-thread-br(latest): 7fd29e78ca2713c7e7e09a96d60820a0dc6d66ff

DrewOJensen, 2023-07-28:
Thanks for the advice to delete the build. That didn't work so I deleted both the IDF and thread folder and redownloaded the latest which got it working for the auto start. I am however still dealing with failing when enabling the web server option as I mentioned at the end of my previous post. This time I'm getting undefined references. Any ideas?

`-- Build files have been written to: /home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/bootloader
[110/111] Generating binary image from built executable
esptool.py v4.6.2
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/bootloader/bootloader.bin
[111/111] cd /home/crazyharry/esp/esp-thread-br/examples/basic_thread_bord...ead-br/examples/basic_thread_border_router/build/bootloader/bootloader.bi
Bootloader binary size 0x51d0 bytes. 0x2e30 bytes (36%) free.
[1258/1261] Linking CXX executable esp_ot_br.elf
FAILED: esp_ot_br.elf 
: && /home/crazyharry/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/esp_ot_br.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld CMakeFiles/esp_ot_br.elf.dir/project_elf_src_esp32s3.c.obj -o esp_ot_br.elf -L/home/crazyharry/esp/esp-idf/components/soc/esp32s3/ld   -L/home/crazyharry/esp/esp-idf/components/esp_rom/esp32s3/ld   -L/home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_system/ld   -L/home/crazyharry/esp/esp-idf/components/esp_phy/lib/esp32s3   -L/home/crazyharry/esp/esp-idf/components/esp_wifi/lib/esp32s3 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/openthread/libopenthread.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__esp-serial-flasher/libespressif__esp-serial-flasher.a  esp-idf/iperf/libiperf.a  esp-idf/espressif__esp_ot_cli_extension/libespressif__esp_ot_cli_extension.a  esp-idf/espressif__esp_rcp_update/libespressif__esp_rcp_update.a  esp-idf/espressif__mdns/libespressif__mdns.a  esp-idf/main/libmain.a  esp-idf/border_router_board/libborder_router_board.a  esp-idf/thread_border_router/libthread_border_router.a  esp-idf/esp_br_http_ota/libesp_br_http_ota.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/esp_ot_br_server/libesp_ot_br_server.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/perfmon/libperfmon.a  esp-idf/touch_element/libtouch_element.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/espressif__esp_ot_cli_extension/libespressif__esp_ot_cli_extension.a  esp-idf/iperf/libiperf.a  esp- [... cut ...]
/home/crazyharry/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/crazyharry/esp/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:623: undefined reference to `otCommissionerAddJoiner'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_143218 and /home/crazyharry/esp/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_143218`

gytxxsy, 2023-07-31:
I think it's necessary to open the `joiner` and `commissioner` via `idf.py menuconfig`. The path is as follows.

Component config > OPenThread > Enable Commissioner
Component config > OPenThread > Enable Joiner

[image]

[image]

[image]

DrewOJensen, 2023-08-01:
Hot dang, thanks for pointing those out. I'll have to dig through to documentation to understand all of those component options. I had wrongly assumed that by enabling the web server feature, that it would automatically enabled any dependencies within the config. I was wrong.

Thanks again for all your help.

closed 2023-08-01 by DrewOJensen