espressif/esp-thread-br #31: Cannot build with web server component

jonny190, 2023-08-15:
When building with the web server compolent enabled i get 

`[100%] Linking CXX executable esp_ot_br.elf
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_join_network_request+0x64): undefined reference to `otJoinerStart'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x34): undefined reference to `otCommissionerGetState'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x38): undefined reference to `otCommissionerStart'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x3c): undefined reference to `otCommissionerAddJoinerWithDiscerner'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj):(.literal.handle_openthread_network_commission_request+0x40): undefined reference to `otCommissionerAddJoiner'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj): in function `handle_openthread_join_network_request':
/home/jonny/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:490: undefined reference to `otJoinerStart'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/esp_ot_br_server/libesp_ot_br_server.a(esp_br_web_api.c.obj): in function `handle_openthread_network_commission_request':
/home/jonny/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:623: undefined reference to `otCommissionerGetState'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/jonny/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:628: undefined reference to `otCommissionerStart'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/jonny/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:642: undefined reference to `otCommissionerAddJoinerWithDiscerner'
/root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/jonny/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:645: undefined reference to `otCommissionerAddJoiner'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/esp_ot_br.elf.dir/build.make:570: esp_ot_br.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:2185: CMakeFiles/esp_ot_br.elf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2, output of the command is in the /home/jonny/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_71955 and /home/jonny/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_71955`

zwx1995esp, 2023-08-16:
You need enable the `commissioner` and `joiner` feature via `idf.py menuconfig`, (Top) → Component config → OpenThread → `Enable Commissioner` and `Enable Joiner`

jonny190, 2023-08-16:
That'd do it thanks

closed 2023-08-16 by jonny190