espressif/esp-thread-br #33: build error : Enable the automatic start mode in Thread Border Router.Enable the web server in Thread Border Router.

MaplestoryAlen, 2023-08-23:
I have purchased a development board, but there was a compilation error when enabling these two features. The log is as follows

Enable the automatic start mode in Thread Border Router 
Output Log：

Building C object esp-idf/thread_border_r...read_border_router.dir/src/border_router_launch.c.objFAILED: esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj 
ccache /home/alen/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1-rc1-8-gea5e0ff298-dirty\" -DMD5_ENABLED=1 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DOPENTHREAD_FTD=1 -DSERIAL_FLASHER_BOOT_HOLD_TIME_MS=50 -DSERIAL_FLASHER_RESET_HOLD_TIME_MS=100 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -Iconfig -I/home/alen/esp-thread-br/examples/common/thread_border_router/include -I/home/alen/esp-idf/components/newlib/platform_include -I/home/alen/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/alen/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/alen/esp-idf/components/freertos/esp_additions/include/freertos -I/home/alen/esp-idf/components/freertos/esp_additions/include -I/home/alen/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/home/alen/esp-idf/components/esp_hw_support/include -I/home/alen/esp-idf/components/esp_hw_support/include/soc -I/home/alen/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/alen/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/alen/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/alen/esp-idf/components/heap/include -I/home/alen/esp-idf/components/log/include -I/home/alen/esp-idf/components/soc/include -I/home/alen/esp-idf/components/soc/esp32s3 -I/home/alen/esp-idf/components/soc/esp32s3/include -I/home/alen/esp-idf/components/hal/esp32s3/include -I/home/alen/esp-idf/components/hal/include -I/home/alen/esp-idf/components/hal/platform_port/include -I/home/alen/esp-idf/components/esp_rom/include -I/home/alen/esp-idf/components/esp_rom/include/esp32s3 -I/home/alen/esp-idf/components/esp_rom/esp32s3 -I/home/alen/esp-idf/components/esp_common/include -I/home/alen/esp-idf/components/esp_system/include -I/home/alen/esp-idf/components/esp_system/port/soc -I/home/alen/esp-idf/components/esp_system/port/include/private -I/home/alen/esp-idf/components/xtensa/include -I/home/alen/esp-idf/components/xtensa/esp32s3/include -I/home/alen/esp-idf/components/lwip/include -I/home/alen/esp-idf/components/lwip/include/apps -I/home/alen/esp-idf/components/lwip/include/apps/sntp -I/home/alen/esp-idf/components/lwip/lwip/src/include -I/home/alen/esp-idf/components/lwip/port/include -I/home/alen/esp-idf/components/lwip/port/freertos/include -I/home/alen/esp-idf/components/lwip/port/esp32xx/include -I/home/alen/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/alen/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/include -I/home/alen/esp-idf/components/openthread/include -I/home/alen/esp-idf/components/openthread/openthread/include -I/home/alen/esp-idf/components/esp_netif/include -I/home/alen/esp-idf/components/esp_event/include -I/home/alen/esp-idf/components/driver/include -I/home/alen/esp-idf/components/driver/deprecated -I/home/alen/esp-idf/components/driver/analog_comparator/include -I/home/alen/esp-idf/components/driver/dac/include -I/home/alen/esp-idf/components/driver/gpio/include -I/home/alen/esp-idf/components/driver/gptimer/include -I/home/alen/esp-idf/components/driver/i2c/include -I/home/alen/esp-idf/components/driver/i2s/include -I/home/alen/esp-idf/components/driver/ledc/include -I/home/alen/esp-idf/components/driver/mcpwm/include -I/home/alen/esp-idf/components/driver/parlio/include -I/home/alen/esp-idf/components/driver/pcnt/include -I/home/alen/esp-idf/components/driver/rmt/include -I/home/alen/esp-idf/components/driver/sdio_slave/include -I/home/alen/esp-idf/components/driver/sdmmc/include -I/home/alen/esp-idf/components/driver/sigma_delta/include -I/home/alen/esp-idf/components/driver/spi/include -I/home/alen/esp-idf/components/driver/temperature_sensor/include -I/home/alen/esp-idf/components/driver/touch_sensor/include -I/home/alen/esp-idf/components/driver/twai/include -I/home/alen/esp-idf/components/driver/uart/include -I/home/alen/esp-idf/components/driver/usb_serial_jtag/include -I/home/alen/esp-idf/components/driver/touch_sensor/esp32s3/include -I/home/alen/esp-idf/components/esp_pm/include -I/home/alen/esp-idf/components/esp_ringbuf/include -I/home/alen/esp-idf/examples/common_components/iperf/include -I/home/alen/esp-idf/components/esp_wifi/include -I/home/alen/esp-idf/components/esp_wifi/wifi_apps/include -I/home/alen/esp-idf/components/esp_phy/include -I/home/alen/esp-idf/components/esp_phy/esp32s3/include -I/home/alen/esp-idf/components/http_parser -I/home/alen/esp-idf/components/esp_http_client/include -I/home/alen/esp-thread-br/components/esp_br_http_ota/include -I/home/alen/esp-idf/components/app_update/include -I/home/alen/esp-idf/components/bootloader_support/include -I/home/alen/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/alen/esp-idf/components/esp_app_format/include -I/home/alen/esp-idf/components/esp_partition/include -I/home/alen/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_rcp_update/include -I/home/alen/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp-serial-flasher/include -I/home/alen/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp-serial-flasher/port -I/home/alen/esp-idf/components/nvs_flash/include -I/home/alen/esp-idf/components/spi_flash/include -I/home/alen/esp-idf/components/vfs/include -I/home/alen/esp-thread-br/examples/common/border_router_board/include -mlongcalls    -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/home/alen/esp-thread-br/examples/basic_thread_border_router=. -fmacro-prefix-map=/home/alen/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj -MF esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj.d -o esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj   -c /home/alen/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c
In file included from /home/alen/esp-idf/components/openthread/include/esp_openthread.h:9,
                 from /home/alen/esp-thread-br/examples/common/thread_border_router/include/border_router_launch.h:14,
                 from /home/alen/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:8:
/home/alen/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c: In function 'ot_task_worker':
/home/alen/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:111:21: error: implicit declaration of function 'esp_openthread_auto_start'; did you mean 'esp_openthread_get_netif'? [-Werror=implicit-function-declaration]
  111 |     ESP_ERROR_CHECK(esp_openthread_auto_start(NULL));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/alen/esp-idf/components/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
cc1: some warnings being treated as errors
[1183/1195] Building C object esp-idf/protocol_exampl...__idf_protocol_examples_common.dir/wifi_connect.c.objninja: build stopped: subcommand failed.

Enable the web server in Thread Border Router.
Output Log：
[1194/1196] Linking CXX executable esp_ot_br.elfFAILED: esp_ot_br.elf 
: && /home/alen/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++  -mlongcalls  -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=/home/alen/esp-thread-br [... cut ...]
/home/alen/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/alen/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:620: undefined reference to `otCommissionerAddJoinerWithDiscerner'
/home/alen/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/alen/esp-thread-br/components/esp_ot_br_server/src/esp_br_web_api.c:623: undefined reference to `otCommissionerAddJoiner'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

zwx1995esp, 2023-08-23:
Hi, could you please check which idf commit hash you used? Maybe you need to checkout to `release/v5.1` branch.

For `undefined reference to otCommissionerAddJoiner` you need to enable the configuration `ot commissioner` via `idf.py menuconfig` and select this configuration:

(Top) → Component config → OpenThread 
`Enable Commissioner`
`Enable Joiner`

zwx1995esp, 2023-08-23:
Also please check if you used the latest main branch of `esp-thread-br`?

MaplestoryAlen, 2023-08-24:
> Also please check if you used the latest main branch of ?`esp-thread-br`

Using the ea5e0ff branch of version 5.1

MaplestoryAlen, 2023-08-24:
> Hi, could you please check which idf commit hash you used? Maybe you need to checkout to branch.`release/v5.1`
> 
> For you need to enable the configuration via and select this configuration:`undefined reference to otCommissionerAddJoiner``ot commissioner``idf.py menuconfig`
> 
> (Top) → Component config → OpenThread `Enable Commissioner` `Enable Joiner`
Okay, thank you

chshu, 2023-09-08:
@MaplestoryAlen Closing, let us know if any other issues.

closed 2023-09-08 by chshu