espressif/esp-thread-br #35: spiffs_image does not exist

alexyuoouy, 2023-10-09:
Hello everyone, I recently encountered a problem. During compilation, it prompts that "spiffs_image" is missing. I don't know how this is generated, and I can't find similar issues online. 
```
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
Processing 5 dependencies:
[1/5] espressif/esp-serial-flasher (0.0.11)
[2/5] espressif/esp_ot_cli_extension (1.0.1)
[3/5] espressif/esp_rcp_update (1.0.1)
[4/5] espressif/mdns (1.2.1)
[5/5] idf (5.1.1)
-- Project sdkconfig file C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/sdkconfig
Loading defaults file C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/sdkconfig.defaults...
Compiler supported targets: xtensa-esp32s3-elf

-- App "esp_ot_br" version: v1.0-beta-3-g568221c
-- Adding linker script C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Users/wangyu/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support border_router_board bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_br_http_ota esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_ot_br_server esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-serial-flasher espressif__esp_ot_cli_extension espressif__esp_rcp_update espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 iperf json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport thread_border_router touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/wangyu/esp/esp-idf/components/app_trace C:/Users/wangyu/esp/esp-idf/components/app_update C:/Users/wangyu/esp/esp-idf/components/bootloader C:/Users/wangyu/esp/esp-idf/components/bootloader_support C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/common/border_router_board C:/Users/wangyu/esp/esp-idf/components/bt C:/Users/wangyu/esp/esp-idf/components/cmock C:/Users/wangyu/esp/esp-idf/components/console C:/Users/wangyu/esp/esp-idf/components/cxx C:/Users/wangyu/esp/esp-idf/components/driver C:/Users/wangyu/esp/esp-idf/components/efuse C:/Users/wangyu/esp/esp-idf/components/esp-tls C:/Users/wangyu/esp/esp-idf/components/esp_adc C:/Users/wangyu/esp/esp-idf/components/esp_app_format C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/components/esp_br_http_ota C:/Users/wangyu/esp/esp-idf/components/esp_coex C:/Users/wangyu/esp/esp-idf/components/esp_common C:/Users/wangyu/esp/esp-idf/components/esp_eth C:/Users/wangyu/esp/esp-idf/components/esp_event C:/Users/wangyu/esp/esp-idf/components/esp_gdbstub C:/Users/wangyu/esp/esp-idf/components/esp_hid C:/Users/wangyu/esp/esp-idf/components/esp_http_client C:/Users/wangyu/esp/esp-idf/components/esp_http_server C:/Users/wangyu/esp/esp-idf/components/esp_https_ota C:/Users/wangyu/esp/esp-idf/components/esp_https_server C:/Users/wangyu/esp/esp-idf/components/esp_hw_support C:/Users/wangyu/esp/esp-idf/components/esp_lcd C:/Users/wangyu/esp/esp-idf/components/esp_local_ctrl C:/Users/wangyu/esp/esp-idf/components/esp_mm C:/Users/wangyu/esp/esp-idf/components/esp_netif C:/Users/wangyu/esp/esp-idf/components/esp_netif_stack C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/components/esp_ot_br_server C:/Users/wangyu/esp/esp-idf/components/esp_partition C:/Users/wangyu/esp/esp-idf/components/esp_phy C:/Users/wangyu/esp/esp-idf/components/esp_pm C:/Users/wangyu/esp/esp-idf/components/esp_psram C:/Users/wangyu/esp/esp-idf/components/esp_ringbuf C:/Users/wangyu/esp/esp-idf/components/esp_rom C:/Users/wangyu/esp/esp-idf/components/esp_system C:/Users/wangyu/esp/esp-idf/components/esp_timer C:/Users/wangyu/esp/esp-idf/components/esp_wifi C:/Users/wangyu/esp/esp-idf/components/espcoredump C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp-serial-flasher C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_rcp_update C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__mdns C:/Users/wangyu/esp/esp-idf/components/esptool_py C:/Users/wangyu/esp/esp-idf/components/fatfs C:/Users/wangyu/esp/esp-idf/components/freertos C:/Users/wangyu/esp/esp-idf/components/hal C:/Users/wangyu/esp/esp-idf/components/heap C:/Users/wangyu/esp/esp-idf/components/http_parser C:/Users/wangyu/esp/esp-idf/components/idf_test C:/Users/wangyu/esp/esp-idf/components/ieee802154 C:/Users/wangyu/esp/esp-idf/examples/common_components/iperf C:/Users/wangyu/esp/esp-idf/components/json C:/Users/wangyu/esp/esp-idf/components/log C:/Users/wangyu/esp/esp-idf/components/lwip C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/main C:/Users/wangyu/esp/esp-idf/components/mbedtls C:/Users/wangyu/esp/esp-idf/components/mqtt C:/Users/wangyu/esp/esp-idf/components/newlib C:/Users/wangyu/esp/esp-idf/components/nvs_flash C:/Users/wangyu/esp/esp-idf/components/openthread C:/Users/wangyu/esp/esp-idf/components/partition_table C:/Users/wangyu/esp/esp-idf/components/perfmon C:/Users/wangyu/esp/esp-idf/components/protobuf-c C:/Users/wangyu/esp/esp-idf/examples/common_components/protocol_examples_common C:/Users/wangyu/esp/esp-idf/components/protocomm C:/Users/wangyu/esp/esp-idf/components/pthread C:/Users/wangyu/esp/esp-idf/components/sdmmc C:/Users/wangyu/esp/esp-idf/components/soc C:/Users/wangyu/esp/esp-idf/components/spi_flash C:/Users/wangyu/esp/esp-idf/components/spiffs C:/Users/wangyu/esp/esp-idf/components/tcp_transport C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/common/thread_border_router C:/Users/wangyu/esp/esp-idf/components/touch_element C:/Users/wangyu/esp/esp-idf/components/ulp C:/Users/wangyu/esp/esp-idf/components/unity C:/Users/wangyu/esp/esp-idf/components/usb C:/Users/wangyu/esp/esp-idf/components/vfs C:/Users/wangyu/esp/esp-idf/components/wear_levelling C:/Users/wangyu/esp/esp-idf/components/wifi_provisioning C:/Users/wangyu/esp/esp-idf/components/wpa_supplicant C:/Users/wangyu/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/build
[266/581] cmd.exe /C "cd /D C:\Users\wangyu\Documents\dev\idf-proj\esp-thread-br\examples\basic_thread_border_router\build\esp-idf\main && C:...sp-thread-br/examples/basic_thread_border_router/build/rcp_fw.bin --page-size=256 --obj-name-len=32 --meta-len=4 --use-magic --use-magic-len"
FAILED: esp-idf/main/CMakeFiles/spiffs_rcp_fw_bin C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/main/CMakeFiles/spiffs_rcp_fw_bin
cmd.exe /C "cd /D C:\Users\wangyu\Documents\dev\idf-proj\esp-thread-br\examples\basic_thread_border_router\build\esp-idf\main && C:\Users\wangyu\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/wangyu/esp/esp-idf/components/spiffs/spiffsgen.py 0xa0000 C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/main/spiffs_image C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/exa [... cut ...]
[1/1] cmd.exe /C "cd /D C:\Users\wangyu\Documents\dev\idf-proj\esp-thread-br\examples\basic_thread_border_router\build\bootloader\esp-idf\esptool_py && C:\Users\wangyu\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Users/wangyu/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/wangyu/Documents/dev/idf-proj/esp-thread-br/examples/basic_thread_border_router/build/bootloader/bootloader.bin"
Bootloader binary size 0x5120 bytes. 0x2ee0 bytes (37%) free.
ninja: build stopped: subcommand failed.
```

zwx1995esp, 2023-10-10:
The `spiffs_image` is created by the script `create_ota_image.py` under the directory `esp-thread-br/examples/basic_thread_border_router`, the related command is in the file [`main/CMakeLists.txt`](https://github.com/espressif/esp-thread-br/blob/main/examples/basic_thread_border_router/main/CMakeLists.txt#L8), so could you please run the generating command manually? 

```
cd <your-local-esp-thread-br-path>/example/basic_thread_border_router

python create_ota_image.py --rcp-build-dir $IDF_PATH/examples/openthread/ot_rcp/build --target-file build/esp-idf/main/spiffs_image/ot_rcp_0/rcp_image
```
to see if there is a file under the directory `example/basic_thread_border_router/build/esp-idf/main/spiffs_image/ot_rcp_0/rcp_image`

alexyuoouy, 2023-10-10:
You are right. Manually executing that command did generate the file. Now the compilation is successful. There might be some reason why this command was not executed during the compilation process. Thank you very much.

alexyuoouy, 2023-10-10:
> The `spiffs_image` is created by the script `create_ota_image.py` under the directory `esp-thread-br/examples/basic_thread_border_router`, the related command is in the file [...]

You are right. Manually executing that command did generate the file. Now the compilation is successful. There might be some reason why this command was not executed during the compilation process. Thank you very much.

alexyuoouy, 2023-10-10:
I found the root cause of the problem. Since I am using the Windows 11 system, the default program to open .py files is not the Python interpreter. Therefore, that line in the CMake file won't be executed correctly.

igrr, 2023-10-10:
@zwx1995esp I think the fix for this would be to update this part:

https://github.com/espressif/esp-thread-br/blob/89fade55f23bb0ef3bf07dbdc24d4fd634d92dde/examples/basic_thread_border_router/main/CMakeLists.txt#L7-L8

like this:
```cmake
    idf_build_get_property(python PYTHON)
    add_custom_target(rcp_image_generation ALL
        COMMAND ${python} ${CMAKE_CURRENT_SOURCE_DIR}/../create_ota_image.py
```

zwx1995esp, 2023-10-11:
Thanks for that Ivan, I will fix it ASAP.

zwx1995esp, 2023-10-11:
I have already created the fix MR in our internal branch, will merge it ASAP. Thanks for your issue @alexyuoouy !

closed 2023-10-17 by espressif-bot