espressif/esp-thread-br #36: Thread communication with external networks is unstable

alexyuoouy, 2023-10-15:
Hello,I've encountered another issue. I've been continuously pinging a node from my computer, and it intermittently connects and disconnects. When it disconnects, it shows a 'Request Timed Out' message. Here's the current network topology: I have a router to which both my computer and OTBR are connected. The router supports IPv6 and does not use DHCPv6. Both my computer and OTBR have generated IPv6 addresses based on the prefix assigned by my ISP. I also have an ESP32-H2-DevKitM-1, which was added as a node to the Thread network. Initially, it acted as a child node, but later it automatically switched to a router role. Currently, I'm using this router node to ping my computer, and it works sometimes. However, when I try to ping this node from my computer, it occasionally succeeds, but I'm encountering the issue I mentioned earlier. I've set the OTBR log level to 'debug,' and from the logs, it seems that OTBR is constantly switching its routing strategy. This might be due to the presence of two IPv6 prefixes in my network – one assigned by the ISP and another by OTBR. This could lead to continuous switching between active and inactive routes, resulting in the issue mentioned above. Here are some log entries.

The routing strategy has changed, and the source address has been modified：
```
D(314305) OPENTHREAD:[D] Platform------: Received spinel frame, flg:0x2, tid:0, cmd:PROP_VALUE_IS, key:STREAM_RAW, len:89, rssi:-12 ...
D(314305) OPENTHREAD:[D] Platform------: ... noise:-128, flags:0x0000, channel:26, lqi:10, timestamp:3737310850, rxerr:0
D(314305) OPENTHREAD:[D] Mac-----------: Received frame from short address 0x4400
D(314305) OPENTHREAD:[D] Mac-----------: Rx security - frame counter 20701
D(314305) OPENTHREAD:[D] Mac-----------: ==============================[RX len=089]==============================
D(314305) OPENTHREAD:[D] Mac-----------: | 69 98 AF 86 FC 00 48 00 | 44 0D DD 50 00 00 01 7B | i./.|.H.D.]P...{
D(314305) OPENTHREAD:[D] Mac-----------: | D0 10 3A C8 7D 86 16 53 | E3 CB B3 FD B5 0B 79 B3 | P.:H}..ScK3}5.y3
D(314305) OPENTHREAD:[D] Mac-----------: | D8 AC 57 C9 C7 74 99 67 | A1 59 28 81 00 78 FA 00 | X,WIGt.g!Y(..xz.
D(314305) OPENTHREAD:[D] Mac-----------: | 01 07 5B 61 62 63 64 65 | 66 67 68 69 6A 6B 6C 6D | ..[abcdefghijklm
D(314305) OPENTHREAD:[D] Mac-----------: | 6E 6F 70 71 72 73 74 75 | 76 77 61 62 63 64 65 66 | nopqrstuvwabcdef
D(314305) OPENTHREAD:[D] Mac-----------: | 67 68 69 BD 12 F7 47 F4 | 0A .. .. .. .. .. .. .. | ghi=.wGt........
D(314305) OPENTHREAD:[D] Mac-----------: ------------------------------------------------------------------------
I(314315) OPENTHREAD:[I] MeshForwarder-: Received IPv6 ICMP6 msg, len:80, chksum:78fa, ecn:no, from:0x4400, sec:yes, prio:normal, rss:-12.0
I(314315) OPENTHREAD:[I] MeshForwarder-:     src:[fdb5:7e4f:e7:1:c87d:8616:53e3:cbb3]
I(314315) OPENTHREAD:[I] MeshForwarder-:     dst:[fdb5:b79:b3d8:ac57:c9c7:7499:67a1:5928]
D(314315) OPENTHREAD:[D] Mac-----------: Idle mode: Radio receiving on channel 26
D (314595) OPENTHREAD: Millisecond timer alarm start running, t0=314591, dt=661
I(314595) OPENTHREAD:[I] RoutingManager: Evaluating routing policy
I(314595) OPENTHREAD:[I] RoutingManager: Found a favored on-link prefix 2408:8215:512a:d6e0::/64
I(314595) OPENTHREAD:[I] RoutingManager: Local on-link prefix state: Advertising -> Deprecating (fdb5:b79:b3d8:ac57::/64)
I(314595) OPENTHREAD:[I] RoutingManager: Evaluating NAT64 prefix
I(314595) OPENTHREAD:[I] RoutingManager: Preparing RA
I(314595) OPENTHREAD:[I] RoutingManager: - PIO fdb5:b79:b3d8:ac57::/64 (valid:1796, preferred:0)
I(314595) OPENTHREAD:[I] RoutingManager: - RIO fdb5:7e4f:e7:1::/64 (lifetime:1800, prf:medium)
I (314595) OPENTHREAD: Received PIO
I(314595) OPENTHREAD:[I] RoutingManager: Sent RA on infra netif 2
D(314595) OPENTHREAD:[D] RoutingManager: =============[[BR-CERT] direction=send | type=RA | len=064]=============
D(314595) OPENTHREAD:[D] RoutingManager: | 86 00 00 00 00 00 00 00 | 00 00 00 00 00 00 00 00 | ................
D(314605) OPENTHREAD:[D] RoutingManager: | 03 04 40 C0 00 00 07 04 | 00 00 00 00 00 00 00 00 | ..@@............
D(314605) OPENTHREAD:[D] RoutingManager: | FD B5 0B 79 B3 D8 AC 57 | 00 00 00 00 00 00 00 00 | }5.y3X,W........
D(314605) OPENTHREAD:[D] RoutingManager: | 18 02 40 00 00 00 07 08 | FD B5 7E 4F 00 E7 00 01 | ..@.....}5~O.g..
D(314605) OPENTHREAD:[D] RoutingManager: ------------------------------------------------------------------------
I(314605) OPENTHREAD:[I] RoutingManager: Will evaluate routing policy in 04:23.327 (263327 msec)
D (314605) OPENTHREAD: Millisecond timer alarm fired
I(314605) OPENTHREAD:[I] RoutingManager: Received RA from fe80:0:0:0:4a27:e2ff:fe14:43c4 on infra netif 2
D(314605) OPENTHREAD:[D] RoutingManager: =============[[BR-CERT] direction=recv | type=RA | len=064]=============
D(314605) OPENTHREAD:[D] RoutingManager: | 86 00 7C E0 00 00 00 00 | 00 00 00 00 00 00 00 00 | ..|`............
D(314605) OPENTHREAD:[D] RoutingManager: | 03 04 40 C0 00 00 07 04 | 00 00 00 00 00 00 00 00 | ..@@............
D(314605) OPENTHREAD:[D] RoutingManager: | FD B5 0B 79 B3 D8 AC 57 | 00 00 00 00 00 00 00 00 | }5.y3X,W........
D(314605) OPENTHREAD:[D] RoutingManager: | 18 02 40 00 00 00 07 08 | FD B5 7E 4F 00 E7 00 01 | ..@.....}5~O.g..
D(314605) OPENTHREAD:[D] RoutingManager: ------------------------------------------------------------------------
I(314605) OPENTHREAD:[I] RoutingManager: - RA Header - default route - lifetime:0
I(314615) OPENTHREAD:[I] RoutingManager: - PIO fdb5:b79:b3d8:ac57::/64 (valid:1796, preferred:0)
D(314615) OPENTHREAD:[D] RoutingManager: Prefix stale timer scheduled in 1799612 ms
D (314725) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcbe1ec
V (314725) esp_netif_lwip: esp_netif_get_ip6_linklocal esp-netif:0x3fcbe1ec
V (314725) esp_netif_lwip: esp_netif_get_ip6_linklocal esp-netif:0x3fcbe1ec
D (314725) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcbe1ec
D (314725) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcbe1ec
V (314725) esp_netif_lwip: esp_netif_get_ip6_linklocal esp-netif:0x3fcbe1ec
V (314725) esp_netif_lwip: esp_netif_get_ip6_linklocal esp-netif:0x3fcbe1ec
D (314725) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fcbe1ec
D (315265) OPENTHREAD: Millisecond timer alarm start running, t0=315261, dt=7547
D (315265) OPENTHREAD: Millisecond timer alarm start running, t0=315262, dt=993
D (315265) OPENTHREAD: Millisecond timer alarm fired
D(315295) OPENTHREAD:[D] Mac-----------: Request to start operation "TransmitDataDirect"
D(315295) OPENTHREAD:[D] Mac-----------: Starting operation "TransmitDataDirect"
D(315295) OPENTHREAD:[D] SubMac--------: RadioState: Receive -> CsmaBackoff
D(315295) OPENTHREAD:[D] SubMac--------: RadioState: CsmaBackoff -> Transmit
D (315295) OPENTHREAD: sent radio frame
D(315295) OPENTHREAD:[D] Platform------: Sent spinel frame, flg:0x2, tid:10, cmd:PROP_VALUE_SET, key:STREAM_RAW, len:90, channel:26, maxbackoffs:4, maxretries:15 ...
D(315295) OPENTHREAD:[D] Platform------: ... csmaCaEnabled:1, isHeaderUpdated:0, isARetx:0, skipAes:0, txDelay:0, txDelayBase:0
D (315315) OPENTHREAD: radio uart read event
D (315315) OPENTHREAD: received hdlc radio frame
D(315315) OPENTHREAD:[D] Platform------: Received spinel frame, flg:0x2, tid:10, cmd:PROP_VALUE_IS, key:LAST_STATUS, status:OK
D(315315) OPENTHREAD:[D] SubMac--------: RadioState: Transmit -> Receive
[... 179 lines cut ...]
D(600365) OPENTHREAD:[D] Mac-----------: Finishing operation "TransmitDataDirect"
I(600365) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 ICMP6 msg, len:80, chksum:795f, ecn:no, to:0x4400, sec:yes, prio:normal
I(600365) OPENTHREAD:[I] MeshForwarder-:     src:[fdb5:b79:b3d8:ac57:c9c7:7499:67a1:5928]
I(600365) OPENTHREAD:[I] MeshForwarder-:     dst:[fdb5:7e4f:e7:1:c87d:8616:53e3:cbb3]
D(600365) OPENTHREAD:[D] Mac-----------: Idle mode: Radio receiving on channel 26
D (600375) OPENTHREAD: radio uart read event
D (600375) OPENTHREAD: received hdlc radio frame
D(600375) OPENTHREAD:[D] Platform------: Received spinel frame, flg:0x2, tid:0, cmd:PROP_VALUE_IS, key:STREAM_RAW, len:89, rssi:-12 ...
D(600375) OPENTHREAD:[D] Platform------: ... noise:-128, flags:0x0000, channel:26, lqi:10, timestamp:4023388389, rxerr:0
D(600375) OPENTHREAD:[D] Mac-----------: Received frame from short address 0x4400
D(600375) OPENTHREAD:[D] Mac-----------: Rx security - frame counter 20837
[... cut ...]
D(600385) OPENTHREAD:[D] Mac-----------: ------------------------------------------------------------------------
I(600385) OPENTHREAD:[I] MeshForwarder-: Received IPv6 ICMP6 msg, len:80, chksum:785f, ecn:no, from:0x4400, sec:yes, prio:normal, rss:-12.0
I(600385) OPENTHREAD:[I] MeshForwarder-:     src:[fdb5:7e4f:e7:1:c87d:8616:53e3:cbb3]
I(600385) OPENTHREAD:[I] MeshForwarder-:     dst:[fdb5:b79:b3d8:ac57:c9c7:7499:67a1:5928]
D(600395) OPENTHREAD:[D] Mac-----------: Idle mode: Radio receiving on channel 26
``` 
I don't know how to configure OTBR or my router to solve this problem.

alexyuoouy, 2023-10-15:
英语很烂，不知道有没有表达清楚，使用中文说一遍：我使用我的电脑一直在ping 一个node，时而通时而断，断的时候提示请求超时。我现在的网络拓扑是这样，我有一个路由器，我的电脑和otbr都连接到了这个路由器，这个路由器支持ipv6，并且没有使用dhcpv6，我的电脑和otbr都生成了运营商分配的ipv6前缀的地址，我手里还有一个ESP32-H2-DevKitM-1，他作为一个node添加进了thread网络，起初他的角色是child，后来自动切换成了router。现在我是用这个作为router的node去ping我的电脑，可以ping通，使用电脑去ping这个node，有时也能ping通，但是会出现我上面说的现象。我将otbr的日志等级设置到了debug，通过日志我看到了otbr似乎一直在切换它的路由策略，因为我现在的网络中存在了两个ipv6前缀，一个是运营商给我分配的，另一个是otbr的，这可能导致某个有效的路由一直在被弃用和被启用之间切换，从而导致了上面的现象。

zwx1995esp, 2023-10-16:
问下，你有wireshark可以抓包嘛？可以抓一下 ping 过程中，时而通时而断的时候wireshark的包吗？流程如下，
1.先用电脑连上路由器，然后打印一下电脑的ip地址
2.BR连上路由器
3.Node连上路由器
4.在node上 输入 ipaddr 把结果记录一下
5.在电脑上打印电脑的所有ip地址和ip路由(windows 输入 route)
6.电脑上打开wireshark，监听连接到路由器的interface
7.电脑开始ping node，复现一下时断时续的问题，然后保存一下wireshark抓的包。

alexyuoouy, 2023-10-16:
> 问下，你有wireshark可以抓包嘛？可以抓一下 ping 过程中，时而通时而断的时候wireshark的包吗？流程如下， 1.先用电脑连上路由器，然后打印一下电脑的ip地址 2.BR连上路由器 3.Node连上路由器 4.在node上 输入 ipaddr 把结果记录一下 5.在电脑上打印电脑的所有ip地址和ip路由(windows 输入 route) 6.电脑上打开wireshark，监听连接到路由器的interface 7.电脑开始ping node，复现一下时断时续的问题，然后保存一下wireshark抓的包。

由于抓包日志里带了太多信息，是否可以提供一个邮箱或者其它联系方式单独发给你。

zwx1995esp, 2023-10-17:
公司邮箱附件大小上限是5M，你可以直接打一个压缩包，上传到github这个上面就可以(把压缩包拖到comment框里面就行)。

zwx1995esp, 2023-10-18:
请尝试修改这个代码:
$IDF_PATH/components/lwip/lwip/src/core/ipv6/nd6.c line 549:
https://github.com/espressif/esp-lwip/blob/5c4f899f8e7b58ef53522b6f832fc940e29bf0b9/src/core/ipv6/nd6.c#L549
```
   nd6_send_na(inp, &target_address, ND6_FLAG_SOLICITED | ND6_FLAG_OVERRIDE);
```

改为这个:

```
  nd6_send_na(inp, &target_address, ND6_FLAG_ROUTER | ND6_FLAG_SOLICITED | ND6_FLAG_OVERRIDE);
```

alexyuoouy, 2023-10-18:
好的，非常感谢，我会试一下，我还有个问题请教一下，如果node使用了matter（比如esp-matter里的light例程），该怎样连接br和电脑端的chip-tool通信？

chshu, 2023-10-18:
br 不需要做任何改动，只要电脑和 br 能ping 通，chip-tool 就可以和 br 通信。

在用 chip-tool 给 Thread 设备配网时，`operationalDataset` 是从 BR 读出来的 dataset:
```
dataset active -x
```

chip-tool 指令参考
https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html#commissioning

alexyuoouy, 2023-10-19:
> 请尝试修改这个代码: $IDF_PATH/components/lwip/lwip/src/core/ipv6/nd6.c line 549: https://github.com/espressif/esp-lwip/blob/5c4f899f8e7b58ef53522b6f832fc940e29bf0b9/src/core/ipv6/nd6.c#L549
> 
> ```
>    nd6_send_na(inp, &target_address, ND6_FLAG_SOLICITED | ND6_FLAG_OVERRIDE);
> ```
> 
> 改为这个:
> 
> ```
>   nd6_send_na(inp, &target_address, ND6_FLAG_ROUTER | ND6_FLAG_SOLICITED | ND6_FLAG_OVERRIDE);
> ```
> 
> 应该会对你的问题有帮助

我连续ping了近一个小时，没有再出现请求超时的情况，这个更改是否考虑提交到esp-idf中？

alexyuoouy, 2023-10-19:
> br 不需要做任何改动，只要电脑和 br 能ping 通，chip-tool 就可以和 br 通信。
> 
> 在用 chip-tool 给 Thread 设备配网时，`operationalDataset` 是从 BR 读出来的 dataset:
> 
> ```
> dataset active -x
> ```
> 
> chip-tool 指令参考 https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html#commissioning

这个文档描述的好像是通过ble给节点配网，那是否我的电脑还需要一个ble-dongle，这个ble-dongle有什么要求吗，笔记本电脑带的蓝牙行不行？我可以通过手机给matter over thread的节点配网吗，通过二维，使用哪个app？

chshu, 2023-10-19:
笔记本带的蓝牙是可以的。

chip-tool 是 matter 日常开发用的工具，也可以用几个生态的 手机 APP (Google Home, Apple Homekit, Amazon Alexa)，需要搭配生态的音箱作为 BR 才能用。

zwx1995esp, 2023-10-19:
> 我连续ping了近一个小时，没有再出现请求超时的情况，这个更改是否考虑提交到esp-idf中？

这个我们会改到idf中。

alexyuoouy, 2023-10-19:
OK，非常感谢，问题都被解决了，closed

alexyuoouy, 2023-10-23:
感谢各位解答，没有更多问题了。

closed 2023-10-23 by chshu