espressif/esp-thread-br #39: OT CLI extension can't handle WiFi PSK with spaces

gregbreen, 2023-11-01:
The OT CLI extension can't correctly parse a PSK with spaces. Even if the PSK is wrapped in quotes. It's because it parses as arguments: https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_wifi_cmd.c#L211

zwx1995esp, 2023-11-01:
Hi, if you want to input a space as a character of the PSK, you can use `\` to escape it. For example, if your PSK is 
```
espressif  123
```
then you can input 
```
wifi connect -s myssid -p espressif\ \ 123
```

zwx1995esp, 2023-11-02:
> Hi, if you want to input a space as a character of the PSK, you can use `\` to escape it. For example, if your PSK is
> 
> ```
> espressif  123
> ```
> 
> then you can input
> 
> ```
> wifi connect -s myssid -p espressif\ \ 123
> ```

@gregbreen Hi, I believe this can help you with your issue, so, could you please confirm that and if your issue is solved, could you please close this issue?

closed 2023-11-13 by chshu

gregbreen, 2024-01-02:
Ah OK thank you.