espressif/esp-thread-br #41: Feature try-out exploration [Matter controller + Border Router]

supersunil, 2023-11-15:
Hello there,
I would like to combine the [esp-matter-controller](https://github.com/espressif/esp-matter/tree/main/examples/controller) with the [esp-thread-br](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router) as a single application.

What is the suggested way of getting this started?

By the way, are there any internal development for the same planned already?

Curious to know.

chshu, 2023-11-16:
We have an example to integrate Thread BR and matter controller, the code is under internal review. I will update here once it's available on Github.

supersunil, 2023-11-16:
@chshu - Thanks for your response. Would you mind asking the ETA?

chshu, 2023-11-19:
> @chshu - Thanks for your response. Would you mind asking the ETA?

It will be available in the next week.

chshu, 2023-11-24:
The Thread BR and Matter controller integration example is now available: https://github.com/espressif/esp-matter/tree/main/examples/controller, please refer to this commit https://github.com/espressif/esp-matter/commit/fd033440c7a9147f45800eb9d2e7e4c9e258e46a for the major change.

supersunil, 2023-11-28:
Hey @chshu - Thanks for sharing this. After having a look, my first question is, are the pairing functionalities (pairing ble-wifi and pairing ble-thread) been purged in this release?

If so, this integration needs additional device acting as Matter controller for commissioning things to the network?

supersunil, 2023-11-28:
Yeah, if so, we need the `chip-tool` running, right? I'm looking for getting the controller with commissioning ability running combined with `esp-thread-br`, how can I do it?

chshu, 2023-11-28:
> Yeah, if so, we need the `chip-tool` running, right? I'm looking for getting the controller with commissioning ability running combined with `esp-thread-br`, how can I do it?

The commission command is also supported: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#pairing-commands

But note that it uses the local test PAA for device attestation, it's only an experimental feature, can't be used for production.

Commissioner is normally the Phone APP, and need some cloud based solution for NOC signing.