espressif/esp-thread-br #42: Multicast forwarding

briansuyu, 2023-11-20:
Hi, during testing the basic_thread_boarder_router exsample, i was stucked doing the multicast forwarding codelab, 

after using mcast commond, my esp32boarder router and other 2 esp32h2 FTD all joined ff04::123 multicast group
[image]
[image]

but when try to ping ff04::123 address from outside windows machine it doesn't response (the windwos route table seems alright, and ping each FTD with there ULA address works)
[image]

can i ask am i missing any config steps to make this work? Should i add some code to register the netif: en interface to the multicast group as well? (i guess my esp32 boarder router didn't receive the ff04::123 address multicast data because it didn't register it in the outside network?)

zwx1995esp, 2023-11-20:
Could you please set this command to BR and to see the BBR is enabled or not?

`> bbr`

briansuyu, 2023-11-21:
> Could you please set this command to BR and to see the BBR is enabled or not?
> 
> `> bbr`

not enabled
[image]

briansuyu, 2023-11-21:
after enabling bbr, the multicast forwarding works now. thanks

briansuyu, 2023-12-25:
hi, about the multicast forwarding,  some other issue comes out when enabling bbr and the multicast not working again. The code was not changed, only some modification in menuconfig about the dataset. Here is the error i see
[image]
[image]
Can i ask what possible reason could cause this error? I tried clone a completely new clean esp br code and compiled again still the same

zwx1995esp, 2023-12-26:
Could you please enlarge the configuration to `32` and try again?

(Top) → Component config → mDNS → (16) Maximum actions pending to the server

zwx1995esp, 2023-12-26:
Also you need to set the `mcast join ff04::123` on a non-BR device. You can flash and run the example `$ESP_IDF/examples/openthread/ot_cli` on an ESP32H2/ESP32C6 and join the network formed by BR, then use this command to listen the multicast address on this device.

zwx1995esp, 2023-12-26:
I guess you want to try the multicast forwarding feature of the ESP BR, so you need to try like the previous comment.

But if you just want to let the BR join the multicast listenning group on the WiFi netif and ping the BR from host, you can just use the lwip address adding command like this `ip add st ff04::123`(BTW, you can use the lwip address show command `ip print` to log the address allocated on the BR, you can confirm the multicast address does register on the ST netif) then you can ping this multicast address from HOST.

briansuyu, 2023-12-26:
yes i was trying the multicast forwarding feature. I cleaned everything and tried again with the ot_cli exsample on a h2 mini devboard and increased the mDNS max actions pending size on the esp32 br. Device joined the network formed by the br and can be pined from linux host, bbr enabled, but multicast still fails.
[image]
here is the log on br side.
[image]
[image]

but i can ping the multicast address from windows powershell
[image]

both machines and br connected to the same switch and router. 
Can i ask is there anything configuration i should set on the linux host?

zwx1995esp, 2023-12-26:
Yes, for the linux host, I think you can have try with set the `ttl` 64(also named hop limit), some linux host will set 1 as the ttl default value, and maybe you need also specify the net interface in the ping command on the linux host. You can have a try with this command on linux:

```
ping ff04::123 -I <interface name> -t 64
```

Tips: As for the interface name you can obtain it using the command `ifconfig`.

briansuyu, 2023-12-26:
thanks a lot, with the ttl setting everything finalling works. Thanks

closed 2023-12-26 by briansuyu

zwx1995esp, 2023-12-26:
Great! Actually, in our docs, we have already recommend to set the ttl `64` when using a multicast address as a ping dest address. you can refer to [here](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/multicast_forwarding.html#reaching-the-multicast-group-from-the-wi-fi-network-via-icmp)