espressif/esp-thread-br #43: Enable TREL from menuconfig

opieters, 2023-11-24:
Hi

I'm using this project as a base to make a network extender of the thread network with multiple partitions. According to Thread 1.2, the border routers of both partitions should communicate over TREL. In the build log, I see that the TREL-related libraries are compiled, but the `trel` command is not available in the CLI. Moreover, in menuconfig there is no options to enable/disable TREL. 

The `sdkconfig.defaults` does contain code to able TREL (`CONFIG_OPENTHREAD_TREL=y`), but this flag is removed in `sdkconfig` by the build system.

Is the omission of TREL intentional? Can I easily add it as an additional option to the build system? So far, I've been unsuccessful to get it working locally. Thanks!

zwx1995esp, 2023-11-24:
Sorry, we does not support this feature now, this configuration is a remained one which we add the supports for the legacy version of TREL. On the removing legacy TREL feature PR, we forget to remove this config in the sdkconfig.defaults.

opieters, 2023-11-24:
What would be needed to add this feature?

zwx1995esp, 2023-11-28:
Hi, a new MDNS feature is needed. For the esp-mdns, it does not support a listening mode, just like `avahi-browse -a`. We have already sync this feature requirement to MDNS group, and it is under developing.

chshu, 2023-11-28:
@opieters Want to note that even without TREL, the Thread devices in two separate partitions could still communicate with each other, as long as the Thread BRs are in the same local network, it can be achieved by the Thread 1.3.0 Border Router.

TREL is a feature defined in Thread 1.2, but it's not a certification ready feature in Thread 1.3.0 yet. The BR certification in Thread 1.3.0 mainly covers the Matter application requirements, and ESP Thread BR solution is already Thread 1.3.0 certificated component.

Could you let us know your use case and why TREL is required?

opieters, 2023-11-28:
Thank you for the update. We are currently using TREL to extend the coverage of our Thread network using several embedded-linux devices. These devices contain significant processing power, which is not required to simply extend the network (as the density of connected devices is quite low). Consequently, we are evaluating the use of an ESP-based BR to extend the network, in a more cost-effective way.

Since the current implementation already uses TREL, we would initially look into this type of connection to have a more consistent structure / simplify the software.

The devices use a propriety protocol, so we are not looking into matter at the moment.

gytxxsy, 2024-11-13:
Currently, the TREL feature is supported and an example is provided:[ot_trel](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_trel).

The OpenThread border router also supports the TREL feature, and you can enable it as follows:
```
idf.py menuconfig → Component config → OpenThread → Enable Thread Radio Encapsulation Link (TREL)
```

To verify the TREL functionality, you can use ESP devices to form the following topology:

```
ESP32-S3(run ot_trel example)
               |
           (Wi-Fi link)
               |
Border Router Board(run border router example)----(15.4 link)---ESP32-H2(run ot_cli example)
```

The ot_cli device can send messages to the ot_trel device. The related examples are as follows:
* [ot_trel](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_trel)
* [ot_cli](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli)
* [border router](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router)

It should be noted that there is currently a bug in mDNS that affects the Border Router's browse functionality. To work around this issue, you need to first start the Border Router and connect it to a Wi-Fi network, then start other OT TREL devices. The fix for this bug has already been merged into the main branch and will take effect in the next mDNS version update. Please refer to [here](https://github.com/espressif/esp-protocols/pull/661).

chshu, 2024-12-24:
The TREL codelab is available here: https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/trel.html

closed 2024-12-24 by chshu