espressif/esp-thread-br #44: Using SPI for RCP communication crashes the host

theomeyer-tado, 2023-11-24:
I'm trying to use SPI for Host<>RCP communication but it seems to somehow crash/hang the Thread Border Router firmware on the host. The logs indicate that the RCP is reset successfully, so I expect the connection and the communication between the S3 and the H2 over SPI to be ok. When using UART everything is working properly.

The crash do not reset the S3 : the prompt for the OpenThread CLI is printed and then I can't send anything and there is no output anymore. I checked the signal on the U0RXD line of the S3, and when sending data the voltages does not drop down to 0V so I suspect the S3 is trying to actually drive this pin (it should not). The SPI pins are correctly configured (and not using the U0RXD pin obviously).

I'm using ESP-IDF v5.1.1 and this esp-thread-br [commit ](https://github.com/espressif/esp-thread-br/commit/160d31d7608416b56d4f4c08d15de7f57a81cd16). Both ot_rcp and esp-thread-br are configured for SPI RCP communication using `CONFIG_OPENTHREAD_RCP_SPI` in ot_rcp and `CONFIG_OPENTHREAD_RADIO_SPINEL_SPI` in esp-thread-br config. Hardware is a custom board based on the Espressif ESP-Thread BR&Zigbee GW rev1.1, with slight pinout adjustment.

My updated sdkconfig.default : 
```
# ESP32-S3 is the host SoC on ESP-Thread-Border-Router board
CONFIG_IDF_TARGET="esp32s3"

#
# libsodium
#
CONFIG_LIBSODIUM_USE_MBEDTLS_SHA=y
# end of libsodium

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# end of Serial flasher config

#
# Partition Table
#
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000
CONFIG_PARTITION_TABLE_MD5=y
# end of Partition Table

#
# Compiler options
#
CONFIG_COMPILER_OPTIMIZATION_SIZE=y

#
# Newlib
#
CONFIG_NEWLIB_NANO_FORMAT=y

#
# mbedTLS
#
CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_SSL_PROTO_DTLS=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECJPAKE=y
# end of TLS Key Exchange Methods
CONFIG_MBEDTLS_ECJPAKE_C=y
# end of mbedTLS

#
# OpenThread
#
CONFIG_OPENTHREAD_ENABLED=y
CONFIG_OPENTHREAD_BORDER_ROUTER=y
CONFIG_OPENTHREAD_TREL=y
CONFIG_OPENTHREAD_CLI_OTA=y
# end of OpenThread

#
# OpenThread Border Router Example
#
CONFIG_OPENTHREAD_BR_AUTO_UPDATE_RCP=y
CONFIG_OPENTHREAD_BR_START_WEB=y
#
# External coexist wire type and pin config
#
CONFIG_ESP_WIFI_EXTERNAL_COEXIST_ENABLE=y
CONFIG_EXTERNAL_COEX_ENABLE=y
CONFIG_EXTERNAL_COEX_WIRE_TYPE=2
CONFIG_EXTERNAL_COEX_REQUEST_PIN=13
CONFIG_EXTERNAL_COEX_GRANT_PIN=14
CONFIG_EXTERNAL_COEX_PRIORITY_PIN=12
# end of External coexist wire type and pin config
# end of OpenThread Border Router Example

#
# lwIP
#
CONFIG_LWIP_IPV6_FORWARD=y
CONFIG_LWIP_IPV6_NUM_ADDRESSES=8
CONFIG_LWIP_MULTICAST_PING=y
CONFIG_LWIP_NETIF_STATUS_CALLBACK=y
CONFIG_LWIP_HOOK_IP6_ROUTE_DEFAULT=y
CONFIG_LWIP_HOOK_ND6_GET_GW_DEFAULT=y
CONFIG_LWIP_HOOK_IP6_INPUT_CUSTOM=y
CONFIG_LWIP_IPV6_AUTOCONFIG=y
CONFIG_LWIP_HOOK_IP6_SELECT_SRC_ADDR_CUSTOM=y
# end of lwIP

#
# mDNS
#
CONFIG_MDNS_STRICT_MODE=y
CONFIG_MDNS_MULTIPLE_INSTANCE=y
# end of mDNS

#
# ESP System Settings
#
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=3584
# end of ESP System Settings

#
# PHY
#
# No need for USB
CONFIG_ESP_PHY_ENABLE_USB=n
# end of PHY

#
# ESP Border Router Board Kit
#
CONFIG_ESP_BR_BOARD_DEV_KIT=n
# Using standard UART for console
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=n
# end of ESP Border Router Board Kit

#
# Board Configuration
#
CONFIG_PIN_TO_RCP_RESET=48
CONFIG_PIN_TO_RCP_BOOT=11
# TX/RX swapped in esp_ot_config.h file
CONFIG_PIN_TO_RCP_TX=9
CONFIG_PIN_TO_RCP_RX=10
# Use SPI for normal communication with RCP
CONFIG_OPENTHREAD_RADIO_SPINEL_SPI=y
CONFIG_PIN_TO_RCP_SCLK=45
CONFIG_PIN_TO_RCP_MISO=35
CONFIG_PIN_TO_RCP_MOSI=47
CONFIG_PIN_TO_RCP_CS=21
# end of Board Configuration
```

Logs : 

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (27) boot: compile time Nov 24 2023 15:27:52
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (79) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (94) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (102) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (109) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=533b8h (340920) map
I (191) esp_image: segment 1: paddr=000733e0 vaddr=3fc97f00 size=04f4ch ( 20300) load
I (195) esp_image: segment 2: paddr=00078334 vaddr=40374000 size=07ce4h ( 31972) load
I (204) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=f6678h (1009272) map
I (386) esp_image: segment 4: paddr=001766a0 vaddr=4037bce4 size=0c1ach ( 49580) load
I (406) boot: Loaded app from partition at offset 0x20000
I (406) boot: Disabling RNG early entropy source...
I (418) cpu_start: Multicore app
I (418) cpu_start: Pro cpu up.
I (418) cpu_start: Starting app cpu, entry point is 0x4037554c
0x4037554c: call_start_cpu1 at C:/Users/theo.meyer/Documents/Firmware/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (436) cpu_start: Pro cpu start user code
I (436) cpu_start: cpu freq: 160000000 Hz
I (436) cpu_start: Application information:
I (439) cpu_start: Project name:     tr04_esp_thread_br
I (445) cpu_start: App version:      v1.0-beta-16-g160d31d-dirty
I (452) cpu_start: Compile time:     Nov 24 2023 15:31:07
I (458) cpu_start: ELF file SHA256:  8ee48eae8194761f...
I (464) cpu_start: ESP-IDF:          v5.1.1
I (469) cpu_start: Min chip rev:     v0.0
I (473) cpu_start: Max chip rev:     v0.99
I (478) cpu_start: Chip rev:         v0.2
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FCAE800 len 0003AF10 (235 KiB): DRAM
I (496) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (503) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (509) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (516) spi_flash: detected chip: gd
I (520) spi_flash: flash io: dio
W (524) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (537) sleep: Configure to isolate all GPIO pins in sleep state
I (544) sleep: Enable automatic switching of GPIO sleep configuration
I (551) app_start: Starting scheduler on CPU0
I (556) app_start: Starting scheduler on CPU1
I (556) main_task: Started on CPU0
I (566) main_task: Calling app_main()
�������������I (646) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (646) OPENTHREAD: spinel SPI interface initialization completed
I (646) obtr_web: <=======================server start========================>

I (656) obtr_web: http://0.0.0.0:80/index.html

I(666) OPENTHREAD:[I] Platform------: RCP reset: RESET_POWER_ON
I(676) OPENTHREAD:[I] Platform------: Software reset RCP successfully
I (676) obtr_web: <===========================================================>

I (646) main_task: Returned from app_main()
I(756) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I (756) esp_ot_br: Internal RCP Version: openthread-esp32/e088c3766b-5beae1437; esp32h2;  2023-11-24 14:26:22 UTC
I (776) esp_ot_br: Running  RCP Version: openthread-esp32/e088c3766b-5beae1437; esp32h2;  2023-11-24 14:26:22 UTC
I (786) OPENTHREAD: OpenThread attached to netif
> >
```

chshu, 2023-11-26:
@theomeyer-tado would you mind to try it again with IDF v5.1.2, which we have tested recently.

zwx1995esp, 2023-11-27:
@theomeyer-tado Hi, from your description of the stuck issue, I guess it might be the known issue of the USJ(USB-Jtag): when printing the boot logs, any input will make the console stuck. 

One things that I want to confirm from your side: 
1. Which serial tools did you use? I guess it might not be the esp-monitor right?

If you used the esp-monitor, some warning logs might be out:

```
I(630) OPENTHREAD:[I] Settings------: ... pid:0x726a5fb8, mlecntr:0xbd5, maccntr:0xbba, mliid:478e781070658133}
I (650) OPENTHREAD: OpenThread attached to netif
> Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
> Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
```
Could you please try to input anything while the boot log is printing to reproduce this stuck issue? Also try to input nothing to avoid this stuck issue?

theomeyer-tado, 2023-11-27:
@zwx1995esp I'm using UART for the console, not the internal USB/JTAG adapter of the S3 (`CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=n`). In fact I'm disabling the USB PHY completely using `CONFIG_ESP_PHY_ENABLE_USB=n` . Also, I'm using esp-monitor through a call to `esp-idf.py monitor` to get these logs. I'm on Windows 11 and using PowerShell 7 if that matters.

> @theomeyer-tado would you mind to try it again with IDF v5.1.2, which we have tested recently.
@chshu I will try that but would prefer to solve it with v5.1.1 since I plan to include external code later specifically written for this version of IDF

zwx1995esp, 2023-11-28:
@theomeyer-tado Sorry, I used your config and re-flash more than 20 times, and did not reproduce this stuck issue....
Could you please try a little more to find the patterns in reproducing the issue?

theomeyer-tado, 2023-11-28:
> @theomeyer-tado Sorry, I used your config and re-flash more than 20 times, and did not reproduce this stuck issue.... Could you please try a little more to find the patterns in reproducing the issue?

Have you been testing on this particular [commit ](https://github.com/espressif/esp-thread-br/commit/160d31d7608416b56d4f4c08d15de7f57a81cd16)and with esp-idf v5.1.1 ?

After further investigation, it appear that the ESP32 itself does not get stuck I now have another task running and periodically outputting logs which is working as expected. What stops working is the OpenThread CLI prompt though.

zwx1995esp, 2023-11-29:
> Have you been testing on this particular [commit ](https://github.com/espressif/esp-thread-br/commit/160d31d7608416b56d4f4c08d15de7f57a81cd16)and with esp-idf v5.1.1 ?

Yes, I try to reproduce the issue you mentioned at this code base, but finally failed.

> After further investigation, it appear that the ESP32 itself does not get stuck I now have another task running and periodically outputting logs which is working as expected. What stops working is the OpenThread CLI prompt though.

So you mean that the issue might be the cli task is pending and never scheduled? And the only way to reproduce this issue is re-flash the binary file and sometimes it will occur (or maybr it is an inevitable problem)? Do you have any other way to reproduce it?

theomeyer-tado, 2023-11-30:
I tracked down the issue to GPIO0 being set to output level high at some point in the initialization of OpenThread (probably during RCP SPI initialization. In my board GPIO0 is connected to U0RX for entering programming mode using the RX pin : IO0 driven high prevent correct signalling on the RX pin, so the CLI does not receive any character.
GPIO0 should not be touched by the OpenThread code though.