espressif/esp-thread-br #45: BR causing network changes triggering kubernetes pod restarts... Is it running DHCP on WiFi network?

DanTup, 2023-11-24:
I picked up the border router and have been following [these instructions](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html) to set it up. The only config changes I made were to add the WiFi details and enable the web server.

When the device boots, it caused all of the kubernetes pods on one of my servers to reboot 😱

This turns out to be because [kubernetes restarts pods](https://github.com/canonical/microk8s/issues/3575#issuecomment-1503518335) when IP addresses on the host changes. Looking through the config files I found a `CONFIG_LWIP_DHCPS` setting that I thought maybe was causing the BR to run DHCP that might be triggering the network change on my server.

I tried setting `CONFIG_LWIP_DHCPS=n` and re-flashing, but unfortunately this doesn't appear to have made a difference - I'm still seeing the server triggering restarts when the BR boots.

I have some options for changing kubernetes to not restart on these kinds of changes, but I'm trying to understand exactly what the BR is doing that might cause this (just so I have a better understanding of how things work). I expected it to be a DHCP client on my WiFi network, but I didn't expect it to do anything that could affect the network/IPs on other devices on the network.

So I guess I just have some questions:

- What is the BR doing that could change the network/IPs on other devices on the network - is it running DHCP on the WiFi side?
- What exactly does `CONFIG_LWIP_DHCPS` do? I couldn't find any config explaining exactly what DHCP it's running (I expected maybe it will run DHCP on the Thread network, but not WiFi), but setting it to `n` also doesn't appear to have changed anything.

Thank you! :-)

DanTup, 2023-11-25:
Explicitly setting `--bind-address=0.0.0.0` in Kubernetes prevented the pods from restarting when the BR joined the network.

Running `ip address` on the Kubernetes host before/after booting the BR results shows this extra entry showing up when the BR has booted:

[image]

Unfortunately I don't understand what this line means. I don't know if it indicates that the BR is providing something like DHCP, or whether this is completely normal (and Kubernetes restarting pods because of it is overly-aggressive).

Perhaps this is more of a networking/k8s question than BR, but if anyone has any insight I'd love to know. Thanks!

chshu, 2023-11-26:
It's not caused by the DHCP, but the IPv6 ND RA message sent by the Thread BR.

You may refer to the section 9.6.2 in Thread 1.3 specification. The Wi-Fi devices will add an extra IPv6 address based on the PIO from the RA message, and use it to communicate with the Thread devices behind the Thread BR. This is how the Thread BR support the bidirectional IPv6 connectivity.

But I don't understand why it causes the kubernetes pod to restart, maybe some issue in the kubernetes pod itself.

DanTup, 2023-11-26:
@chshu thanks! I can't find a copy of the spec (apparently you have to _request_ it from the Thread website?!). I understand "RA message" as being "Router Advertisement" but not sure what you mean by PIO (I searched but couldn't find much).

Do you mean that the thread BR will essentially assign new IPv6 addresses for all existing devices on my network?

> But I don't understand why it causes the kubernetes pod to restart, maybe some issue in the kubernetes pod itself.

There's a service in k8s (or at least microk8s) that monitors for network changes and triggers a restart. I think it's because by default it listens on all interfaces and any changes to those could affect the pods. I've been able to fix this by having k8s only bind to 0.0.0.0 (which I think means it will only be affected by IPv4 changes and therefore ignores this change). The default behaviour seems a bit agressive to me (especially if whatevever the BR is doing is fairly normal). I might open an issue with Kubernetes once I have a better understanding of exactly what you mean by the above.

Thank you!

DanTup, 2023-11-26:
I missed the "ND" part of your message above and think I understand better now.

I've opened https://github.com/canonical/microk8s/issues/4317 with some details to a) verify whether this is working as intended, and b) have my issue documented somewhere others can find it if they have similar issues.

I don't think there's anything for you to do here, thank you for your responses @chshu :-)

closed 2023-11-26 by DanTup

chshu, 2023-11-27:
@DanTup The Thread 1.3 spec can be downloaded from this link: https://www.threadgroup.org/ThreadSpec.

Feel free to contact us for any other findings and questions.