espressif/esp-thread-br #46: Improve documentation for exactly EXTERNAL_COEX_ENABLE does and when to use it

DanTup, 2023-11-25:
I picked up the [Thread Border Router](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html) board and was following the instructions at https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html

It says:

> Enable EXTERNAL_COEX_ENABLE option if you want to enable the RF External coexistence.
> 
> Note: To enable external coexistence of the Thread Border Router, enable the EXTERNAL_COEX_ENABLE option of $IDF_PATH/examples/openthread/ot_rcp before building the RCP Image.

However, I can't find any information on what this means. My understanding was that the reason for using two boards (as there are on the TBR board) was for coexistance, however it's off by default and the docs don't give any indication of under what circumstances you would/would not want to use this.

Thanks!

chshu, 2023-11-26:
Please refer to this doc for the external coexistence design: https://www.espressif.com.cn/sites/default/files/documentation/external_coexistence_design_en.pdf

The 3-wire coex mechanism is supported in the ESP Thread BR SDK. When enabled, S3 acts as the coex master, H2 acts as the slave, the two will not transmit at the same time to void RF interference. It helps to improve the performance especially when Wi-Fi and 802.15.4 are operating on a close frequency.

The implementation is built in the SDK itself, which work closely with the HW coex module. We don't expect user to modify the detailed logic. Let us know if you have any specific requirement on the coex design.

DanTup, 2023-11-26:
> The 3-wire coex mechanism is supported in the ESP Thread BR SDK. When enabled, S3 acts as the coex master, H2 acts as the slave, the two will not transmit at the same time to void RF interference. [...]

So would you recommend enabling it? It's not clear to me what the trade-off is like between less interference and each radio having to keep pausing for the other.

I think whatever the answer to that question is would be useful to include in the notes at https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html so it's clearer _why_ you might choose to use it when working through the guide :-)

chshu, 2023-11-27:
The coex is not a mandatory feature for a product, the two RFs could anyway work together without enabling coex, since there are two separate RF paths.

If the application could ensure that Wi-Fi and 802.15.4 will operate on different frequency, then Coex is not necessary. But if the two work on a close frequency, then the RF interference is significant, where the coex would be helpful.

Thanks for the suggetion, we will enhance the docs with more description about coex.

DanTup, 2023-11-27:
> If the application could ensure that Wi-Fi and 802.15.4 will operate on different frequency, then Coex is not necessary. But if the two work on a close frequency, then the RF interference is significant, where the coex would be helpful.

Got it, thank you :)

closed 2023-12-08 by espressif-bot