espressif/esp-thread-br #52: How many services can be registered on the esp32 boarder router srp server?

briansuyu, 2023-12-21:
hi, can i ask is there a max number of services that can be registered on the esp32 boarder router as a srp server? And how could i increase it? And could i set up a central SRP server on a linux server for several different thread networks all connected in the same local Ethernet  network? If this can be achieved how should i do that?

zwx1995esp, 2023-12-22:
For your question about the max number of services which can be registered on the BR, the only limitation is the size of the static array which is used for record the MDNS service. You can set this via `idf.py menuconfig`

```
(Top) → Component config → mDNS  →  (10)Max number of services
```

And for your question `And could i set up a central SRP server on a linux server for several different thread networks all connected in the same local Ethernet network? If this can be achieved how should i do that`

Do you want that the host can access the service which is pubilshed on any thread devices? 

If yes, I think currently the implemention of the SRP can reach your demand. Each thread network should have one BR to forward the packets between wifi network and thread network. And the host can access any service published at any thread netwrok. No other things are needed to be done.

briansuyu, 2023-12-22:
ok thanks a lot

closed 2023-12-22 by briansuyu