espressif/esp-thread-br #53: Unable to build basic_thread_border_router

manmountain, 2023-12-27:
I have followed this guide: https://openthread.io/guides/border-router/espressif-esp32 to build basic_thread_border_router example for the official ESP Thread Border Router board but I get an error

`
/home/_USERNAME_/Sources/Repos/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:78:24: error: implicit declaration of function 'ESP_OPENTHREAD_DEFAULT_HOST_CONFIG'; did you mean 'ESP_OPENTHREAD_DEFAULT_PORT_CONFIG'? [-Werror=implicit-function-declaration]
   78 |         .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(),
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                        ESP_OPENTHREAD_DEFAULT_PORT_CONFIG
/home/_USERNAME_/Sources/Repos/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:76:56: error: missing braces around initializer [-Werror=missing-braces]
   76 |     esp_openthread_platform_config_t platform_config = {
      |                                                        ^
/home/_USERNAME_/Sources/Repos/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:76:56: error: missing braces around initializer [-Werror=missing-braces]
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
`

Is there a newer guide or do I miss something in my setup? In `idf.py menuconfig` I'm enabling `ESP Thread Border Router Example > Enable the automatic start mode in Thread Border Router` and set my WIFI credentials in `Example Connection Configuration > connect using Wi-Fi interface`. Is there any other options in `idf.py menuconfig` I'm missing?

manmountain, 2023-12-27:
My ESP Thread Border Router board is version V1.2 if that makes any differences.

zwx1995esp, 2023-12-27:
Ah, you need to use the IDF v5.1.2 branch, I will create a MR to update the instruction of this [set-up-repositories](https://openthread.io/guides/border-router/espressif-esp32#set-up-repositories)

zwx1995esp, 2023-12-27:
Please have a try with the IDF branch `v5.1.2` and the ESP-THREAD-BR branch `v1.0`

manmountain, 2023-12-27:
It worked with the IDF branch `v5.1.2` and the ESP-THREAD-BR branch `v1.0`

closed 2023-12-27 by manmountain

zwx1995esp, 2024-01-02:
@manmountain FYI, the docs updating MR has already been created and waiting for reviewing and merging. Please see https://github.com/openthread/ot-docs/pull/149