espressif/esp-thread-br #55: CAD Model for Border Router

TCWilliamson, 2024-01-22:
Hello,

would it be possible to get a 3D CAD Export for the Border Router? I am designing an enclosure for it.

Thank you!

chshu, 2024-01-23:
The 3D CAD file is not ready yet, will post here once it's available.

TCWilliamson, 2024-04-12:
> The 3D CAD file is not ready yet, will post here once it's available.

Any luck yet? It would help as we need to design a top board to go on where the ethernet plugs in, we want to make sure we align the sizes & pins with our CAD & the new PCB. It is important. We will be looking at several hundred of these units soon.

closed 2024-05-17 by espressif-bot

chshu, 2024-05-17:
@TCWilliamson @martin8310 Please find the CAD files here: https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#cad