espressif/esp-thread-br #56: Build fail for C6

orenc75, 2024-02-05:
Building on ESP32-C6 fails for both esp-idf-v5.1.2 and esp-idf-v5.2-beta2.

For example, on esp-idf-v5.2-beta2, using ESP32-C6, I get the following error:

In file included from C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:17:
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c: In function 'app_main':
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_config.h:54:36: error: 'CONFIG_PIN_TO_RCP_MOSI' undeclared (first use in this function); did you mean 'CONFIG_PIN_TO_RCP_BOOT'?
   54 |                     .mosi_io_num = CONFIG_PIN_TO_RCP_MOSI, \
      |                                    ^~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:77:25: note: in expansion of macro 'ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG'
   77 |         .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_config.h:54:36: note: each undeclared identifier is reported only once for each function it appears in
   54 |                     .mosi_io_num = CONFIG_PIN_TO_RCP_MOSI, \
      |                                    ^~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:77:25: note: in expansion of macro 'ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG'
   77 |         .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_config.h:55:36: error: 'CONFIG_PIN_TO_RCP_MISO' undeclared (first use in this function); did you mean 'CONFIG_PIN_TO_RCP_RX'?
   55 |                     .miso_io_num = CONFIG_PIN_TO_RCP_MISO, \
      |                                    ^~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:77:25: note: in expansion of macro 'ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG'
   77 |         .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_config.h:56:36: error: 'CONFIG_PIN_TO_RCP_SCLK' undeclared (first use in this function); did you mean 'CONFIG_PIN_TO_RCP_RX'?
   56 |                     .sclk_io_num = CONFIG_PIN_TO_RCP_SCLK, \
      |                                    ^~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:77:25: note: in expansion of macro 'ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG'
   77 |         .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_config.h:66:37: error: 'CONFIG_PIN_TO_RCP_CS' undeclared (first use in this function); did you mean 'CONFIG_PIN_TO_RCP_RX'?
   66 |                     .spics_io_num = CONFIG_PIN_TO_RCP_CS,  \
      |                                     ^~~~~~~~~~~~~~~~~~~~
C:/Users/ORENC/Private/Thread/esp-thread-br/examples/basic_thread_border_router/main/esp_ot_br.c:77:25: note: in expansion of macro 'ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG'
   77 |         .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\ORENC\Private\Thread\esp-thread-br\examples\basic_thread_border_router\build\log\idf_py_stderr_output_48496 and C:\Users\ORENC\Private\Thread\esp-thread-br\examples\basic_thread_border_router\build\log\idf_py_stdout_output_48496

chshu, 2024-02-05:
The basic_thread_border_router example in esp-thread-br repo works as is on [ESP Thread BR Board](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#wi-fi-based-thread-border-router), which is ESP32-S3 + ESP32+H2.

Please use [ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) example in esp-idf on ESP32-C6, but the single SoC solution is not recommended for production.

In order to setup the Thread BR with C6 + H2, here are the steps for your reference:

- build and flash [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) to ESP32-H2 devkit
- use [ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) example in esp-idf
  - idf.py set-target esp32c6
  - configure OPENTHREAD_RADIO_TYPE to "Connect via UART" (note: C6 has 802.15.4, so the default is "Native 15.4 radio")
  - update the correct [C6 pin numbers](https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_br/main/esp_ot_config.h#L40) connected to H2's TX and RX
  - build and flash the example to ESP32-C6 devkit
- connect C6 and H2 with three wires (TX, RX, GND), be sure to close the H2 console monitor, since the UART0 is used for RCP UART

Then there should be some output as below:
```
I (611M�I (621) OPENTHREAD: spinel UART interface initialization completed
I (621) main_task: Returned from app_main()
I(621) OPENTHREAD:[I] P-RadioSpinel-: RCP reset: RESET_POWER_ON
I(631) OPENTHREAD:[I] P-RadioSpinel-: Software reset RCP successfully
I(671) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I (671) OPENTHREAD: OpenThread attached to netif
> scan

| PAN  | MAC Address      | Ch | dBm | LQI |
+------+------------------+----+-----+-----+
| b335 | 92eacce0b84594c1 | 11 | -63 |   9 |
| 1a65 | e6bd350d91e0c183 | 13 | -45 |   9 |
| 1a65 | 9eed1d95cf1004b3 | 13 | -44 |   8 |
| 35e9 | 8122334455667705 | 15 | -55 |  10 |
| e2b2 | 0e5ba4c40bf0b6be | 16 | -49 |  10 |
| e2b2 | 360d34996a2c9e34 | 16 | -43 |  10 |
| 4d83 | ba9741cd44273f44 | 18 | -45 |   9 |
| 472f | da75c0910ffbb2a7 | 24 | -53 |  10 |
| 8df9 | c60e461cbc93936c | 25 | -58 |   7 |
| 45f7 | 7e062787e985c640 | 25 | -57 |   9 |
Done
>
```

orenc75, 2024-02-05:
Thanks, I was actually trying to setup it as C6+H2, so I will follow the ot_br example.

Maybe the README.md for this project should be updated since it might be a bit confusion since it state that the project support any ESP32-S & ESP32-C systems:

"Hardware Platforms
Wi-Fi based Thread Border Router
The Wi-Fi based ESP Thread Border Router consists of two SoCs:

- An ESP32 series Wi-Fi SoC (ESP32, **ESP32-C, ESP32-S**, etc) loaded with ESP Thread Border Router and OpenThread Stack.
- An ESP32-H 802.15.4 SoC loaded with OpenThread RCP."

chshu, 2024-02-05:
[ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) example is much easier to get start with. While it's still doable to run the basic_thread_border_router example on C6+H2, just need to connect more GPIO pins as described here: https://github.com/espressif/esp-thread-br?tab=readme-ov-file#standalone-modules.

Your previous compile issue could be resolved by enabling `OPENTHREAD_RADIO_SPINEL_UART` config option.

zwx1995esp, 2024-02-06:
@orenc75 Hi, could you please show us how did you connect the C6 with H2? You said you connect 9 pins, which pins did you use?

orenc75, 2024-02-06:
Hi, I set the following pins:

C6 side    H2 Side
--------    ---------
  GND          G
    19            TX
    18            RX
     7             RST
     8             9
     10           2
     11           3
     12           0
     13           1

I also set in menuconfig the following pins:
ESP Thread Border Router Example -> Board Configuration -> (19) Pin to RCP TX
                                                                                                     (18) Pin to RCP RX
I did not changed the RCP Reset (7) and RCP Boot (8) pins in menuconfig

zwx1995esp, 2024-02-07:
@orenc75 Hi I tried to use the uart 0 for C6 to communicate with H2, and also used the GPIO19 and GPIO18 on c6, all things work fine. And what is the GPIO 10~13 used for in your apps? Did you enable RCP SPI in your project? If you use uart, please check both BR and RCP side, you should enable `OPENTHREAD_RADIO_SPINEL_UART` for BR and `OPENTHREAD_RCP_UART` for RCP.

If you want to use SPI for communication between BR and RCP. You should enable `OPENTHREAD_RADIO_SPINEL_SPI` for BR and `OPENTHREAD_RCP_SPI` for RCP. But notice, for the chapter 7.1 of the [esp32c6 TRM](https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf), it said that `GPIO10 ~ GPIO11 are not led out to any chip pins, thus not available to users`. So GPIO 10 and GPIO 11 should not be used for the user. Since different chips may have some difference, this repo is designed for the [ESP Thread Border Router Board](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#esp-thread-border-router-board), if you want to use C6 for your project, we recommend reading the TRM first before choosing the GPIO you used.

BTY, did you try this example?
> The basic_thread_border_router example in esp-thread-br repo works as is on [ESP Thread BR [...]

orenc75, 2024-02-07:
@zwx1995esp 
Thank you very much for reproducing my setup.  I followed your guidelines and it still not working for me. I got the same error as I got before. I wonder if there is something wrong with my HW or am I missing something in the setup. 

Regarding the setup using ot_br as @chshu wrote. It is not working for me either. I connect the three pins and followed the steps and this is the error I got:
Wrote 1710368 bytes (974093 compressed) at 0x00010000 in 22.5 seconds (effective 609.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 106...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (106 compressed) at 0x00008000 in 0.0 seconds (effective 907.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
3.8_env/Scripts/python.exe;;C:/Users/ORENC/Private/Thread/esp-idf-v5.2-rc1/components/esptool_py/esptool/esptool.py;--chip;esp32c6 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/ORENC/Private/Thread/esp-idf-v5.2-rc1/examples/openthread/ot_br/build -P C:/Users/ORENC/Private/Thread/esp-idf-v5.2-rc1/components/esptool_py/run_serial_tool.cmake"
Executing action: monitor
Running idf_monitor in directory C:\Users\ORENC\Private\Thread\esp-idf-v5.2-rc1\examples\openthread\ot_br
Executing "C:\Users\ORENC\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe C:\Users\ORENC\Private\Thread\esp-idf-v5.2-rc1\tools/idf_monitor.py -p com6 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c6 --revision 0 --decode-panic backtrace C:\Users\ORENC\Private\Thread\esp-idf-v5.2-rc1\examples\openthread\ot_br\build\esp_ot_br.elf --force-color -m 'C:\Users\ORENC\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe' 'C:\Users\ORENC\Private\Thread\esp-idf-v5.2-rc1\tools\idf.py' '-p' 'com6'"...
--- esp-idf-monitor 1.4.0 on com6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xde0
load:0x4086e610,len:0x2e04
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2-rc1-dirty 2nd stage bootloader
I (24) boot: compile time Feb  7 2024 19:11:33
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 001a9000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=42138020 size=4fb08h (326408) map
I (159) esp_image: segment 1: paddr=0005fb30 vaddr=40800000 size=004e8h (  1256) load
I (160) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=13729ch (1274524) map
I (424) esp_image: segment 3: paddr=001972c4 vaddr=408004e8 size=169e0h ( 92640) load
I (445) esp_image: segment 4: paddr=001adcac vaddr=40816ed0 size=03c50h ( 15440) load
I (455) boot: Loaded app from partition at offset 0x10000
I (455) boot: Disabling RNG early entropy source...
I (467) cpu_start: Unicore app
W (476) clk: esp_perip_clk_init() has not been implemented yet
I (482) cpu_start: Pro cpu start user code
I (483) cpu_start: cpu freq: 160000000 Hz
I (483) cpu_start: Application information:
I (486) cpu_start: Project name:     esp_ot_br
I (491) cpu_start: App version:      v5.2-rc1-dirty
I (496) cpu_start: Compile time:     Feb  7 2024 19:11:07
I (502) cpu_start: ELF file SHA256:  c76836d87...
I (508) cpu_start: ESP-IDF:          v5.2-rc1-dirty
I (513) cpu_start: Min chip rev:     v0.0
I (518) cpu_start: Max chip rev:     v0.99
I (523) cpu_start: Chip rev:         v0.0
I (528) heap_init: Initializing. RAM available for dynamic allocation:
I (535) heap_init: At 4082C260 len 000503B0 (320 KiB): RAM
I (541) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (547) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (554) spi_flash: detected chip: generic
I (558) spi_flash: flash io: dio
W (562) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (575) sleep: Configure to isolate all GPIO pins in sleep state
I (582) sleep: Enable automatic switching of GPIO sleep configuration
[... 46 lines cut ...]
40839490: 0x00000004 0x00000013 0x00000012 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408394b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408394d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001 0x00000000 0x0001c200 0x00000003
408394f0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000004 0xffffffff 0xffffffff 0x00000000
40839510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40839530: 0x00000000 0x00000000 0x4213ad2c 0x00000a0a 0x00000000 0x00000000 0x00000000 0x40809366
40839550: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40839570: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150
40839590: 0x40838bb0 0x00000193 0x4081ad40 0x4081ad40 0x40839590 0x4081ad38 0x00000014 0x40839f98
[... cut ...]
40839670: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40839690: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0000 0x00000000
408396d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000060 0x00000000 0x00000000 0x40839748
408396f0: 0x40839758 0x40839768 0x42051920 0x420518ae 0x40839768 0x00000000 0x00000000 0x00000000
40839710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: c76836d87

Rebooting...

zwx1995esp, 2024-02-08:
I see, so could you please try these steps again on the `v5.2-rc1` commit `bd18b8ba6a`?

0.let's checkout the ESP-IDF to this commit
```
$ git checkout bd18b8ba6a
$ git submodule update --init --recursive
```
(Notice, please use `git status` to double check no files were modified.)

1.compile the RCP after removing all previous build files.
```
> cd $IDP_PATH/examples/openthread/ot_rcp
> rm -rf ./build sdkconfig
> idf.py set-target esp32h2
> idf.py -p <your-h2-port> build flash
```

2. compile the BR after removing all previous build files.
```
> cd $IDP_PATH/examples/openthread/ot_br
> rm -rf ./build sdkconfig
> idf.py set-target esp32c6
> idf.py menuconfig
```
Only enable this config, please do not modify any others: 
```
(Top) → Component config → OpenThread → Config the Thread radio type (Connect via UART) 
```
[image]
Then flash to board
```
> idf.py -p <your-c6-port> build flash
```

3. Connect C6 with BR:
```
C6      <----->   H2
GPIO4   <----->   TX
GPIO5   <----->   RX
GND     <----->   GND
```

4.Open the C6 monitor
```
idf.py -p <your-c6-port> monitor
```

zwx1995esp, 2024-02-08:
If all things go as expected, you will see
[image]

orenc75, 2024-02-08:
@zwx1995esp 
Thank you very much for the detailed instructions.
I followed your instructions and got the same problem I got same error I got before.
This is the error:
I (612) main_task: Returned from app_main()
W(2622) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout
W(4632) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout
E(4632) OPENTHREAD:[C] P-RadioSpinel-: Failed to reset RCP!
E(4642) OPENTHREAD:[C] Platform------: ResetRcp() at radio_spinel.cpp:229: Failure

abort() was called at PC 0x42006b1b on core 0
0x42006b1b: syscall_not_implemented_aborts at C:/Users/ORENC/Private/Thread/esp-idf/components/newlib/syscalls.c:27

Stack dump detected
Core  0 register dump:
MEPC    : 0x408004e8  RA      : 0x40808286  SP      : 0x40839380  GP      : 0x408176a0
0x408004e8: panic_abort at C:/Users/ORENC/Private/Thread/esp-idf/components/esp_system/panic.c:472
0x40808286: __ubsan_include at C:/Users/ORENC/Private/Thread/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407ea714  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x408393e4  A0      : 0x408393ac  A1      : 0x408393e2
A2      : 0x00000000  A3      : 0x408393d9  A4      : 0x00000001  A5      : 0x4082c000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x00000001
S4      : 0x00000000  S5      : 0x4081fb5c  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Thinking that my C6 is defective, I flashed the "blink" example to both C6 and H2, and both worked flawlessly.

zwx1995esp, 2024-02-09:
> @zwx1995esp Thank you very much for the detailed instructions. I followed your instructions and got the same problem I got same error I got before. This is the error: I (612) main_task: Returned [...]

So, could you please take a photo showing that how do you connect the C6  with H2?

orenc75, 2024-02-09:
Thank you

[image]

zwx1995esp, 2024-02-09:
could you please try to add this code after [this line](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/port/esp_openthread_uart.c#L66)?
```
        if (config->tx_pin != -1) {
            gpio_set_drive_capability(config->tx_pin, GPIO_DRIVE_CAP_3);
        }
        if (config->rx_pin != -1) {
            gpio_set_drive_capability(config->rx_pin, GPIO_DRIVE_CAP_3);
        }
```
(before calling the function `uart_set_pin`)
It seems your jump line is a little long, let set the gpio driver level to the max and then have a try, notice, both RCP and BR needs to re-flash with this changes.

orenc75, 2024-02-09:
I added the lines (Your marked line is at line #65 in my active code), and flashed both sides.
I still get the same error:
I (965) main_task: Returned from app_main()
W(2975) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout
W(4985) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout
E(4985) OPENTHREAD:[C] P-RadioSpinel-: Failed to reset RCP!
E(4995) OPENTHREAD:[C] Platform------: ResetRcp() at radio_spinel.cpp:229: Failure

abort() was called at PC 0x42006b1b on core 0
0x42006b1b: syscall_not_implemented_aborts at C:/Users/ORENC/Private/Thread/esp-idf/components/newlib/syscalls.c:27

Stack dump detected
Core  0 register dump:
MEPC    : 0x408004e8  RA      : 0x40808286  SP      : 0x40839380  GP      : 0x408176a0
0x408004e8: panic_abort at C:/Users/ORENC/Private/Thread/esp-idf/components/esp_system/panic.c:472
0x40808286: __ubsan_include at C:/Users/ORENC/Private/Thread/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407ea6e4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x408393e4  A0      : 0x408393ac  A1      : 0x408393e2
A2      : 0x00000000  A3      : 0x408393d9  A4      : 0x00000001  A5      : 0x4082c000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x00000001
S4      : 0x00000000  S5      : 0x4081fb5c  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000

Do you want me to find a shorter jump lines?

zwx1995esp, 2024-02-09:
> I added the lines (Your marked line is at line #65 in my active code), and flashed both sides. I still get the same error: I (965) main_task: Returned from app_main() W(2975) OPENTHREAD:[W] [...]

Oh, too bad, it does not make sense... You can have a try with a shorter jump line.

I compile the .bin file to you, and could you please try with this bin file building from my side? Let's try if the file built from my side can work or not. If it also did not work, it might be some HW issue related to your board.
[test.zip](https://github.com/espressif/esp-thread-br/files/14218325/test.zip)
Please download this .zip file and unzip to your local directory, for example, my folder is `~/playground/github_issue`
1. Flash to C6 using this command:
```
python -m esptool --chip esp32c6 -b 2000000 -p /dev/ttyUSB0 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 C6_BR/bootloader.bin 0x8000 C6_BR/partition-table.bin 0x10000 C6_BR/esp_ot_br.bin
```
[image]
then flash H2.
```
python -m esptool --chip esp32h2 -b 2000000 -p /dev/ttyUSB2 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 48m 0x0 RCP_H2/bootloader.bin 0x8000 RCP_H2/partition-table.bin 0x10000 RCP_H2/esp_ot_rcp.bin
```
[image]
Then connect the C6 with H2, also using this map:
```
C6      <----->   H2
GPIO4   <----->   TX
GPIO5   <----->   RX
GND     <----->   GND
```
[image]

zwx1995esp, 2024-02-09:
Tips: if you see an error "no module named esptool", You can install it via command `python -m pip install esptool`

orenc75, 2024-02-09:
ok, now it is working!

[image]

The message seems different from what you posted here: 
https://github.com/espressif/esp-thread-br/issues/56#issuecomment-1933299473

Is it ok? So I got a problem with my build process?

zwx1995esp, 2024-02-09:
> ok, now it is working!
> 
> ![image](https://private-user-images.githubusercontent.com/12273488/303583012-921bd081-7213-423b-8d4b-c9d79d21de12.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0NjQ4ODIsIm5iZiI6MTcwNzQ2NDU4MiwicGF0aCI6Ii8xMjI3MzQ4OC8zMDM1ODMwMTItOTIxYmQwODEtNzIxMy00MjNiLThkNGItYzlkNzlkMjFkZTEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA5VDA3NDMwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1ZWYyM2E3NzU0ZGYxZmQzM2FhMTQ0MjNjYjM4ZDY0NDQwZjk0MDBkZTJhYzZiZmM5ZDNmM2JiOGMxZDI3YWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OTc-GUWJMSrwFNJUNvTNBSDBj6sAcJfLs2k5CaXVmU0)
> 
> The message seems different from what you posted here: [#56 (comment)](https://github.com/espressif/esp-thread-br/issues/56#issuecomment-1933299473)
> 
> Is it ok? So I got a problem with my build process?

Yes, could you please double check with these steps again? 
[#56(comment)](https://github.com/espressif/esp-thread-br/issues/56#issuecomment-1933296338)

orenc75, 2024-02-09:
ok, I started a fresh installation on a different computer.
First, I built and compiled ot_rcp, then I used the monitor command to check (my ot_rcp, yours ot_br) and it worked.
Second, I build and compiled ot_br, then I used the monitor command to check (my ot_rcp, my ot_br) and it failed...

I documented the steps I did so maybe you can find what am I doing wrong. I am using windows environment.

1. Install ESP-IDF using windows installer from [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#get-started-step-by-step)

2. Update branch
```
git checkout bd18b8ba6a
git submodule update --init --recursive
```
git status results:
[image]

3. Build, flash and check ot_rcp
```
cd %IDF_PATH%/examples/openthread/ot_rcp
del sdkconfig
rmdir build /s/q
idf.py set-target esp32h2
idf.py -p com5 build flash
idf.py -p com6 monitor
```
Result: Success

4. Build, flash and check ot_br
```
cd %IDF_PATH%/examples/openthread/ot_br
del sdkconfig
rmdir build /s/q
idf.py set-target esp32c6
idf.py menuconfig : (Top) → Component config → OpenThread → Config the Thread radio type (Connect via UART) 
idf.py -p com6 build flash
idf.py -p com6 monitor
```
Result: Failed

orenc75, 2024-02-10:
I installed a fresh Linux installation and repeated the steps in https://github.com/espressif/esp-thread-br/issues/56#issuecomment-1933299473 .
Amazingly, but I still get the same error... I am not sure what am i doing wrong.

I compared the bin files I got from @zwx1995esp and the file size are different:

For ot_rcp:
                               my <---> zwx1995esp 
bootloader.bin :       19120 <---> 19280
partition-table.bin :  3072 <---> 3072
esp_ot_br.bin :       193312 <---> 195024

For ot_br:
                               my <---> zwx1995esp 
bootloader.bin :       21568 <---> 21584
partition-table.bin :  3072 <---> 3072
esp_ot_br.bin :       1709280 <---> 1696656

So I definitely missing something in the complication since it happen in both Windows and Linux and on fresh installations of es-idf for both.
Any idea?

orenc75, 2024-02-15:
An update:
I managed to make the full esp-thread-br to work on C6+H2.
I think the trick was to switch to non-stable version of esp-idf v5.3 and to follow [this](https://openthread.io/codelabs/esp-openthread-hardware#2) guide.

So now it is working. Thank you all.

orenc75, 2024-02-15:
It is working now

closed 2024-02-15 by orenc75