espressif/esp-thread-br #58: WEB GUI fails to compile for C6+H2 combination

orenc75, 2024-02-15:
Hi,

The thread border router is working great under v5.3 with C6 instead of S3 as the BR and H2 as the RCP. Connection via UART.
However, when I set "Enable the web server in Thread Border Router", the build fails with the following log:

Executing action: all (aliases: build)
Running ninja in directory /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32c6
Processing 5 dependencies:
[5/5] idf (5.3.0)
-- Project sdkconfig file /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/sdkconfig
Loading defaults file /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/sdkconfig.defaults...
warning: unknown kconfig symbol 'LWIP_FORCE_ROUTER_FORWARDING' assigned to 'y' in /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/sdkconfig.defaults
-- App "esp_ot_br" version: v1.0-25-gc7f0103
-- Adding linker script /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_system/ld/esp32c6/[sections.ld.in](http://sections.ld.in/)
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.api.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.rvfp.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.version.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.phy.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.coexist.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.net80211.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.pp.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.wdt.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.heap.ld
-- Adding linker script /home/eagle/esp_repeat/esp-idf/components/soc/esp32c6/ld/esp32c6.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_br_http_ota esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_ot_br_server esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-serial-flasher espressif__esp_ot_cli_extension espressif__esp_rcp_update espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 iperf json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table protobuf-c protocol_examples_common protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport thread_border_router ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/eagle/esp_repeat/esp-idf/components/app_trace /home/eagle/esp_repeat/esp-idf/components/app_update /home/eagle/esp_repeat/esp-idf/components/bootloader /home/eagle/esp_repeat/esp-idf/components/bootloader_support /home/eagle/esp_repeat/esp-idf/components/bt /home/eagle/esp_repeat/esp-idf/components/cmock /home/eagle/esp_repeat/esp-idf/components/console /home/eagle/esp_repeat/esp-idf/components/cxx /home/eagle/esp_repeat/esp-idf/components/driver /home/eagle/esp_repeat/esp-idf/components/efuse /home/eagle/esp_repeat/esp-idf/components/esp-tls /home/eagle/esp_repeat/esp-idf/components/esp_adc /home/eagle/esp_repeat/esp-idf/components/esp_app_format /home/eagle/esp_repeat/esp-idf/components/esp_bootloader_format /home/eagle/esp_repeat/esp-thread-br/components/esp_br_http_ota /home/eagle/esp_repeat/esp-idf/components/esp_coex /home/eagle/esp_repeat/esp-idf/components/esp_common /home/eagle/esp_repeat/esp-idf/components/esp_eth /home/eagle/esp_repeat/esp-idf/components/esp_event /home/eagle/esp_repeat/esp-idf/components/esp_gdbstub /home/eagle/esp_repeat/esp-idf/components/esp_hid /home/eagle/esp_repeat/esp-idf/components/esp_http_client /home/eagle/esp_repeat/esp-idf/components/esp_http_server /home/eagle/esp_repeat/esp-idf/components/esp_https_ota /home/eagle/esp_repeat/esp-idf/components/esp_https_server /home/eagle/esp_repeat/esp-idf/components/esp_hw_support /home/eagle/esp_repeat/esp-idf/components/esp_lcd /home/eagle/esp_repeat/esp-idf/components/esp_local_ctrl /home/eagle/esp_repeat/esp-idf/components/esp_mm /home/eagle/esp_repeat/esp-idf/components/esp_netif /home/eagle/esp_repeat/esp-idf/components/esp_netif_stack /home/eagle/esp_repeat/esp-thread-br/components/esp_ot_br_server /home/eagle/esp_repeat/esp-idf/components/esp_partition /home/eagle/esp_repeat/esp-idf/components/esp_phy /home/eagle/esp_repeat/esp-idf/components/esp_pm /home/eagle/esp_repeat/esp-idf/components/esp_psram /home/eagle/esp_repeat/esp-idf/components/esp_ringbuf /home/eagle/esp_repeat/esp-idf/components/esp_rom /home/eagle/esp_repeat/esp-idf/components/esp_system /home/eagle/esp_repeat/esp-idf/components/esp_timer /home/eagle/esp_repeat/esp-idf/components/esp_wifi /home/eagle/esp_repeat/esp-idf/components/espcoredump /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp-serial-flasher /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_rcp_update /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__mdns /home/eagle/esp_repeat/esp-idf/components/esptool_py /home/eagle/esp_repeat/esp-idf/components/fatfs /home/eagle/esp_repeat/esp-idf/components/freertos /home/eagle/esp_repeat/esp-idf/components/hal /home/eagle/esp_repeat/esp-idf/components/heap /home/eagle/esp_repeat/esp-idf/components/http_parser /home/eagle/esp_repeat/esp-idf/components/idf_test /home/eagle/esp_repeat/esp-idf/components/ieee802154 /home/eagle/esp_repeat/esp-idf/examples/common_components/iperf /home/eagle/esp_repeat/esp-idf/components/json /home/eagle/esp_repeat/esp-idf/components/log /home/eagle/esp_repeat/esp-idf/components/lwip /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/main /home/eagle/esp_repeat/esp-idf/components/mbedtls /home/eagle/esp_repeat/esp-idf/components/mqtt /home/eagle/esp_repeat/esp-idf/components/newlib /home/eagle/esp_repeat/esp-idf/components/nvs_flash /home/eagle/esp_repeat/esp-idf/components/nvs_sec_provider /home/eagle/esp_repeat/esp-idf/components/openthread /home/eagle/esp_repeat/esp-idf/components/partition_table /home/eagle/esp_repeat/esp-idf/components/protobuf-c /home/eagle/esp_repeat/esp-idf/examples/common_components/protocol_examples_common /home/eagle/esp_repeat/esp-idf/components/protocomm /home/eagle/esp_repeat/esp-idf/components/pthread /home/eagle/esp_repeat/esp-idf/components/riscv /home/eagle/esp_repeat/esp-idf/components/sdmmc /home/eagle/esp_repeat/esp-idf/components/soc /home/eagle/esp_repeat/esp-idf/components/spi_flash /home/eagle/esp_repeat/esp-idf/components/spiffs /home/eagle/esp_repeat/esp-idf/components/tcp_transport /home/eagle/esp_repeat/esp-thread-br/examples/common/thread_border_router /home/eagle/esp_repeat/esp-idf/components/ulp /home/eagle/esp_repeat/esp-idf/components/unity /home/eagle/esp_repeat/esp-idf/components/usb /home/eagle/esp_repeat/esp-idf/components/vfs /home/eagle/esp_repeat/esp-idf/components/wear_levelling /home/eagle/esp_repeat/esp-idf/components/wifi_provisioning /home/eagle/esp_repeat/esp-idf/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/build
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32c6
[... cut ...]
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_3229 and /home/eagle/esp_repeat/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_3229

It seems like the defined partitions are too small to accumulate the web server.
Maybe the partition definitions are wrong for C6 as BR? I am using ESP32-C6-WROOM-1 MAN8 and ESP32-H2-MINI1 MBN4.

orenc75, 2024-02-15:
ok, I fixed it by setting:
Serial flasher config > Flash size > 8 MB

Also, I changed the partitions.csv to increase the partitions size of ota_0 & ota_1 to be 1664k

I am not sure what the defaults for S3, but maybe you would like to change the default settings for C6?

chshu, 2024-02-18:
@orenc75 The current S3 module on Thread BR board is 4M, the default partition table for S3 is the same to C6, but the C6 firmware seems bigger than S3. So please make the change in your own C6 project.

We will probably produce new Thread BR board with 8MB S3 module, we will change the default partition table then.

chshu, 2024-02-18:
The current Thread BR board with 4M S3 could support Web GUI and OTA, but not too much flash space left.

OTA feature could be disabled during development phase, and `ota_1` partition is not required, so more space left for `ota_0`.

orenc75, 2024-02-18:
OK, thanks for your answer. I am closing the issue.

closed 2024-02-18 by orenc75