espressif/esp-thread-br #60: Build files for ESP Thread Border Router v1.2 board

djorr5, 2024-02-19:
Hi,

Are there build files available for the above board? I would like to add additional circuitry to it without having to re-do all the vias for RF tuning.

chshu, 2024-02-19:
@djorr5 The schematic file is available here: https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#related-documents.

Could you specify what kinds of build files are you looking for? We will check internally if these can be shared.

djorr5, 2024-02-19:
The gerbers basically. I could use the PDF schematic to create them but
ideally the already created ones would be a massive time saver. Also you
have a lot of vias on the board, I assume for RF and you board would be
tuned already. I don't want to have to do that part again.

chshu, 2024-02-19:
@djorr5 The NDA may be required to share the gerber file, could you please get in touch with us via https://www.espressif.com/en/contact-us/sales-questions.

djorr5, 2024-02-21:
Hi. Every time I try to submit the form at sales-question it fails.

chshu, 2024-02-22:
@djorr5 Could you send an email to sales@espressif.com?

martin8310, 2024-02-26:
I would be happy about these files too! Tasmota does that too ;-)
[image]

chshu, 2024-05-17:
Please contact us via https://www.espressif.com/en/contact-us/sales-questions to request the build/gerber files.

closed 2024-05-17 by chshu