espressif/esp-thread-br #62: WEB GUI issue "ErrorMessage": "404 Not Found"

martin8310, 2024-02-26:
Hello,

1. is the doc correct "https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html"? Somone told me that "idf.py menuconfig" is enough to activate the web gui.

2. menu "ESP Thread Border Router Example --> Enable the web server in Thread Border Router." is activited but the failure is still aviable.
{
    "ErrorCode": "404",
    "ErrorMessage": "404 Not Found"
}

3. console log:
[1B][0;32mI (1576433) obtr_web: -------------------------------------------[1B][0m
[1B][0;32mI (1576433) obtr_web: /[1B][0m
[1B][0;32mI (1578593) OPENTHREAD: Received PIO[1B][0m
[1B][0;32mI (1582253) OPENTHREAD: Received PIO[1B][0m
[1B][0;32mI (1582253) OPENTHREAD: Received RIO[1B][0m
[1B][0;32mI (1582253) OPENTHREAD: prefix FD31:99BC:430:1:: lifetime 1800
[1B][0m
[1B][0;32mI (1585673) OPENTHREAD: Received PIO[1B][0m

regards
Martin 
[idf_py_stdout_output_20944.log](https://github.com/espressif/esp-thread-br/files/14402619/idf_py_stdout_output_20944.log)

zwx1995esp, 2024-02-26:
Hi @martin8310 Could you please show me the website you access on the host? And also show me the IPv4 address on the BR? Also please check if the the BR and the host are connected to the same AP.

martin8310, 2024-02-26:
Hello zwx1995esp,

The esp is definitely on the same network. Querying the API also works.

[image]
[image]

martin8310, 2024-02-26:
@zwx1995esp

Thanky, now it works ;-) "http://192.168.1.22:80/index.htm"

martin8310, 2024-02-26:
okay, thank you. now it looks good. But doc is not up to date I think. "Enable the `CONFIG_OPENTHREAD_BR_START_SERVER` option to enable the Web Server feature" is not necessary.

I think this is the correct code in the file "sdkconfig"
`CONFIG_OPENTHREAD_BR_START_WEB=y`

zwx1995esp, 2024-02-26:
> I think this is the correct code in the file "sdkconfig" `CONFIG_OPENTHREAD_BR_START_WEB=y`

Thanks for letting us know about that, we will update the docs soon.

zwx1995esp, 2024-02-26:
> @zwx1995esp
> 
> It works only one time with "http://192.168.1.22:80/index.html". And now the same issue ;-(

Oh, bad things, So could you please try to ping the BR on your host when you get the `404` error?
Just `ping 192.168.1.22` on your host to double check the connection between your host and BR.

martin8310, 2024-02-26:
Hello zwx,

i,m little bit confuesd now, with "http://192.168.1.22/index.html" it works. I hope it will stay alive so :-)

Ping is also fine:
C:\Users\marti\esp\v5.2\esp-idf>ping 192.168.1.22

Ping wird ausgeführt für 192.168.1.22 mit 32 Bytes Daten:
Antwort von 192.168.1.22: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.1.22: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.1.22: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.1.22: Bytes=32 Zeit=3ms TTL=64

closed 2024-02-26 by martin8310