espressif/esp-thread-br #63: [Question] Auto Start Thread network with specified hex value

mmt-10, 2024-03-07:
Hi,

As my understanding, WiFi SSID/PASS can set menuconfig.
I would like to use the specified hex value which `dataset active -x` generates with auto start.
How can I set hex value in menuconfig?

zwx1995esp, 2024-03-08:
Hi, we do not support setting the whole hex dataset in auto start mode. You can have a try with this API to set your dataset `otDatasetSetActiveTlvs`, for details of this api, you can refer to [openthread source code](https://github.com/openthread/openthread/blob/main/include/openthread/dataset.h#L417).

mmt-10, 2024-03-08:
@zwx1995esp 
Thank you for your reply.
I check source code of `esp-thread-br`.
I will write hex dataset to below line, it must be overwrite `dataset`.
[otOperationalDatasetTlvs dataset;](https://github.com/espressif/esp-thread-br/blob/c7f01039780fb2f72cfcd8f015ce52e922469ab0/examples/common/thread_border_router/src/border_router_launch.c#L128)

BTW, I would like to add hex dataset option in menuconfig.
how do I add uint8_t array option in menuconfig?
[otOperationalDatasetTlvs](https://github.com/espressif/openthread/blob/af5938e389be40650507748272bb6c6b3a2de2cf/include/openthread/dataset.h#L264)

zwx1995esp, 2024-03-08:
> @zwx1995esp Thank you for your reply. I check source code of `esp-thread-br`. I will write hex dataset to below line, it must be overwrite `dataset`. [otOperationalDatasetTlvs [...]

How about just add it in your program directly? I mean that you can define a macro in your program, for example:
```
#define OPENTHREAD_ACTIVE_DATASET_MANUAL "0e080000000000010000000300001835060004001fffe002081ed74143359b24030708fd721e9dfc4af9d605107ca397ebdbe65f37b510c9ff037d39bd030f4f70656e5468726561642d36333738010263780410a88354f3e63c96d227d694d7dc9030f80c0402a0f7f8"
```
And you can parse this string macro in some C codes, also you can refer this [API](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread.cpp#L43C15-L43C35) in our ESP idf. 

Also some logics are the same as parsing the active dataset, you can refer to [these](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread.cpp#L135-L147)

mmt-10, 2024-03-08:
Thank you for your reply and advice.
I will try it.

closed 2024-03-08 by mmt-10