espressif/esp-thread-br #64: basic_thread_border_router build fails with ethernet support

purha, 2024-03-07:
basic_thread_border_router build fails when disabling wifi and enabling ethernet support with error:

> fatal error: esp_eth.h: No such file or directory
>    17 | #include "esp_eth.h"
>       |          ^~~~~~~~~~~
> compilation terminated.
> 
> Compilation failed because protocol_examples_common.h (in "protocol_examples_common" component) includes esp_eth.h, provided by esp_eth component(s).
> However, esp_eth component(s) is not in the requirements list of "protocol_examples_common".
> To fix this, add esp_eth to REQUIRES list of idf_component_register call in /code/thread-border-router/esp-idf/examples/common_components/protocol_examples_common/CMakeLists.txt.

couldn't get it working by adding esp_eth to the CMakeLists, any guidance how to build it?

zwx1995esp, 2024-03-08:
Hi @purha , could you please share which idf commit and esp-thread-br commit did you use? Did you try the idf branch v5.1.2? I just compile the esp-thread-br[commit c7f010397] with the idf v5.1.2[commit 482a8fb2d7], disable the config `EXAMPLE_CONNECT_WIFI`, enable the config `EXAMPLE_CONNECT_ETHERNET`, no errors come out.

purha, 2024-03-08:
For esp-idf, I use v5.1.2 / 482a8fb2d7, and for esp-thread-br I'm using latest from main, tried commit c70103 but the same issue remains (ran clean beforehand)

here's the relevant lines from sdkconfig:

> # CONFIG_EXAMPLE_CONNECT_WIFI is not set
> CONFIG_EXAMPLE_CONNECT_ETHERNET=y
> CONFIG_EXAMPLE_ETHERNET_EMAC_TASK_STACK_SIZE=2048
> CONFIG_EXAMPLE_USE_SPI_ETHERNET=y
> # CONFIG_EXAMPLE_USE_DM9051 is not set
> CONFIG_EXAMPLE_USE_W5500=y
> # CONFIG_EXAMPLE_USE_OPENETH is not set

I'm trying to build this on mac os sonoma, apple m1

here's the error:

> [ 98%] Built target __idf_wifi_provisioning
> [ 98%] Built target __idf_espressif__esp_ot_cli_extension
> [100%] Building C object esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj
> [100%] Linking C static library libesp_ot_br_server.a
> [100%] Built target __idf_esp_ot_br_server
> In file included from /code/thread-border-router/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:42:
> /code/thread-border-router/esp-idf/examples/common_components/protocol_examples_common/include/protocol_examples_common.h:17:10: fatal error: esp_eth.h: No such file or directory
>    17 | #include "esp_eth.h"
>       |          ^~~~~~~~~~~
> compilation terminated.
> make[3]: *** [esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj] Error 1
> make[2]: *** [esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/all] Error 2
> make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
> make: *** [flash] Error 2
> Compilation failed because protocol_examples_common.h (in "protocol_examples_common" component) includes esp_eth.h, provided by esp_eth component(s).
> However, esp_eth component(s) is in the private requirements list of "protocol_examples_common".
> To fix this, move esp_eth from PRIV_REQUIRES into REQUIRES list of idf_component_register call in /code/thread-border-router/esp-idf/examples/common_components/protocol_examples_common/CMakeLists.txt.
> make failed with exit code 2, output of the command is in the /code/thread-border-router/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_8843 and /code/thread-border-router/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_8843

here's the stderr:

> Compiler supported targets: xtensa-esp32s3-elf
> 
> In file included from /code/thread-border-router/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:42:
> /code/thread-border-router/esp-idf/examples/common_components/protocol_examples_common/include/protocol_examples_common.h:17:10: fatal error: esp_eth.h: No such file or directory
>    17 | #include "esp_eth.h"
>       |          ^~~~~~~~~~~
> compilation terminated.
> make[3]: *** [esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj] Error 1
> make[2]: *** [esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/all] Error 2
> make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
> make: *** [flash] Error 2

kepstin, 2024-03-22:
I am seeing this same error when building on Linux using esp-idf 5.1.2

When I build with commit ec65dc8a17a6a4a0006871d3c5a7d5ba43801486 then the build is successful
When I build with commit c7f01039780fb2f72cfcd8f015ce52e922469ab0 then the build fails, with the error as described.

The configuration changes I have made are select `OPENTHREAD_RADIO_SPINEL_UART`, enable `OPENTHREAD_BR_AUTO_START` and `OPENTHREAD_BR_START_WEB`, disable `EXAMPLE_CONNECT_WIFI`, enable `EXAMPLE_CONNECT_ETHERNET` (and set ethernet SPI device configuration)

zwx1995esp, 2024-03-22:
> I am seeing this same error when building on Linux using esp-idf 5.1.2
> 
> When I build with commit [ec65dc8](https://github.com/espressif/esp-thread-br/commit/ec65dc8a17a6a4a0006871d3c5a7d5ba43801486) then the build is successful When I build with commit [c7f0103](https://github.com/espressif/esp-thread-br/commit/c7f01039780fb2f72cfcd8f015ce52e922469ab0) then the build fails, with the error as described.
> 
> The configuration changes I have made are select `OPENTHREAD_RADIO_SPINEL_UART`, enable `OPENTHREAD_BR_AUTO_START` and `OPENTHREAD_BR_START_WEB`, disable `EXAMPLE_CONNECT_WIFI`, enable `EXAMPLE_CONNECT_ETHERNET` (and set ethernet SPI device configuration [according to documentation](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#ethernet-based-thread-border-router))

Hi, did you try the esp-thread-br v1.0 tags? In our docs, we recommend to use `esp-thread-br tags v1.0` with `esp-idf tags v5.1.2`.

kepstin, 2024-03-22:
The v1.0 tag works with esp-idf v5.1.2, yes.

Should I use a different version of esp-idf if i want to build the main branch of esp-thread-br? Which version, 5.2.1?

zwx1995esp, 2024-03-25:
Hi, @kepstin I forgot to enable `OPENTHREAD_BR_AUTO_START`, with enabling this, I reproduce this issue. Will fix it asap.

chshu, 2024-03-27:
The compile issue has been fixed by https://github.com/espressif/esp-thread-br/commit/6d1b5662dca0a0230095a3314ec50e2ff85ca828

closed 2024-03-27 by chshu