espressif/esp-thread-br #65: DNS IPv4 Address Resolution on ESP32 Border Router

cedriczou13, 2024-03-11:
I'm utilizing the `otDnsClientResolveIp4Address to initiate a DNS query for resolving the IPv4 address of a specified hostname from a end device. This functionality operates successfully on my Raspberry Pi-based border router. However, I'm encountering issues when using an ESP32 border router. Could you inform me if this functionality is supported on the ESP32 platform and guide me on how to properly implement it? 

thanks

Cedric

zwx1995esp, 2024-03-12:
Yes we support this API, so could you please show us more details about your issue?

1. Which type of board did you use? The ESP32-S3 or just ESP32? 
2. And the code base, did you try the esp-thread-br tag v1.0 and esp-idf tag v5.1.2? 
3. How could your issue be reproduced? Could you please share some steps to reproduce your encounted issue?

cedriczou13, 2024-03-12:
Hi thanks for you support, please find bellow the details of my setup and the problem I'm facing:

I am using the ESP Thread Border Router with ESP-S3 and ESP32-H2.
The software versions: ESP-THREAD-BR tag v1.0 and ESP-IDF tag v5.1.2.

**Setup Description:**
My network setup consists of one ESP Thread Border Router and one end-thread device (MTD) based on the nRF52840 chipset.
The objective is to resolve IPv4 addresses via DNS queries. While this functionality works as expected on a Raspberry Pi-based Border Router, it fails on the ESP32 Border Router. This discrepancy suggests that the implementation on the  end-thread device is correct.

**Configurations:**

a. ESP Border Router:
Configuration settings applied:
```
CONFIG_OPENTHREAD_DNS64_CLIENT=y
CONFIG_OPENTHREAD_DNS_SERVER_ADDR="8.8.8.8"
```
I can successfully resolve IP addresses using the CLI command dns resolve4 example.com 8.8.8.8. However, I encounter an UDP message error during this process, the relevance of which is unclear to me.
[image]

b. On the End Device:

Upon connection to the border router, I execute the following commands:
otSrpClientGetServerAddress()
otSrpClientEnableAutoStartMode()
otDnsClientResolveIp4Address(ot->instance, "example.com", otDnsAddressCallback1, &synt_ipv6_addr, NULL)
In the callback function otDnsAddressCallback1, I receive after fews minute a DNS resolution error code 28 (OT_ERROR_RESPONSE_TIMEOUT), indicating a timeout error.

If additional information or clarification is needed, please let me know.

Thank you for your time and assistance.

zwx1995esp, 2024-03-13:
Hi Cedric, copy that info, I will try to reproduce your issue ASAP, and if I have any process, I will sync here.

zwx1995esp, 2024-03-14:
Hi, sorry I did not reprodece on my side, all things work well on my side with the ESP BR.(But, I do not have a nrf board, so for the thread end device I use an ESP32H2 instead, but I do not thing there are some difference...)

Here is the logs on the H2(thread end device):
```
> 
> dns resolve4 google.com 8.8.8.8

Synthesized IPv6 DNS server address: fd8e:8656:3b2e:2:0:0:808:808
DNS response for google.com. - fd8e:8656:3b2e:2:0:0:8efa:cf0e TTL:300 
Done
> ping fd8e:8656:3b2e:2:0:0:8efa:cf0e

16 bytes from fd8e:8656:3b2e:2:0:0:8efa:cf0e: icmp_seq=3 hlim=112 time=251ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 251/251.0/251 ms.
Done
> 
```
[image]

And this command `dns resolve4 google.com 8.8.8.8` is a thread scope command, might not run on the BR. For the BR if you want to resolve some host using DNS, you need to use some upper layer command. This is a same issue like BR can not ping the host address using a thread ping command. For more details you can refer to this [discussion ](https://github.com/orgs/openthread/discussions/9688#discussioncomment-7771734) on openthread.

So one thing that I need your help:

Do you have a ESP32H2 or ESP32C6 board? Could you please try the thread end device with our ESP boards?

We have an [OT_CLI example](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli) on the IDF, and you can still use the IDF branch v5.1.2.

zwx1995esp, 2024-04-02:
Hi, @cedriczou13 Are there some updates from your side? Does this feature work on your env? If any other questions, feel free to let us know.

chshu, 2024-05-09:
@cedriczou13 Hope the https://github.com/espressif/esp-thread-br/issues/65#issuecomment-1996896888 addressed your issue, feel free to reopen if you have any follow up questions.

closed 2024-05-09 by chshu