espressif/esp-thread-br #66: ESP32S3 is continually rebooting

JoseAntonioMG, 2024-03-30:
I am installing "Esp thread border router" in a standalone kit with an Esp32s3 and an Esp32c6, connected with SPI interface. However, once the flash is done, the serial monitor shows:

```
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (79) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (94) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (102) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (109) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=4c848h (313416) map
I (186) esp_image: segment 1: paddr=0006c870 vaddr=3fc99100 size=037a8h ( 14248) load
I (189) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=ebdbch (966076) map
I (365) esp_image: segment 3: paddr=0015bde4 vaddr=3fc9c8a8 size=01450h (  5200) load
I (366) esp_image: segment 4: paddr=0015d23c vaddr=40374000 size=150b4h ( 86196) load
I (398) boot: Loaded app from partition at offset 0x20000
I (399) boot: Disabling RNG early entropy source...
I (399) cpu_start: Multicore app
I (403) cpu_start: Pro cpu up.
I (406) cpu_start: Starting app cpu,�

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a17
0x40375a17: esp_restart_noos_dig at C:/Users/JoseAntonio/esp/v5.1.3/esp-idf/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (31) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (31) boot: compile time Mar 30 2024 19:34:24
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (99) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (106) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (114) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (121) boot: End of partition table
I (126) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=4c848h (313416) map
I (190) esp_image: segment 1: paddr=0006c870 vaddr=3fc99100 size=037a8h ( 14248) load
I (193) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=ebdbch (966076) map
I (369) esp_image: segment 3: paddr=0015bde4 vaddr=3fc9c8a8 size=01450h (  5200) load
I (371) esp_image: segment 4: paddr=0015d23c vaddr=40374000 size=150b4h ( 86196) load
I (403) boot: Loaded app from partition at offset 0x20000
I (403) boot: Disabling RNG early entropy source...
I (404) cpu_start: Multicore app
I (407) cpu_start: Pro cpu up.
I (411) cpu_start: Starting app cpu,�

```
ESP32S3 is continually rebooting.

I am following the instructions listed here:

[ ESP thread BR » Development Guide  » 2.1. Build and Run](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/index.html)

Please, can you help me?

gytxxsy, 2024-04-01:
The examples in esp-thread-br are usually used for the ESP Thread Border Router/Zigbee Gateway Board ( [hardware](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#wi-fi-based-thread-border-router) ). It is not recommended for standalone Kits (such as pairing an S3 board with a C6 board). If you have to do it, you will need to manually wire the necessary pins to ensure all required connections are made (refer to [here](https://dl.espressif.com/dl/schematics/esp_thread_br_zigbee_gw_schematiccs_v1.2.pdf) ).

[image]

chshu, 2024-05-09:
@JoseAntonioMG Hope the https://github.com/espressif/esp-thread-br/issues/66#issuecomment-2029269223 helped, feel free to reopen for any follow up questions.

closed 2024-05-09 by chshu

JoseAntonioMG, 2024-05-09:
It is working correctly, ESP32S3 + ESP32C6, I have made the connections as shown in the graphic of @gytxxsy 
Thank you very much for the information