espressif/esp-thread-br #67: Update ESP32H2 via UART from ESP32S3

djorr5, 2024-04-29:
Hi,

Question, I have designed a board that has a H2 and S3 and have it working fine for Zigbee gateway application. I was checking on wether I can update the H2 chip from the S3 without SPI integrated. 

In my serial interface the logs show that it does recognize the version of the OT-RCP and tries to update it, but fails to update.

I noticed in the Readme.md page that:

 "Note that the SPI GPIOs are optional, if the UART is selected for communication."

But just want to confirm that SPI connection is also needed to complete the auto update of RCP.

zwx1995esp, 2024-05-06:
Hi, @djorr5 We support the SPI connection and also support the auto update of RCP when using SPI.

chshu, 2024-05-09:
@djorr5 The RCP auto update is done over the UART interface, the same UART interface can also be used for Host-RCP communication. So the SPI connection is optional.

Feel free to reopen if any follow up questions.

closed 2024-05-09 by chshu