espressif/esp-thread-br #68: Failure to set active dataset

motters, 2024-05-23:
Hi,

We've been copy'ing and past the openthread br code into our project which uses ESP-IDF v5.1 with ESP-Ardunio. 

Everything works as expected however it is not possible to set to set a new dataset active, see below:
[image]

We have tried setting the active dataset through code and though the cli and both fail to set it.
[image]
 

Our question is:
  - Does ESP-Ardunio (including Preference class) clash with 'esp_openthread_settings.c'?
  - Is there anything else that could stop the dataset becoming active?

Cheers!

motters, 2024-05-24:
Using nvs_tool.py i can conform he OT dataset is being wrote to flash
[image]

So the question is why can't it read the dataset from flash.

motters, 2024-05-24:
I think the problem is solved. port_config.storage_partition_name was changed from ot_storage to nvs

This commit was a breaking change our orginal BR months back to the current commit 
https://github.com/espressif/esp-thread-br/commit/f3c952a9ce369f5d6b78e27dd764c8af11fb3d19

motters, 2024-06-04:
This was indeed the issue encase anyone else has it.

closed 2024-06-04 by motters

zwx1995esp, 2024-06-04:
Hi, sorry for missing this issue, we recommend using esp-thread-br v1.0 based IDF v5.1.3. So could you please update your codebase?