espressif/esp-thread-br #69: ESP32+ESP32C6 构成的边界路由器测试问题

wnylei, 2024-06-01:
问题现象：当启用ç配置时， 边界路由器不能自动下载rcp固件，不断重启。 关闭Update RCP automatically配置后，边界路由器可正常工作。

日志：
I(918) OPENTHREAD:[I] P-RadioSpinel-: RCP reset: RESET_POWER_ON
I(928) OPENTHREAD:[I] P-RadioSpinel-: Software reset RCP successfully
I (898) main_task: Returned from app_main()
I(968) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I (978) esp_ot_br: Internal RCP Version: openthread-esp32/e7771c75bd-456c44828; esp32c6;  2024-06-01 02:36:09 UTC
I (978) esp_ot_br: Running  RCP Version: openthread-esp32/e7771c75bd-456c44828; esp32c6;  2024-06-01 02:24:23 UTC
I (998) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (1008) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1008) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (2968) RCP_UPDATE: esp_rcp_update(247): Failed to connect to RCP

硬件连接情况：
**ESP32**           **ESP32C6**
GPIO12--------->RST(EN)
GPIO15--------->BOOT(GPIO9)
GPIO13--------->TX（U0TXD）
GPIO14--------->RX(U0RXT)

请问这种情况大概是什么问题导致的？  也尝试手动触发RST和BOOT按钮使C6进行下载模式，但没有什么作用。

wnylei, 2024-06-01:
大概看了 自动更新RCP的部分代码，好像不支持 ESP32C6作为RCP 来自动更新固件。 请问ESP32C6 预计什么时候能支持呢？

wnylei, 2024-06-02:
1，修改依赖组件 espressif/esp-serial-flasher 版本为 "~1.3.0"。
2，修改 esp_ot_config.h 中的 .target_chip = ESP32C6_CHIP。
可以解决这个问题0.0

zwx1995esp, 2024-06-03:
> 1，修改依赖组件 espressif/esp-serial-flasher 版本为 "~1.3.0"。 2，修改 esp_ot_config.h 中的 .target_chip = ESP32C6_CHIP。 可以解决这个问题0.0

是的，要是使用C6作为RCP进行自动更新，就需要修改 esp_ot_config.h 。

closed 2024-06-03 by wnylei