espressif/esp-thread-br #70: Openthread Custom API

im22361, 2024-06-01:
Hello, 

I have been able to set up the ESP32 border router with ESP-C6 and I have been able to send messages using the OpenThread commands.  

I have used the example here for the border router: https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router example 
and the examples here for the node:  https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli example. 

I would like to automatically send and receive messages containing sensor readings. 
How can I integrate custom functions and deactivate the command line so that I can do this? 

I can see that esp provides ot_common_components example, but its  unclear how this can be integrated.

Any support would be appreciated.

zwx1995esp, 2024-06-03:
You can find some helps here [openthread.io cli command reference](https://openthread.io/reference/cli/commands), for any commands you manually set to board, there might be a function call in openthread stack. 

For examples, the command [`thread start`](https://openthread.io/reference/cli/commands#thread_start) is related to the function [`otThreadSetEnabled`](https://openthread.io/reference/group/api-thread-general#group__api-thread-general_1ga20a68a51ce7299b1c74c81812a4854a3).

Tips: the openthread APIs are not thread-safe, so if you want to call some openthread APIs in other task, [the ot lock](https://github.com/espressif/esp-idf/blob/master/components/openthread/include/esp_openthread_lock.h#L48) should be acquired first and released after the finish.

im22361, 2024-06-03:
Thank you for your response
My query was more related to the best way to implement a custom API  such as 
 ot_common_components(https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_common_components/ot_led/) that ESP have provided. 

You mention that openthread APIs are not thread-safe does that mean if we want to implement customAPI we should not use xTaskCreate to run our custom APIs? 

Any support would help thank you

zwx1995esp, 2024-06-03:
> You mention that openthread APIs are not thread-safe does that mean if we want to implement customAPI we should not use xTaskCreate to run our custom APIs?
> 
> Any support would help thank you

Not yet, you can create a task to run the openthread APIs, but when you want to do that, you need to call the function `esp_openthread_lock_acquire` first then call the functions then call `esp_openthread_lock_release` when finished. The lock will make the thread safe.

zwx1995esp, 2024-06-03:
> Thank you for your response My query was more related to the best way to implement a custom API such as ot_common_components(https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_common_components/ot_led/) that ESP have provided.
> 
And for this example, we only create the state changed callback to do some process when thread state changed, there is no logic related to start the thread network. I mean in your application, you need to join a specific thread network first, this can be done via calling the openthread APIs shown in the link. And for your application task you can implement some specific logic to so something you like(maybe as you metioned, send and receive messages containing sensor readings).

Also you can register a state changed callback for doing that. Anyway you like.

closed 2024-07-05 by chshu