espressif/esp-thread-br #74: 边界路由器项目中，添加mqtt组件后，mqtt连接提示Host is unreachable。

wnylei, 2024-06-28:
### 硬件：
ESP32+ESP32C6

### 软件：
[esp-thread-br](https://github.com/espressif/esp-thread-br)

### 情况：
我尝试在**esp-thread-br**示例项目中添加mqtt组件，但当它运行之后，mqtt组件会出现 **Host is unreachable** 不可达的提示。此刻wifi是正常连接的，如下图所示：

[image]

然后，当我把 **launch_openthread_border_router(&platform_config, &rcp_update_config);** 这句代码屏蔽后，mqtt组件是可以正常连接服务器的。如下图所示：
[image]

### 问题：

1. 请问这是什么原因导致的?
2. 我应该如何修改代码，可以实现在esp-thread-br项目中，正常的使用mqtt组件，访问网络？

### 请求指教，万分感谢！

wnylei, 2024-06-30:
### 补充
我发现了它 #71 ，和它应该是同一个问题。

zwx1995esp, 2024-07-01:
Hi @wnylei 所以你遇到的问题也是和[#71](https://github.com/espressif/esp-thread-br/issues/71) 中描述的，在调用esp_netif_set_default_netif这个函数之后，BR就没办法通过getaddrinfo获取云服务器的ip地址了是嘛？

wnylei, 2024-07-01:
> Hi @wnylei 所以你遇到的问题也是和[#71](https://github.com/espressif/esp-thread-br/issues/71) 中描述的，在调用esp_netif_set_default_netif这个函数之后，BR就没办法通过getaddrinfo获取云服务器的ip地址了是嘛？

是的。另外，我尝试打开mqtt的调试日志之后，发现mqtt在发起连接请求的时候，是可以解析到正确的ip地址的（也有可能是DNS缓存？），然后就是一直提示**主机不可达**。  去掉**esp_netif_set_default_netif**之后，就一切正常了。

zwx1995esp, 2024-07-01:
> > Hi @wnylei 所以你遇到的问题也是和[#71](https://github.com/espressif/esp-thread-br/issues/71) 中描述的，在调用esp_netif_set_default_netif这个函数之后，BR就没办法通过getaddrinfo获取云服务器的ip地址了是嘛？
> 
> 是的。另外，我尝试打开mqtt的调试日志之后，发现mqtt在发起连接请求的时候，是可以解析到正确的ip地址的（也有可能是DNS缓存？），然后就是一直提示**主机不可达**。 去掉**esp_netif_set_default_netif**之后，就一切正常了。

Ok, 感谢您帮忙确认这个问题， @wnylei  可以麻烦再帮忙做一个实验嘛？就是打印一下不可达的主机地址是什么，这个地址有可能用的不是DNS刚刚解析的ipv4地址，我怀疑这里面有点问题。

如果这个地址也是预期的，我这边尝试复现一下，然后可以提供一下您这边是调用的哪个上层接口去和云进行建立链接的嘛？非常感谢

wnylei, 2024-07-01:
@zwx1995esp 

打开debug日志后，解析到的ip地址是期望的IP地址。如下：
```
D (44273) esp-tls: [sock=57] Resolved IPv4 address: 82.157.23.88
D (44283) esp-tls: [sock=57] Connecting to server. HOST: gateway-test.afnsmart.com, Port: 9883
E (44283) esp-tls: [sock=57] connect() error: Host is unreachable
E (44293) esp-tls: Failed to open new connection
E (44303) transport_base: Failed to open a new connection
E (44313) mqtt_client: Error transport connect
D (44313) event: running post MQTT_EVENTS:0 with handler 0x400f0ac8 and context 0x3ffe714c on loop 0x3ffe7090
0x400f0ac8: u_mqtt_event_handler at /Users/wanghm/Code/nl/esp/thread-esp-project/projects/normal_br/components/u_common_mqtt/src/u_common_mqtt.c:42
```
另外，我使用的是官方的mqtt组件，先进行初始化，随后调用 **esp_mqtt_client_start**API，开始建立链接。

 ### 太客气了，这是我应该做的。有任何疑问或需要的测试，随时联系。

zwx1995esp, 2024-07-01:
好的，我们这边后续尝试复现一下这个问题，有进展会在这两个issue中同步。

zwx1995esp, 2024-07-01:
有一个初步的猜想，这个ipv4的地址，lwip层不知道route是什么，esp_netif_set_default_netif这个不设置的话，默认的default netif是wifi，这个地址会放到wifi侧建立tcp，但是加了这个esp_netif_set_default_netif之后，可能会跑到thread侧建立，就会失败。我们复现一下，再debug看看。

wnylei, 2024-07-01:
> 有一个初步的猜想，这个ipv4的地址，lwip层不知道route是什么，esp_netif_set_default_netif这个不设置的话，默认的default netif是wifi，这个地址会放到wifi侧建立tcp，但是加了这个esp_netif_set_default_netif之后，可能会跑到thread侧建立，就会失败。我们复现一下，再debug看看。

是的，我也是这个想法。 
我之前大概看了下代码，
对于默认的 **wifi：netif** 它的 route_prio 是 100. 
对于默认的 **eth：netif（有线接口）** 它的 route_prio 是50. 
对于当前示例项目中 **thread：netif** 它的route_prio是 15.
我猜测应该是这个优先级的问题？所以去掉 **esp_netif_set_default_netif** 之后，不论是使用wifi还是eth，都能mqtt正常联接互联网。

wnylei, 2024-07-01:
我目前更想知道增加 esp_netif_set_default_netif（openthread_netif） 这行代码的目的是什么，去掉它会影响什么功能嘛。
当我把它屏蔽掉之后，经过简单的测试，发现thread网络和wifi网络都可以很好的工作。
当然，我还没有进行全部功能的测试。

zwx1995esp, 2024-07-01:
这个是当时BR过Thread官方的certification的时候，有一些multicast的地址再做forwarding的时候，没有正确的forward到Thread层，所以加了这个函数。这个问题的解决方案，可能需要我们先复现一下，然后内部评估一下。如果您的项目中不涉及multicast的转发，可以临时注释这个函数作为workaround。

chshu, 2024-09-06:
Duplicate with https://github.com/espressif/esp-thread-br/issues/71, closing.

@wnylei please update your default netif as required in your application, we will follow up in https://github.com/espressif/esp-thread-br/issues/71

closed 2024-09-06 by chshu