espressif/esp-thread-br #75: esp-thread-br built successfull but

jeffbrown57, 2024-06-30:
I  purchased the espressif thread border router board ( the one with esp32s3 + esp32h2 on one card) , built ot-rcp , built
the esp-thread-br ( using esp-idf 5.1.3 ) flashed it , formed a thread network with web-gui , i can see it using the discovery app
on iphone _meshcop._udp ( esp-ot-br ) homeassistant sees it as a Thread device , when i attempt to add an onvis s4  matter over thread device using homekit all i ever get is , you need a "Thread Border Router" error message on iphone trying to add accessory.
I took the smart plug to a friends who has a google nest mini ( with built in thread border router ) and it adds it to "google home" successfully and works great. Anybody have any ideas to share to help ???

Thanks a million

chshu, 2024-06-30:
You may need to install the OpenThread border router integration to homeassistant, the discussion here may help: https://community.home-assistant.io/t/cant-add-otbr-credentials-on-ha/721986

jeffbrown57, 2024-06-30:
> 
You may need to install the OpenThread border router integration to homeassistant, the discussion here may help: https://community.home-assistant.io/t/cant-add-otbr-credentials-on-ha/721986

So i attempted to install the OTBR integration and i got "Already configured. Only a single configuration possible."

I find this all very confusing, i previously installed integration "Thread" which shows my esp board as a Border Router
under Thread integration, i assumed this was the one the OTBR integration was referring to as "already configured" so
i deleted that and restarted HAOS , upon restart i had a notification that the Thread device was found , i "ignored" that
and proceeded to attempt installing OTBR integration which failed in the same way  "Already configured. Only a single configuration possible." . Help me understand what the "Thread" integration is and how does it co-exist with "OTBR"
integration ( to be certain im referring tio Integrations and NOT to "Addons"

jeffbrown57, 2024-06-30:
Does the actual HAOS integration OTBR assume that you have an RCP dongle (esp32h2) physically plugged into
the computer running homeassistant ? i currently do not , my esp board flashed with esp_thread_br is a separate device

DejinChen, 2024-07-02:
Hi, so there is no "Open Thread Border Router" integration in your homeassistant, but you can not install it due to ""Already configured. Only a single configuration possible", right?

The homeassistant i used was running on raspberry pi 4. Following the [instruction](https://community.home-assistant.io/t/cant-add-otbr-credentials-on-ha/721986), i could add Thread device to homeassistant.
"Open Thread Border Router" integration requires the REST API of Thread border, so homeassistant can get and set the dataset of TBR. In "Thread" integration, the found border router could be added to the preferred network and homeassistant can use it to commissioning Thread device.
[image].

chshu, 2024-08-02:
@jeffbrown57 Hope https://github.com/espressif/esp-thread-br/issues/75#issuecomment-2202342516 helped, feel free to reopen if any follow up questions.

closed 2024-08-02 by chshu