espressif/esp-thread-br #76: Web GUI isn't running after startup

jeremyherbert, 2024-07-03:
Hi,

I have the ESP thread border router dev kit with the W5500 addon. I have built and flashed the firmware using the instructions here: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html (including the exact esp-idf version). I have also enabled the Web interface in the sdkconfig, but it doesn't run the web server on startup (nmap says the port is closed, it doesn't work in my browser either). I searched the `sdkconfig` file for `CONFIG_OPENTHREAD_BR_START_SERVER` but it is not present. I have attached the generated file.

I am able to ping the device over the ethernet connection and the serial interface is working fine too.

[sdkconfig_br.txt](https://github.com/user-attachments/files/16076777/sdkconfig_br.txt)

zwx1995esp, 2024-07-03:
Hi, could you please share some logs from your BR? That might be helpful to know what happens.

zwx1995esp, 2024-07-03:
Another tips, did you add the port in the url explicitly? For example: `http://192.168.200.98:80/index.html`

jeremyherbert, 2024-07-04:
Here is the boot log:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a0b
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (26) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (26) boot: compile time Jul  3 2024 03:30:57
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (78) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (86) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (93) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (101) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (108) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (116) boot: End of partition table
I (120) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=56468h (353384) map
I (192) esp_image: segment 1: paddr=00076490 vaddr=3fc99100 size=051b8h ( 20920) load
I (197) esp_image: segment 2: paddr=0007b650 vaddr=40374000 size=049c8h ( 18888) load
I (202) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=100dd8h (1052120) map
I (395) esp_image: segment 4: paddr=00180e00 vaddr=403789c8 size=106d0h ( 67280) load
I (419) boot: Loaded app from partition at offset 0x20000
I (420) boot: Disabling RNG early entropy source...
I (420) cpu_start: Multicore app
I (424) cpu_start: Pro cpu up.
I (427) cpu_start: Starting app cpu, entry point is 0x403755f8
I (0) cpu_start: App cpu up.
I (443) cpu_start: Pro cpu start user code
I (443) cpu_start: cpu freq: 160000000 Hz
I (443) cpu_start: Application information:
I (443) cpu_start: Project name:     esp_ot_br
I (443) cpu_start: App version:      v1.0-41-g48b2e47-dirty
I (444) cpu_start: Compile time:     Jul  3 2024 03:44:22
I (444) cpu_start: ELF file SHA256:  b008406bc33837b1...
I (444) cpu_start: ESP-IDF:          v5.1.3
I (444) cpu_start: Min chip rev:     v0.0
I (445) cpu_start: Max chip rev:     v0.99
I (445) cpu_start: Chip rev:         v0.2
I (445) heap_init: Initializing. RAM available for dynamic allocation:
I (445) heap_init: At 3FCACF88 len 0003C788 (241 KiB): DRAM
I (446) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (446) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (446) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (448) spi_flash: detected chip: generic
I (448) spi_flash: flash io: dio
I (448) sleep: Configure to isolate all GPIO pins in sleep state
I (449) sleep: Enable automatic switching of GPIO sleep configuration
I (449) app_start: Starting scheduler on CPU0
I (450) app_start: Starting scheduler on CPU1
I (450) main_task: Started on CPU0
I (460) main_task: Calling app_main()
I (540) RCP_UPDATE: RCP: using update sequence 0
I (540) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (540) OPENTHREAD: spinel UART interface initialization completed
I (540) main_task: Returned from app_main()
I(540) OPENTHREAD:[I] P-RadioSpinel-: RCP reset: RESET_POWER_ON
I(550) OPENTHREAD:[I] P-RadioSpinel-: Software reset RCP successfully
I(590) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I(600) OPENTHREAD:[I] Settings------: Read NetworkInfo {rloc:0xb400, extaddr:a2a5326caefa6ec2, role:leader, mode:0x0f, version:4, keyseq:0x0, ...
I(600) OPENTHREAD:[I] Settings------: ... pid:0x4c8a61da, mlecntr:0x2b30, maccntr:0xbc6, mliid:b691c08d603609ce}
I (610) esp_ot_br: Internal RCP Version: openthread-esp32/e7771c75bd-456c44828; esp32h2;  2024-07-03 02:46:45 UTC
I (610) esp_ot_br: Running  RCP Version: openthread-esp32/e7771c75bd-456c44828; esp32h2;  2024-07-03 02:46:45 UTC
I (620) OPENTHREAD: OpenThread attached to netif
> I (640) esp_eth.netif.netif_glue: 02:00:00:12:34:56
I (640) esp_eth.netif.netif_glue: ethernet attached to netif
I (650) ethernet_connect: Waiting for IP(s).
I (2650) ethernet_connect: Ethernet Link Up
I (3650) esp_netif_handlers: example_netif_eth ip: 192.168.1.179, mask: 255.255.255.0, gw: 192.168.1.1
I (3650) ethernet_connect: Got IPv4 event: Interface "example_netif_eth" address: 192.168.1.179
I (4540) ethernet_connect: Got IPv6 event: Interface "example_netif_eth" address: fe80:0000:0000:0000:0000:00ff:fe12:3456, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4540) example_common: Connected to example_netif_eth
I (4540) example_common: - IPv4 address: 192.168.1.179,
I (4540) example_common: - IPv6 address: fe80:0000:0000:0000:0000:00ff:fe12:3456, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I(4540) OPENTHREAD:[N] RoutingManager: BR ULA prefix: fd2b:587a:7efe::/48 (loaded)
I(4550) OPENTHREAD:[N] RoutingManager: Local on-link prefix: fd82:1da8:68ac:2f3::/64
I (4560) OPENTHREAD: Platform UDP bound to port 49153
I(4560) OPENTHREAD:[N] Mle-----------: Role disabled -> detached
I (4590) OT_STATE: netif up
I (4600) OPENTHREAD: NAT64 ready
```

Yes, I did try to put the port in explicitly, but it didn't change the outcome.

I should also note that the docs say that the kconfig is `CONFIG_OPENTHREAD_BR_START_SERVER` but in reality it looks to be actually `CONFIG_OPENTHREAD_BR_START_WEB`: https://github.com/espressif/esp-thread-br/blob/5d692a398877c0c4ed425580af1394f0d5d230b7/examples/basic_thread_border_router/main/esp_ot_br.c#L46C5-L46C35

jeremyherbert, 2024-07-04:
I think it's possible that this has always been broken? The line here is waiting for a wifi IP address, but I don't think it will ever trigger if an ethernet address is assigned because it is waiting for the event `IP_EVENT_STA_GOT_IP` but not `IP_EVENT_ETH_GOT_IP`.

https://github.com/espressif/esp-thread-br/blob/5d692a398877c0c4ed425580af1394f0d5d230b7/components/esp_ot_br_server/src/esp_br_web.c#L1222

jeremyherbert, 2024-07-04:
Turns out that is indeed the problem, please see the fix in my PR.

closed 2024-07-05 by espressif-bot