espressif/esp-thread-br #78: Form thread network in Web GUI doesn't work

jeremyherbert, 2024-07-04:
I am trying to form a network using the web GUI and the default settings in the GUI, but I get an error message:

[image]

[image]

It appears that the webserver on the ESP32-S3 (thread border router dev kit) is rejecting the POST request:

[image]

jeremyherbert, 2024-07-04:
The fix is to go into `menuconfig` and change `Component config` => `HTTP Server` => `Max HTTP Request Header Length` to a bigger number (I changed from 512 to 2048). Perhaps the default values should be updated in the project? My HTTP headers are nothing unusual, so it seems like the default value is not large enough.

zwx1995esp, 2024-09-06:
Hi, we have the tests on our ENV using Firefox version:  129.0.2(64bit), all things work fine. I think this issue should be related to the browse. Change the menuconfig seems work.

chshu, 2024-09-19:
@jeremyherbert It seems not a SDK issue, you can change the default configuration in your application.

closed 2024-09-19 by chshu