espressif/esp-thread-br #79: Bi-directional IPv6 connectivity in Codelab have only uni-directional code example

digidhamu, 2024-07-09:
I could not get the ping command to work from the BR Thread CLI when connecting to Linux or Mac IPv6 addresses. Could you please add more steps for bi-directional connectivity using the ping command? This example only covers one direction, but not from the CLI. 

[3.1. Bi-directional IPv6 Connectivity](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/connectivity.html#bi-directional-ipv6-connectivity)

zwx1995esp, 2024-07-09:
Hi, @digidhamu Did you ping the host address from thread BR device? If yes, I hope the explaination here: https://github.com/orgs/openthread/discussions/9688#discussioncomment-7771734 can help you a lot.

zwx1995esp, 2024-07-09:
And the examples related in the docs, it shows that the topology should be like:
```
           Wi-Fi           Thread
(Host) <----------> (BR) <===========> (Thread CLI device)
```

The (Thread CLI device) device and (Host) can access from each other.

digidhamu, 2024-07-09:
> Hi, @digidhamu Did you ping the host address from thread BR device? If yes, I hope the explaination here: https://github.com/orgs/openthread/discussions/9688#discussioncomment-7771734 Hope it can help you a lot.

Yes. Would be great if we add this in the document with simple example

digidhamu, 2024-07-09:
> And the examples related in the docs, it shows that the topology should be like:
> 
> ```
>            Wi-Fi           Thread
> (Host) <----------> (BR) <===========> (Thread CLI device)
> ```
> 
> The (Thread CLI device) device and (Host) can access from each other.

Thread CLI device works perfectly. However, BR also uses Thread CLI hence the confusion

closed 2024-09-06 by chshu