espressif/esp-thread-br #80: Unable to get the Thread network set up with the ESP TBR

pavel808, 2024-07-11:
I have the ESP TBR : https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html

I am following the instructions here to build and flash the host ESP32-C3 SOC as follows : https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#configure-esp-thread-border-router

I am having  issues. The documentation is not clear. I'd apreciate any help with this. Thanks in advance.

- Is it necessary to flash the ESP32-H2 with the $IDF_PATH/examples/openthread/ot_rcp application? 

- After flashing the host ESP32-C3, and setting it up, is it also necessary to access the terminal on the ESP32-H2 and perform these actions here? : https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#attach-the-cli-device-to-the-thread-network

- When I configure the ESP32-C3 application to connect to my wi-fi network, it connects fine. However, when it reboots, the connection is no longer there. Is this normal behavior, and if so, how can I get it to retain the connection between bootups ? 

-  I have tried to commission a Matter device via chip-tool and the Thread network without any success as follows : `chip-tool pairing ble-thread 0x7293 hex:0e080000000000010000000300001435060004001fffe0020834e3cb7122aa4c300708fd5a7186ca4db72c0510429b56c68b2e5c96bd450f682378f4a6030f4f70656e5468726561642d3562636301025bcc04104344baf60e321da843a3d97cf2796e060c0402a0f7f8 20202021 3840` 

Does this command look ok ? It fails as follows : 

`Run command failure: examples/chip-tool/commands/common/CHIPCommand.cpp:617: CHIP Error 0x00000032: Timeout`

chshu, 2024-07-17:
@pavel808 If you have the ESP Thread Border Router board, please follow this instruction step by step: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run. In this case, both S3 and H2 firmwares will be combined and flashed to S3, S3 will download the firmware to H2 over UART automaticlly on the first boot.

If you don't have the ESP Thread Border Router board, but want to setup with standalong boards, please follow this instruction: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br#hardware-required, connect the two SoCs via uart, then flash ot_br firmware to host SoC, flash ot_rcp firmware to H2 SoC, so the two will work together as a Thread BR.

To test matter Thread commissioning, you need a BR setup (either the ESP ESP Thread Border Router board or two standalong SoCs combination) and a Thread matter device (ESP32-H2), the matter command looks good, please ensure the Matter Thread device is ready and broadcasting the BLE service.

pavel808, 2024-07-17:
@chshu Again, thanks for your reply. I have the ESP Thread Border Router board which has both the S3 and H2 . I have followed the instructions here, flashed and configured the S3 as instructed : https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run 

For my Thread end devices I have these ESP32-C6 development boards : https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html#getting-started . 
These should support Thread as an end device according to the documentation? 

The device is ready and is broadcasting the BLE service when I try to commission. I can commission via Wi-fi without any issues, just via Thread it is failing as in the post above. I'm not sure what is going wrong here.

chshu, 2024-07-18:
follow up in https://github.com/espressif/esp-thread-br/issues/81

closed 2024-07-18 by chshu